zwave-js/zwave-js #4466: Often dead FIBARO Roller Shutter 2 nodes in Zwave-js

vladimiro13, 2022-04-13:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

After migration from open zwave to zwave-js I have problems with often dead state with all my fibaro Roller Shutter 2 (fw 25.25).  I have zwave gw AEOTEC Z-Stick Gen5. Sometimes the nodes fall for only a few minutes, sometimes for a longer period and no action helps and they have to recover on their own (various long time). Automation for zwave ping on all nodes every 10 minutes and active sending of reports (power / status) from fibaro nodes didn't helped me. 
I attach a sample of dead state of one of the shutter nodes (Node 010), a similar situation is for each one node (at different times).
I also attach a part of the silly log at the time of the crash the one (node 10) of the nodes
[image]
[dead_node_10_20220413-11:35:04_log_silly.txt](https://github.com/zwave-js/node-zwave-js/files/8481253/dead_node_10_20220413-11.35.04_log_silly.txt)
.

### Device information

Manufacturer: Fibaro
Model name: Roller Shutter 2
Node ID in your network: Node 010
Zwave GW: AEOTEC Z-Stick Gen5
Driver version: 8.11.9
Server version: 1.15.0
Home assistant version: 2022.4.1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: homeassistant/amd64-addon-zwave_js:0.1.55

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I migrated zwave nodes from open zwave to zwave-js.

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

2022-04-13T09:34:02.141Z SERIAL « 0x010c0004000606310504220000e5                                      (14 bytes)
2022-04-13T09:34:02.142Z CNTRLR   [Node 006] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
2022-04-13T09:34:02.144Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Power: 0 => 0                  [Endpoint 0]
2022-04-13T09:34:02.146Z SERIAL » [ACK]                                                                   (0x06)
2022-04-13T09:34:02.147Z DRIVER « [Node 006] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
[... 366 lines cut ...]
2022-04-13T09:35:11.280Z SERIAL « 0x01070013870000066a                                                 (9 bytes)
2022-04-13T09:35:11.281Z SERIAL » [ACK]                                                                   (0x06)
2022-04-13T09:35:11.282Z DRIVER « [REQ] [SendData]
                                    callback id:     135
                                    transmit status: OK
2022-04-13T09:35:11.286Z CNTRLR « [Node 014] ping successful
2022-04-13T09:35:15.330Z SERIAL « 0x010c0004000e06310504220007ea                                      (14 bytes)
2022-04-13T09:35:15.331Z CNTRLR   [Node 014] [Multilevel Sensor] Power: metadata updated            [Endpoint 0]
2022-04-13T09:35:15.331Z CNTRLR   [Node 014] [~] [Multilevel Sensor] Power: 0.7 => 0.7              [Endpoint 0]
2022-04-13T09:35:15.331Z SERIAL » [ACK]                                                                   (0x06)
2022-04-13T09:35:15.332Z DRIVER « [Node 014] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Power
                                      scale: Watt
                                      value: 0.7
2022-04-13T09:35:29.091Z SERIAL « 0x0110000400080a32022144000000d8000064                              (18 bytes)
2022-04-13T09:35:29.093Z CNTRLR   [Node 008] [~] [Meter] value[65537]: 2.16 => 2.16                 [Endpoint 0]
2022-04-13T09:35:29.095Z SERIAL » [ACK]                                                                   (0x06)
2022-04-13T09:35:29.096Z DRIVER « [Node 008] [REQ] [ApplicationCommand]
                                  └─[MeterCCReport]
                                      type:       Electric
                                      scale:      kWh
                                      rate type:  Consumed
                                      value:      2.16
                                      time delta: 0 seconds

AlCalzone, 2022-04-13:
Please **attach** the logfile as a file and don't just paste it into the issue text.

That said, your stick is temporarily not able to get a response from the device (NoACK), which is usually an indicator of a bad connectivity. Pinging the nodes just works around the symptoms.

I suggest you do the following:
1. read the intro to the troubleshooting guide https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index
2. make sure you've followed the general tips here: https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting - a bad controller placement can easily cause what you're experiencing
3. Try doing a network heal, especially if you've moved the controller as part of step 2.
4. If that doesn't help, use zwavejs2mqtt to check the health of each node's route to the controller (node 1). You might find some don't have a strong link

vladimiro13, 2022-04-13:
Hi. Just a while ago, several points fell again, so I'm adding a newer log again.
Thanks for help.
[zwave_js_new.log](https://github.com/zwave-js/node-zwave-js/files/8482780/zwave_js_new.log)

AlCalzone, 2022-04-14:
> And before that I didn't have this problem for about 3 years via open zwave

I've lost track of how often I had this same argument. You did have the problem, you probably just didn't notice, since the commands are retried internally and best case all you see are longer delays. I've used OZW before I started Z-Wave JS and it simply didn't surface the communication errors as visibly as Z-Wave JS does.

`NoAck` means the communication between the controller and node is disrupted physically, as in packets get lost on the air. The host software cannot influence this at all. This instance where node 7 retries sending its report a few times while the controller is having trouble reaching node 8 looks like this is a bidirectional problem:
```
# outgoing message to node 8
2022-04-13T14:55:00.015Z DRIVER » [Node 008] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      219
                                  └─[NoOperationCC]
...
2022-04-13T14:55:00.026Z DRIVER « [RES] [SendData]
                                    was sent: true

# node 7 tries to reach the controller a few times but doesn't get an acknowledgement
2022-04-13T14:55:01.254Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Power
                                      scale: Watt
                                      value: 0
...
2022-04-13T14:55:01.263Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Power
                                      scale: Watt
                                      value: 0
...
2022-04-13T14:55:01.758Z DRIVER « [Node 007] [REQ] [ApplicationCommand]
                                  └─[MultilevelSensorCCReport]
                                      type:  Power
                                      scale: Watt
                                      value: 0

# controller failed to reach node 8
2022-04-13T14:55:02.148Z DRIVER « [REQ] [SendData]
                                    callback id:     219
                                    transmit status: NoAck
```

I'm not sure if you followed my advice from the previous post already, especially the part about the controller placement and USB extension:
> [zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting) - a bad controller placement can easily cause what you're experiencing

If not, I strongly suggest you start with that - if I had to estimate, the success rate of that simple change is like 80-90%.

The information isn't included in your log, but since many nodes seem to go dead at the same time, I'd guess that most of them route through a single node and that node is having problems. Optimizing the controller placement and then healing the network can alleviate this problem by giving the nodes good alternative routes.

vladimiro13, 2022-04-15:
Hi AlCalzone, thanks for your reply and time. 

As for the location of the controller, as I wrote before the distance of controller to the node is from 7 to 30 meters, while the frequency of problems is the same for most nodes, even the nearest ones.
But I have a USB controller behind the TV, so I'll try to change the position and I will see if there is a change.

I also want to ask you about the mentioned healing of the network or roads to nodes. Did you mean action/button "Heal Network" in Diagnostics options?
Or how can I affect these paths between zwave nodes?

Thanks.

AlCalzone, 2022-04-16:
> As for the location of the controller, as I wrote before the distance of controller to the node is from 7 to 30 meters, while the frequency of problems is the same for most nodes, even the nearest ones.

The distance does not matter if the controller or a node is affected by radio interference. https://www.usb.org/sites/default/files/327216.pdf gives some details, including the noise spectrum of USB 3.0, which is actually the highest around the 900 MHz that Z-Wave uses:
[image]

Frequent reports and bulk commands that are repeated due to connectivity issues certainly make the matter worse.

> I also want to ask you about the mentioned healing of the network or roads to nodes. Did you mean action/button "Heal Network" in Diagnostics options?
Or how can I affect these paths between zwave nodes?

In zwavejs2mqtt, you can either heal the entire network (Actions -> Advanced -> Heal Network) or individual problematic nodes (Open node dropdown -> Advanced -> Heal Node). Both actions try to optimize the routes.

You can also test the connection quality of each node on the network map:
[image]

vladimiro13, 2022-04-16:
Hi AlCalzone.
Thanks for the answers. I will try to change the position of the USB controller next week and let me know if it helped. According to the description, this can really be related, as I use USB 3.1 ports on the intel nuc with a directly connected USB zwave dongle. I will try to use a USB extension cable.
I have also just increased the frequency of reports, I will see if the dead status will not increase again.

Next question, to render a real zwave map I need to install zwavejs2mqtt and this is not a problem besides running zwave-js configuration?

Thanks.

AlCalzone, 2022-04-19:
You can switch between the two addons, both work with the official Z-Wave JS integration:
https://www.home-assistant.io/integrations/zwave_js/#how-do-i-switch-between-the-official-z-wave-js-add-on-and-zwavejs2mqtt

vladimiro13, 2022-04-26:
Hi AlCalzone. 
I am happy to confirm that by changing the position of the zwave USB stick away from Intel nuc (and USB 3 ports), my problem with the frequent dead status of all nodes has been eliminated. Thank you for your help.

closed 2022-04-26 by vladimiro13