zwave-js/zwave-js #4468: Vision Security Door Sensors state sometimes not correct

dawiinci, 2022-04-14:


robertsLando, 2022-04-14:
https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs

dawiinci, 2022-04-14:
Here it is:
[zwavejs.log](https://github.com/zwave-js/node-zwave-js/files/8487844/zwavejs.log)

AlCalzone, 2022-04-14:
> Interestingly if I refresh the values in the interface, it will report the correct state at wakeup.

Yeah, that's because you instructed zwave-js to ask the device for all its values.

There is no communication at all with the device in your log. In order to troubleshoot, the log needs to run long enough so it contains the situation where the sensor status becomes incorrect.

dawiinci, 2022-04-16:
The state was frozen at 'on' from 15:55 until 18:08.

The time of the log is - 2 hours.

I have deactivated the manipulation entity in HA. But that shouldn't make a difference.

It is Node 025 (see first and second opening of door - the first one stayed open for HA):

```
2022-04-16T13:55:35.971Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 255
2022-04-16T13:55:35.972Z CNTRLR   [Node 025] treating BasicCC::Set as a report
2022-04-16T13:55:35.972Z CNTRLR   [Node 025] [~] [Binary Sensor] Any: false => true                 [Endpoint 0]
2022-04-16T13:55:35.990Z SERIAL « 0x0112000400190a710507ff00ff07020000ae0022                          (20 bytes)
2022-04-16T13:55:35.991Z CNTRLR   [Node 025] [~] [Notification] notificationMode: "pull" [Endpoint 0] [internal]
                                   => "pull"
2022-04-16T13:55:35.991Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T13:55:36.002Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   Home Security
                                      notification status: 255
                                      notification state:  Intrusion
2022-04-16T13:55:36.003Z CNTRLR   [Node 025] [~] [Notification] Home Security[Sensor status]: 2 =>  [Endpoint 0]
                                  2
2022-04-16T13:55:36.015Z SERIAL » 0x010e00137e07600d00022501ff254b41                                  (16 bytes)
2022-04-16T13:55:36.015Z DRIVER » [Node 126] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      75
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      0
                                    │ destination: 2
                                    └─[BinarySwitchCCSet]
                                        target value: true
2022-04-16T13:55:36.017Z SERIAL « [ACK]                                                                   (0x06)
2022-04-16T13:55:36.023Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-04-16T13:55:36.023Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T13:55:36.023Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-04-16T13:55:36.065Z SERIAL « 0x011800134b00000501c37f7f7f7f000003490000000301000030              (26 bytes)
2022-04-16T13:55:36.066Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T13:55:36.066Z DRIVER « [REQ] [SendData]
                                    callback id:            75
                                    transmit status:        OK, took 50 ms
                                    repeater node IDs:      73
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 100 kbit/s
                                    ACK RSSI:               -61 dBm
                                    ACK RSSI on repeaters:  N/A
                                    ACK channel no.:        0
                                    TX channel no.:         0
...

2022-04-16T13:55:40.028Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 0
2022-04-16T13:55:40.039Z SERIAL « 0x0112000400190a7105070000ff07020000b000c3                          (20 bytes)
2022-04-16T13:55:40.040Z CNTRLR   [Node 025] [~] [Notification] notificationMode: "pull" [Endpoint 0] [internal]
                                   => "pull"
2022-04-16T13:55:40.040Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T13:55:40.040Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[NotificationCCReport]
                                      notification type:   Home Security
                                      notification status: 255
                                      notification state:  Intrusion
2022-04-16T13:55:40.153Z SERIAL « 0x0112000400190a7105070000ff07020000b000c3                          (20 bytes)
2022-04-16T13:55:40.154Z CNTRLR   [Node 025] [~] [Notification] notificationMode: "pull" [Endpoint 0] [internal]
                                   => "pull"
2022-04-16T13:55:40.154Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T13:55:40.154Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
[... 20 lines cut ...]
                                      notification state:  Intrusion
2022-04-16T16:08:44.719Z CNTRLR   [Node 025] [~] [Notification] Home Security[Sensor status]: 2 =>  [Endpoint 0]
                                  2

2022-04-16T16:08:48.338Z SERIAL « 0x010b0004001903200100b0007b                                        (13 bytes)
2022-04-16T16:08:48.339Z SERIAL » [ACK]                                                                   (0x06)
2022-04-16T16:08:48.339Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[BasicCCSet]
                                      target value: 0
2022-04-16T16:08:48.340Z CNTRLR   [Node 025] treating BasicCC::Set as a report
[... cut ...]
                                  └─[NotificationCCReport]
                                      notification type:   Home Security
                                      notification status: 255
                                      notification state:  Intrusion
2022-04-16T16:08:48.377Z CNTRLR   [Node 025] [~] [Notification] Home Security[Sensor status]: 2 =>  [Endpoint 0]
                                  2
2022-04-16T16:09:03.303Z SERIAL « 0x010f0004001d07600d02022503ffbc00e6                                (17 bytes)

```

dawiinci, 2022-04-18:
If I interpret this correctly, the state has been sent correctly. However HA, did not reflect this, right?

dawiinci, 2022-04-19:
Please have a look at the raw file:
[zwavejs_2022-04-16.log](https://github.com/zwave-js/node-zwave-js/files/8514886/zwavejs_2022-04-16.log)

AlCalzone, 2022-04-20:
Hmm in that log I see `treating BasicCC::Set as a report` under every incoming `BasicCCSet` command. Some of these are kept as is and `currentValue` gets updated. Some are also mapped to the `Binary Sensor` value, depending on the device class: `[Node 025] treating BasicCC::Set as a report` and some of the devices also send a `Binary Sensor Report` prior to the Basic CC Set, which is why the Basic CC Set doesn't always cause the binary sensor to be updated (because it already did).

I can't find any updates in the log that aren't reflected as a state change.

AlCalzone, 2022-04-21:
Upon further investigation, this is probably a duplicate of https://github.com/zwave-js/node-zwave-js/issues/4467. Please check if this is fixed for you in v9.0.3, if not feel free to reopen.

closed 2022-04-21 by AlCalzone