zwave-js/zwave-js #4478: Getting "ZWaveError: Not a valid NVM3 page! (ZW0283)" when trying nvm2json

keassol, 2022-04-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Trying to use nvm2json to convert .bin file

What do you observe?
Error: ZWaveError: Not a valid NVM3 page! (ZW0283)

What did you expect to happen?
File to be converted to .json file

Steps to reproduce the behavior:
  1. Load zwavejs2mqtt and create NVM backup from control panel or use the Aeotec backup tool
  2. npx @zwave-js/nvmedit nvm2json --in ./500Old.bin --out ./500Old.json --verbose
  3. get error "ZWaveError: Not a valid NVM3 page! (ZW0283)"

### Device information

Manufacturer: Aeotec
Model name: ZStick 5
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

Driver Version: 8.11.9
Server Version: 1.15.0

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Not necessarily changed anything.  Have been successfully using HomeAssistant zwave_js integration but temporarily installed zwavejs2mqtt to extract the NVM backup .bin.  Currently using Aeotec zstick5 and trying to use the zwave-js/nvmedit tool to convert bin file for use on a Zooz ZST10 700 controller. Also extracted the bin file from the Aeotec backup tool with the same results.

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

using nvmedit tool - not relevant.

keassol, 2022-04-18:
Sorry should have left full error:

`ZWaveError: Not a valid NVM3 page! (ZW0283)
    at readPage (/home/poser/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/nvm3/page.js:21:15)
    at parseNVM (/home/poser/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/nvm3/nvm.js:22:57)
    at nvmToJSON (/home/poser/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/convert.js:568:36)
    at Object.handler (/home/poser/.npm/_npx/98595b58b0c35f8f/node_modules/@zwave-js/nvmedit/build/cli.js:41:40) {
  code: 283,
  context: undefined,
  transactionSource: undefined
`

AlCalzone, 2022-04-19:
Which firmware version is your stick on? Converting the NVM is only supported for SDK 6.61+, which means the stick must be on firmware 1.2.

keassol, 2022-04-19:
You are right, it looks like that is the problem.  The stick is an original z-stick 5 (not 5+) and there doesn't appear to be a firmware update available for it anywhere that I can find.  

As best as I can tell the only path forward is to migrate to a 5+ stick using the Aeotec backup/restore tool first before being able to migrate it to the Zooz 700.  Thank you for letting me know and thank you for your work on this project.

AlCalzone, 2022-04-19:
Aeotec pulled the update due to some issues. If you ask them, they might just give it to you anyways.

keassol, 2022-04-20:
That is a great suggestion.  I have submitted a ticket with them requesting it.  Thanks again!

closed 2022-04-20 by keassol