zwave-js/zwave-js #4479: Voltage Calibration

sbur83, 2022-04-19:
Hi, I have an Australian version of the 'Aeotec, Smart Switch 6' added to zwave2mqtt home assistant add-on. The model No. ZW110-B09 is not listed as a supported device however it is discovered and included in my device list and functions as expected. 
The problem is the voltage value is reporting 275v and the outlet is 240v (confirmed with a multi meter).
Could someone please advise if and how I can calibrate this to the correct value?
Thanks
[image]
.

AlCalzone, 2022-04-20:
> The problem is the voltage value is reporting 275v and the outlet is 240v

You're going to have to ask Aeotec about this. If the device reports 275V (and I don't see any config parameters for this [here](https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x0060:0.0)), this is an issue with the device.

sbur83, 2022-04-21:
> > is not listed as a supported device however it is discovered and included in my device list and functions as expected
> 
> I think there should be a configuration value for that, if your device is not listed in db configuration values are missing

...Thanks, device is listed but not specifically the Australian model. 

> You're going to have to ask Aeotec about this. If the device reports 275V (and I don't see any config parameters for this [here](https://devices.zwave-js.io/?jumpTo=0x0086:0x0003:0x0060:0.0)), this [...]

Thanks again, Happy for you to close this. I will try contact Aeotec and update any further information.

closed 2022-04-21 by sbur83