zwave-js/zwave-js #4485: Philio PSR03-B remote needs compat flags to interview correctly and be usable

goldfoxo, 2022-04-20:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
As far as I can tell, the device is just a little dumb during interview, and poorly implemented by only using Basic CC Set.

What do you observe?
These remotes are 'scene controllers' with two operating modes, one is Simple AV Control, and the second, 'Lighting control', which sends a Basic Set command to devices in each button's respective association group (groups 2-9 for buttons 1-8 respectively). As the former isn't implemented the only useful method is the latter.

I've noticed two issues with the lighting group/basic set mode.

I'm trying to use the remotes as generic remotes that Home Assistant can respond to button presses with.
By adding the controller to each association group 2-9, and setting each 'Lighting Group x: Basic Set Value' config value to a different number, then each button press simply sends a different number Basic CC Set to the controller. This is the only effective way to use these remotes with a controller as far as I can gather.

Issue 1 - when these remotes are interviewed, they fail to report support for Basic CC. In the logs I see eg. -

> 2022-04-20T04:17:03.888Z CNTRLR [Node 013] Interviewing Basic...
> 2022-04-20T04:17:03.888Z CNTRLR » [Node 013] querying Basic CC state...
> ...snip...
> 2022-04-20T04:17:04.966Z CNTRLR [Node 013] Timed out while waiting for a response from the node (ZW0201)
> 2022-04-20T04:17:04.967Z CNTRLR [Node 013] No response to Basic Get command, assuming the node does not suppor
> t Basic CC...

The remotes seem to go to sleep before answering Basic support. They always need two wakes to complete interview.
When this happens, things don't seem to populate through correctly end to end, ie. the number.remote_name_basic entity in Home Assistant never gets populated or updated. In zwavejs2mqtt when I press a button for the first time, a 'Basic v0' value section appears with the correct updates to currentValue, but things still don't flow through to HA.

I accidentally discovered that the remotes do successfully report Basic CC support if I happen to be repeatedly pressing any of the remote buttons during interview, I suppose that keeps the remote awake at just the right moment. Forcing support for Basic in the device compat seems to work around the quirk though.

Issue 2 - is the fact that the device only ever uses Basic CC Set for its functionality. Given these get handled as reports they're hard to handle in HA. Setting treatBasicSetAsEvent: true in compat to trigger events instead allows the device to work easily as a remote.

What did you expect to happen?
Device reports its functions successfully, and can be used as a remote as intended

Steps to reproduce the behavior:
1. Interview remote node
2. Wake up remote with its wake button combo. It starts interviewing but stops and sleeps again. It always requires two wakes to complete interview
3. Basic CC won't be discovered properly, resulting in it not reporting to HA
4. Interview remote node again, waking twice as before but mash normal remote buttons repeatedly throughout.
5. Basic CC will be discovered properly. Button presses can be automated on now by watching the Basic currentValue, but pressing the same button in a row won't be possible. Needs events instead.

### Device information

https://devices.zwave-js.io/?jumpTo=0x013c:0x0008:0x001d:0.0
Manufacturer: Philio
Model name: PSR03-B
Node ID in your network: 13, 14, 15, 16

I believe the PSR03-A will probably need the same workarounds, it's a two button version of the same thing, but don't have one to test. (PSR03-C won't though, it's only a panic button)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
I use `Z-Wave JS to MQTT` community addon (currently 0.38.0 using the below) as driver to the Z-Wave JS HA integration
`zwavejs2mqtt` 6.7.2
`zwave-js` 9.0.2

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I've added compat flags to a device override file to test the fixes and they work well:

```
        "compat": {
                "treatBasicSetAsEvent": true,
                "commandClasses": {
                        "add": {
                                // The device times out when interviewing and doesn't correctly
                                // report it supports Basic CC, so ensure it gets added
                                "0x20": {
                                        "isSupported": true,
                                        "version": 1
                                }
                        }
                }
        },
```

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2022-04-20.log normal interview and a few button presses](https://github.com/zwave-js/node-zwave-js/files/8518891/zwavejs_2022-04-20.log)
[zwavejs_2022-04-20-mashing.log interview while mashing buttons](https://github.com/zwave-js/node-zwave-js/files/8518969/zwavejs_2022-04-20-mashing.log)
[zwavejs_2022-04-20-compat.log after adding compat flags and a few button presses](https://github.com/zwave-js/node-zwave-js/files/8518974/zwavejs_2022-04-20-compat.log)

AlCalzone, 2022-04-20:
Can you test again with just `treatBasicSetAsEvent` set to `true`?
This flag should force-add support for the Basic CC and prevent it from being removed during the interview (like in your log snippet above).

goldfoxo, 2022-04-20:
Gotcha.

Yep, it seems to all work well with only treatBasicSetAsEvent: true

AlCalzone, 2022-04-20:
@zwave-js-bot add compat flag to `0x013c:0x0008:0x001d`
```jsonc
// The device uses Basic Set to indicate pressed buttons in Lighting Control Mode
"treatBasicSetAsEvent": true
```

AlCalzone, 2022-04-20:
Looking at the manual I'm assuming the -A variant works the same way.

@zwave-js-bot add compat flag to `0x013c:0x0008:0x0026`
```jsonc
// The device uses Basic Set to indicate pressed buttons in Lighting Control Mode
"treatBasicSetAsEvent": true
```

PR #4487 (2022-04-20): fix(config): treat Basic Set as event for Philio PSR03-B
fixes: #4485

**TODO:**
- [x]	Change PR title to be more specific
closed 2022-04-20 by AlCalzone