zwave-js/zwave-js #4493: [feat] Fire Alarm Support

XtremeOwnageDotCom, 2022-04-21:
So, I have a few First alert z-wave fire alarms. For the most part, these work with zwavejs2mqtt / home assistant without issues. They report sirens/alerts/battery information.

But, there is no support for interacting with the device at all. Ie, silencing the alarm, testing / triggering the alarm, etc.

However, I recently heard a rumor that if you use a smartthings hub, there are options for silencing the alarm, as well as testing the alarm. 

This is a feature request to investigate/add in the missing commands/functionality to z-wave based fire alarms.

Here is all of the related documentation I can find online regarding this....

https://s7d9.scene7.com/is/content/NewellRubbermaid/DASH/S7_int/Home_and_Sec/2020/BRK_ZCOMBO_Z-Wave_Specification_11.0%5B2%5D.pdf

I personally, cannot see any documentation in the reference to support if this functionality exists or not, however, I am also not at all familiar with z-wave command classes, or z-wave in general.

AlCalzone, 2022-04-21:
> However, I recently heard a rumor that if you use a smartthings hub, there are options for silencing the alarm, as well as testing the alarm.

Wonder which undocumented feature they use for that. The command classes mentioned in your linked manual don't include that functionality:
[image]

It **might** be that they react to Notifications sent to them. But that would require:
1. knowing which ones
2. sending them

The latter is possible, but not straightforward or integrated in HA:
- https://zwave-js.github.io/zwavejs2mqtt/#/guide/mqtt?id=send-command
- https://zwave-js.github.io/node-zwave-js/#/api/CCs/Notification?id=sendreport

AlCalzone, 2022-04-21:
In any case, if this isn't implemented in a standard way, there won't be native support in the driver. I can assist with getting this to work if you find more information though.

closed 2022-04-21 by AlCalzone

XtremeOwnageDotCom, 2022-04-21:
@AlCalzone This is inline with what I was expecting. But, I do appreciate someone more familiar with z-wave taking a look at it!