zwave-js/zwave-js #4502: Unknown RFRegion after Factory Reset of 700 series stick

mundschenk-at, 2022-04-23:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I did a factory reset of my 700 series Z-stick.

What do you observe?
`GetRFRegion` returns `Unknown`

What did you expect to happen?
`GetRFRegion` being set to `EU`.

### Device information

Manufacturer: Aeotec
Model name: Z-Stick 7
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

```
2022-04-23T10:24:43.210Z DRIVER » [REQ] [SerialAPISetup]
                                    command: GetPowerlevel
                                    payload: 0x08
2022-04-23T10:24:43.215Z SERIAL « [ACK]                                                                   (0x06)
2022-04-23T10:24:43.218Z SERIAL « 0x0106010b0820944f                                                   (8 bytes)
2022-04-23T10:24:43.221Z SERIAL » [ACK]                                                                   (0x06)
2022-04-23T10:24:43.224Z DRIVER « [RES] [SerialAPISetup]
                                    command:               GetPowerlevel
                                    normal powerlevel:     3.2 dBm
                                    output power at 0 dBm: 14.8 dBm
2022-04-23T10:24:43.249Z SERIAL » 0x0104000b20d0                                                       (6 bytes)
2022-04-23T10:24:43.250Z DRIVER » [REQ] [SerialAPISetup]
                                    command: GetRFRegion
                                    payload: 0x20
2022-04-23T10:24:43.254Z SERIAL « [ACK]                                                                   (0x06)
2022-04-23T10:24:43.255Z SERIAL « 0x0105010b20fe2e                                                     (7 bytes)
2022-04-23T10:24:43.256Z SERIAL » [ACK]                                                                   (0x06)
2022-04-23T10:24:43.257Z DRIVER « [RES] [SerialAPISetup]
                                    command: GetRFRegion
                                    region:  Unknown
```

mundschenk-at, 2022-04-23:
I've also gotten an error when converting the NVM backup to JSON. I'll send you the file via mail:

```
$ npx @zwave-js/nvmedit nvm2json --in NVM_2022-04-23.bin --out nvm.json
ZWaveError: Object 332800 not found! (ZW0282)
```

AlCalzone, 2022-04-26:
The only thing right now I can imagine that could cause the incorrect powerlevel and region settings, assuming your NVM backup is from after the factory reset, would be that there's a new NVM file format that the stick expects. I can't find any updates to the Z/IP Gateway SDK after 7.17.0 though

AlCalzone, 2022-04-26:
So, I just compared your NVM backup with one I got from a user whose stick is reporting correct RF settings. The format seems to be the same.

mundschenk-at, 2022-04-26:
> The only thing right now I can imagine that could cause the incorrect powerlevel and region settings, assuming your NVM backup is from after the factory reset, would be that there's a new NVM file format that the stick expects. I can't find any updates to the Z/IP Gateway SDK after 7.17.0 though

Yes, the backup was made after the factory reset (but after adding the test node, obviously, so it's not an entirely "fresh" backup). Is the factory reset a function of the stick that the driver just triggers, or is it more involved? Because it looks like that reset might not have happened "all the way"? I didn't pull the stick after resetting, should I have done that?

AlCalzone, 2022-04-27:
AFAIK the factory reset is a one-off API call. It takes a few seconds during which the stick writes factory-default data into the NVM, whereever that comes from. I'm not sure if the NVM gets wiped before that and if the entire NVM gets rewritten or just the protocol part and not the application part where the RF settings are stored.

mundschenk-at, 2022-04-28:
What's the best way to test any of this? Redo a factory reset and see if the RF settings are mis-reported again (followed by a restore)?

AlCalzone, 2022-04-29:
I guess the following could work:
1. Factory Reset
2. NVM Backup
3. Check if the settings are incorrect
4. Try to correct them if they are
5. NVM Backup again

I can then compare what the differences in the NVM are, if there are any.

mundschenk-at, 2022-05-02:
I tried doing that, but the download somehow replaced the file of the same name not sure how that can happen, normally the newer file gets an index number appended. After the reset, `getRFRegion` returned `unknown` and `setRFRegion` failed. When I force-restarted the driver (via Settings/Zwave/Save), `getRFRegion` immediately returned `Europe`. I could not replicate this phenomenon on a second reset (i.e. region was always set to `Europe`).

So it might be that the driver somehow doesn't read the properties correctly after the reset command is issued? Either because the stick needs more time or there simply is some hickup in the driver initialization?

AlCalzone, 2022-05-03:
> Either because the stick needs more time or there simply is some hickup in the ~~driver~~ initialization?

My money is on this. I've been observing different weird behavior when switching from PC Controller to Z-Wave JS, where the controller ID would be 0 until the next stick restart.

darkbasic, 2022-12-11:
I've already heard other people complaining about this and if I recall it correctly it wasn't anything specific about zwave-js, rather about the new newest firmwares which allow you to manually select the rf instead of flashing a separate file.

AlCalzone, 2023-02-16:
I recently fixed a few things in the NVM conversion routines, including a change of the RF config format in firmware 7.18.1.
Going forward, Z-Wave JS makes sure to write the NVM in the highest version it definitely has support for, and lets the firmware itself take care of further upgrades.

So, with https://github.com/zwave-js/node-zwave-js/releases/tag/v10.10.0 this should not be an issue anymore (after restoring the NVM anew). If it still happens, please reopen.

closed 2023-02-16 by AlCalzone