zwave-js/zwave-js #4508: [Bug]: Timeout respons - node communication version 9

prilly-dev, 2022-04-26:


AlCalzone, 2022-04-26:
> it seems the RTT + 1 sec might be a problem when route changes and rtt avrage goes from 30-40ms to 1200ms then it hits timeout.

The initially large delay may be an incorrect test result caused by the target node being a FLiRS device. Those take up to a second to wake up and then stay awake for the remaining communication.

The initial delay should also be part of the RTT, so the lock gets ~2s to respond after that.

Anyways - driver logs of the timed out communication please!

prilly-dev, 2022-04-26:
This was just an example of a timeout, in this particular test we only see one test with high latency, but i practice there are multiple ones, let me retest and upload a screenshot.

All troubleshooting of connectivity issues has been tested, if i downgrade to zwavejs version 8 this is no longer a issue, there are no connection issues.

I will get a logg for you

AlCalzone, 2022-04-26:
> there are no connection issues.

Your health check results beg to differ, but I'll wait for the log.

prilly-dev, 2022-04-26:
I did test with zstick gen5 stick and only one node id lock 150 in a new network, timeour reponses are seen there also and this is happening when the stick is exactly 1 meter away from the lock on a extension cord. I also have RTH-K18 RF SPECTRUM ANALYZATOR, that i have concluded that there are no interference that can explain this behaviour. And again, downgrading to zwavejs version 8 fixes the issue

prilly-dev, 2022-04-26:
There are multiple reasons this healthcheck can produce issues like this, interference is just one of them. I will provide some logs later today

prilly-dev, 2022-04-26:
So you said:
`
The initially large delay may be an incorrect test result caused by the target node being a FLiRS device. Those take up to a second to wake up and then stay awake for the remaining communication`

When route changes and the communication goes through a new node that needs wakeup, what mechanism is making shure the delay is not introduced again? As i initialy wrote this problems with latency is happening when route changes

AlCalzone, 2022-04-26:
Routing only uses mains powered nodes as repeaters, not sleeping or FLiRS nodes.

muddro1, 2022-04-26:
So just re-upped back to 6.7.4 and did a reinterview on my BE469ZP. Same issue I had before, with an incomplete interview. You'll notice it isn't even able to grab the manufacturer, product, product code, and in the past, it also wouldn't pick up the firmware. It is also missing a bunch of the typical values categories, namely the one for user codes:

[image][zwavejs_2022-04-26.log](https://github.com/zwave-js/node-zwave-js/files/8563264/zwavejs_2022-04-26.log)

I've also attached my logs.

AlCalzone, 2022-04-26:
@muddro1 your lock doesn't respond at all to the requests during the interview. It tells the driver which commands are supported securely, then there's radio silence aside from the protocol-level acknowledgements (which are received almost immediately btw).

Right at the very end, it responds once to a query that was sent 11 seconds earlier - but not the ones that were sent prior.

muddro1, 2022-04-26:
> @muddro1 your lock doesn't respond at all to the requests during the interview. It tells the driver which commands are supported securely, then there's radio silence aside from the protocol-level [...]

Didn't seem to have this problem before and not sure why its arising. The lock is less than 5 feet from the controller, and the controller is on a USB. Any troubleshooting I can do?

AlCalzone, 2022-04-26:
> Any troubleshooting I can do?

I'm inclined to say this is on the lock side and not the controller side.
- Switch batteries
- Exclude, factory reset, re-include the lock

prilly-dev, 2022-04-26:
> > Any troubleshooting I can do?
> 
> I'm inclined to say this is on the lock side and not the controller side.
> 
>     * Switch batteries
> 
>     * Exclude, factory reset, re-include the lock

I have done this multiple times, its weird we see excatly the same issues on atlest 3 different locktypes with both s0 and s2, where s0 seems to be the worst

muddro1, 2022-04-26:
> I have done this multiple times, its weird we see excatly the same issues on atlest 3 different locktypes with both s0 and s2, where s0 seems to be the worst

Got it working again, but had to replace batteries. Ways I got it working before was also unplugging batteries and replugging them. Odd that its only 4 feet away and the batteries reported @ 97% and would need to be changed.

AlCalzone, 2022-04-26:
> I have done this multiple times, its weird we see excatly the same issues on atlest 3 different locktypes with both s0 and s2, where s0 seems to be the worst

I went by the log I got from @muddro1. Your issue may be different.

prilly-dev, 2022-04-26:
so i have done some logging in silly mode, i have done a interview of the node that is the id lock 150 and refreshed usercode v1 class.

what i noticed:

protocol & route speed: Z-Wave, 40 kbit/s, why is it 40kbit, i thought it was supposed to be 100kbit?
ACK RSSI:               Receiver saturated, seems there is something saturated?

it also seems that the latency is consistent <200ms, that is good. but when the timeout occures reciver is complaining its saturated?

[zwavejs_2022-04-26.log](https://github.com/zwave-js/node-zwave-js/files/8564871/zwavejs_2022-04-26.log)

AlCalzone, 2022-04-27:
@prilly-dev
The communication looks okay-ish. Round-trip times of 100..150 ms at 40 kbps, which does seem normal. However, I'm seeing this over 200 times in your log:
```
ACK RSSI:               Receiver saturated
```
which means the RSSI at the controller was too high to measure precisely. Not sure if this is a problem, you're the first user where I saw this at all. Doesn't sound good though.

---

Then there's a longer stretch of timeouts, but this is due to a different issue:
Z-Wave JS queries the Configuration CC without security encapsulation, because the lock didn't report the CC to require security. These are the ones that don't:
```
2022-04-26T15:49:35.883Z CNTRLR « [Node 025] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Manufacturer Specific
                                  · Security
                                  · Device Reset Locally
                                  · Powerlevel
                                  · Configuration
                                  controlled CCs:
```
and these are the ones that do:
```
2022-04-26T15:49:36.551Z CNTRLR « [Node 025] received secure commands (S0)
                                  supported CCs:
                                  · Version
                                  · Door Lock
                                  · User Code
                                  · Association
                                  · Association Group Information
                                  · Notification
                                  · Firmware Update Meta Data
                                  · Battery
                                  controlled CCs:
```
However, the lock responds with security-encapsulated configuration reports, which are not considered to be responses to the un-encapsulated requests:
```
2022-04-26T15:50:00.322Z DRIVER » [Node 025] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      101
                                  └─[ConfigurationCCGet]
                                      parameter #: 4
...
2022-04-26T15:50:01.215Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[ConfigurationCCReport]
                                        parameter #: 4
                                        value size:  1
                                        value:       2
```
IMO this is a **bug in the lock firmware**.

The next stretch of timeouts happen when the user codes are queried. At first the lock claims that it supports 109 codes:
```
2022-04-26T15:49:56.844Z DRIVER « [Node 025] [REQ] [ApplicationCommand]
                                  └─[SecurityCCCommandEncapsulation]
                                    │ sequenced: false
                                    └─[UserCodeCCUsersNumberReport]
                                        supported users: 109
```
but later when Z-Wave JS queries all codes, it only responds twice:
* Request for code 1 returns code 1
* Request for code 2 also returns code 1 (!)
* Requests for codes 3..109 time out (no response)

In the meantime, the lock does promptly respond to the Nonce requests and there are no dropped commands, so this doesn't seem to be a physical communication issue.

I'm also leaning towards calling this a **bug in the lock firmware**.

prilly-dev, 2022-04-27:
I did found at what this saturated is, if you look closely it specifies node 5 as repeater node. Communication goes from node 25 (lock) -> node 5 (fibaro dimmer 2) - node 1 (zstick gen 5). Node 5 is less then 20 cm from the lock, its built inn the wall right by the door. So its the reciver at node 5 that reports saturated not the controller. All messages sent from the lock -> controller seems to be repeated by node 5. What i found out is that saturated reciver is of no problem ref: https://drzwave.blog/category/z-wave-mesh/

What i did notice is that there are expired nonce when refreshing usercode v1 from the lock, parameters is default not set in the lock either if usercodes are not enabled and set from the user side. It seems it claims that values are not passed from the lock it might be the lock is not passing them because its actualy not set and there are no values.

AlCalzone, 2022-04-27:
> it might be the lock is not passing them because its actualy not set and there are no values

Doesn't matter, it still has to respond:
[image]

and tell us that the code isn't occupied using the User ID status:
[image]

AlCalzone, 2022-04-27:
FWIW, this isn't true:
> its the reciver at node 5 that reports saturated not the controller.

The repeater ACK is a separate field in the transmit report and it is N/A in your case. Not all nodes support reporting it.
The field in your reports is definitely at the controller.

prilly-dev, 2022-04-27:
> FWIW, this isn't true:
> 
> > its the reciver at node 5 that reports saturated not the controller.
> 
> The repeater ACK is a separate field in the transmit report and it is N/A in your case. Not all nodes support reporting it. The field in your reports is definitely at the controller. Here's an example:
> 
> ```
> 2022-04-26T15:47:22.421Z DRIVER « [REQ] [SendData]
>                                     callback id:            3
>                                     transmit status:        OK, took 80 ms
>                                     repeater node IDs:      5, 9
>                                     routing attempts:       1
>                                     protocol & route speed: Z-Wave, 40 kbit/s
>                                     ACK RSSI:               Receiver saturated
>                                     ACK RSSI on repeaters:  N/A, N/A
>                                     ACK channel no.:        1
>                                     TX channel no.:         1
> ```

Yes you probably have right there, but reciver saturated is of no problem.

prilly-dev, 2022-04-27:
it might seems you have right regarding buggy firmware, this id lock is in general a buggy lock. let me see what the maker of thr lock has to say.

as i told you i habe more locks this is a issue with, i have a danalock v3 btzw that also has issues with usercode class, i will provide you with logs from this unit later today.

AlCalzone, 2022-04-27:
Hope you don't also have a Schlage lock, as these seem to be nightmares too.

prilly-dev, 2022-04-27:
[image]

it indeed seems to be undefines values in usercode for the idlock, i will try to sett all values to 2 to see if this might change anything, i have reported this issues to idlock, generic respons from them, i will wait and see what they come up with.

closed 2022-05-15 by zwave-js-assistant[bot]