zwave-js/zwave-js #4517: Fibaro RGBW2 Device Colour not Updating, however controller is reporting the new Colour

amilanov75, 2022-04-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I have had a somewhat challenging day today trying to figure out what's going on whilst testing changing the colour of Fibaro RGBW 2 modules attached to LED strips.

I have tested this issue with 3x separate Fibaro RGBW 2 devices. 
These devices did not appear to be showing this issue when installed on another controller only a week or so ago. 

The issue is that after as little as 1x colour change request, the Fibaro RGBW2 device will not actually change colour, however the controller returns a message to say the colour has actually changed and the new colour is reported in the NR UI.

**Info for analysis**

The following information relates to sending a command to change the colour to light blue (R,G,B = 0,247,255)

1) I have attached an image of the outbound message from NR when changing to light blue (image NR outbound) - proof for me that the command was sent

2) Also attached is an image of the message coming back from the controller showing the value as light blue (image controller output) - evidence that the controller thinks, the colour has changed

3) Plus an image of the NR UI reporting light blue (image NR UI) - evidence that the NR UI also thinks the colour has changed.

4) Finally the log file (zwave-js.log) 
Please start looking from "2022-04-27T19:54:35.103Z NDERED" 
The entries from before are from another capture. 

I'm looking at the logs for the first time myself and think the commandClass 51 to change to light blue can be found here:

2022-04-27T19:54:47.171Z NDERED « [Node: 11] [API: ValueAPI] [Method: setValue]
                                  └─[params]
                                      0: {"commandClass":51,"property":"targetColor","endpoint":1}
                                      1: {"warmWhite":0,"red":0,"green":247,"blue":255}
                                      2: null

I have to admit, I'm not sure what to look for after that.

It's important to say that you may see a Polling request in the log, as I implemented polling 2 seconds after EVERY command today. This is to temporary workaround to this issue: https://github.com/zwave-js/node-zwave-js/issues/4497, until it is resolved. 

### Device information

Fibaro RGBW2 Modules

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [X] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

NR 
module version 6.5.3
Driver version 8.11.2
DB version 8.11.2

### Did you change anything?

no

### If yes, what did you change?

n/a

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Worked on a HC2 controller

### Attach Driver Logfile

[image]
[image]

[image]

[zwave-js.log](https://github.com/zwave-js/node-zwave-js/files/8577515/zwave-js.log)

amilanov75, 2022-04-27:
FYI, I just tested on another part of the network and the same issue is happening for other RGBW2 modules.

AlCalzone, 2022-04-28:
The way I see it, these are two different things happening:
1. The driver is not verifying the color change after setting it, because that would require one query per color channel, which could quickly result in tons of traffic. So all it can do is trust that the change worked if communication had no error (like it currently does), or [use Supervision](https://github.com/zwave-js/node-zwave-js/issues/3517) (but I'm not sure if the RGBW supports that).
2. The device isn't reporting the change, but it is also not sending any message indicating that it failed to set the new color.

> The issue is that after as little as 1x colour change request, the Fibaro RGBW2 device will not actually change colour,

Honestly, I'd contact Fibaro about that.

amilanov75, 2022-04-28:
Hi @Alclazone

Further to the info I sent you today via PM, and thanks to two amazing friends who have help me immensely over the past 48 hours, after testing the same module (Fibaro RGBW2) on 2x 500 series controllers:
1) zwavejs via nr-zwavejs
2) openHAB

The colour syncing issue does not happen when using openHAB.

Attached are the zniffer logs from both openHAB and zwavejs. The commandClass being used is the same = 0x33 i.e. 51

Please refer to line 2 of the openHab log and line 1 of the zniffer log to compare side-by-side.

As per my PM, I have tested a RGBW2 module (earlier today) on a HC2 (300 series) and it works. 
Then I migrated the RGBW2 module back to zwavejs and it doesn’t work again. 

The issues in zwavejs is happening on all devices that I have tested (at least 5 now). 

Can you please look into this as a matter of urgency.

AlCalzone, 2022-04-29:
I took a look at your Zniffer logs. There are three differences between OpenHAB and Z-Wave JS I could spot:
1. OH uses 1 as the source endpoint for multi channel encapsulation, ZJS uses 0. This shouldn't be a problem
2. OH sends a Multilevel Switch Set before each command and verifies each color component change. Z-Wave JS doesn't. This means OH sends 1+1+4=6 messages for each color change and receives 4, Z-Wave JS sends 1 and receives 0.
3. OH uses the default transition duration for the color change commands, the commands from Z-Wave JS are sent with a duration of **69 seconds**!

Not sure where that is configured in node-red (@marcus-j-davies ?), but I'm pretty sure the latter is your problem.

marcus-j-davies, 2022-04-29:
> Z-Wave JS are sent with a duration of 69 seconds!

This will be provided by the Users Message.

Params[0]: The Value ID
Params[1]: The Value
Params[2]: SetValueAPIOptions (if provided)

```javascript
 case 'setValue':
   if (Params.length > 2) {
      await ZWaveNode.setValue(Params[0], Params[1], Params[2]);
    } else {
       await ZWaveNode.setValue(Params[0], Params[1]);
  }
  break;
```

https://github.com/zwave-js/node-red-contrib-zwave-js/wiki/Value-API#setvalue
https://github.com/zwave-js/node-red-contrib-zwave-js/wiki/Value-API#setvalue-with-options

AlCalzone, 2022-04-29:
Interestingly, the driver logs posted above use a transition duration of 0s, so this was somehow changed between tests

closed 2022-04-29 by AlCalzone

amilanov75, 2022-04-29:
@AlCalzone 

It looks like our paths collided.

My friend came to the same conclusion last night and we were in the process of testing it this morning.

Good news is, after a long chat about how great NR is, we tested (with a 1s transition) it and it works!

AlCalzone, 2022-04-29:
Nice. Fix is released in v9.0.7 FYI

amilanov75, 2022-04-29:
Cool, what's the lead time to 9.0.7 being released?

AlCalzone, 2022-04-29:
Roughly minus 43 minutes ;)
[image]