zwave-js/zwave-js #4520: Battery powered devices is reported as node neighbors for powered devices

jmgiaever, 2022-04-28:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

**What causes the bug?**
Not sure.

**What do you observe?**
Battery powered devices (sleeping nodes) is reported as neighbors for powered devices during a network heal. For what I can see all the listed devices is also sleeping on the time when they get listed as a neighbor.

Example of the first healing node in the network:
```
2022-04-27T23:09:44.023Z CNTRLR   [Node 002] healing node...
2022-04-27T23:09:44.023Z CNTRLR » [Node 002] refreshing neighbor list (attempt 1)...
2022-04-27T23:09:44.031Z SERIAL » 0x01050048024efe                                                     (7 bytes)
2022-04-27T23:09:44.032Z DRIVER » [Node 002] [REQ] [RequestNodeNeighborUpdate]
                                    callback id: 78
2022-04-27T23:09:44.035Z SERIAL « [ACK]                                                                   (0x06)
2022-04-27T23:09:44.243Z SERIAL « 0x010500484e21dd                                                     (7 bytes)
2022-04-27T23:09:44.245Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:44.248Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   78
                                    update status: UpdateStarted
2022-04-27T23:09:44.728Z SERIAL « 0x010500484e21dd                                                     (7 bytes)
2022-04-27T23:09:44.731Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:44.733Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   78
                                    update status: UpdateStarted
2022-04-27T23:09:46.237Z SERIAL « 0x010500484e21dd                                                     (7 bytes)
2022-04-27T23:09:46.239Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:46.241Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   78
                                    update status: UpdateStarted
2022-04-27T23:09:49.700Z SERIAL « 0x010a00040021029840be00b4                                          (12 bytes)
2022-04-27T23:09:49.704Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:49.706Z DRIVER « [Node 033] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceGet]
2022-04-27T23:09:54.222Z SERIAL « 0x010500484e22de                                                     (7 bytes)
2022-04-27T23:09:54.226Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:54.229Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
                                    callback id:   78
                                    update status: UpdateDone
2022-04-27T23:09:54.242Z CNTRLR « [Node 002] neighbor list refreshed...
2022-04-27T23:09:54.242Z CNTRLR » [Node 002] Deleting SUC return route...
2022-04-27T23:09:54.252Z SERIAL » 0x01110013210a9880e6d3587ddd41212a054f03                            (19 bytes)
2022-04-27T23:09:54.253Z DRIVER » [Node 033] [REQ] [SendData]
                                  │ transmit options: 0x05
                                  │ callback id:      79
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xe6d3587ddd41212a
2022-04-27T23:09:54.255Z SERIAL « [ACK]                                                                   (0x06)
2022-04-27T23:09:54.260Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-04-27T23:09:54.261Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:54.262Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-04-27T23:09:54.281Z SERIAL « 0x011800134f00000200be7f7f7f7f000003000000000301000006              (26 bytes)
2022-04-27T23:09:54.282Z SERIAL » [ACK]                                                                   (0x06)
2022-04-27T23:09:54.284Z DRIVER « [REQ] [SendData]
[... 111 lines cut ...]
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

`zwavejs2mqtt` version: 6.7.4.e6d17ad
`zwave-js` version: 9.0.4

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### If yes, where did it work?

Can remember seeing this before, on version before v9.

### Attach Driver Logfile

file:///home/joachim/zwavejs_current.log

jmgiaever, 2022-04-28:
Fixed. 

It doesn't seem to be a consistency to the «sleeping nodes» reported as neighbors. Did a reboot and a fresh network heal now, and node 2 now reports (even more) sleeping nodes:

```
2022-04-28T12:42:56.989Z CNTRLR « [Node 002] node neighbors received: 1, 3, 4, 6, 7, 8, 9, 10, 12, 13, 16, 17, 2
                                  0, 22, 30, 32, 33, 37, 44, 45, 47, 78, 81, 82, 85, 90, 96, 97, 99, 104, 107, 1
                                  08, 109, 111, 112, 118, 119, 122, 125, 126, 127, 132, 134, 182, 183, 185, 188,
                                   189, 192, 193, 194, 195, 196, 197, 199, 201, 202
2022-04-28T12:42:56.989Z CNTRLR   [Node 003] added to healing queue for listening nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 008] added to healing queue for listening nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 020] added to healing queue for listening nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 022] added to healing queue for listening nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 078] added to healing queue for sleeping nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 081] added to healing queue for sleeping nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 090] added to healing queue for sleeping nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 104] added to healing queue for sleeping nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 132] added to healing queue for sleeping nodes
2022-04-28T12:42:56.990Z CNTRLR   [Node 134] added to healing queue for sleeping nodes
2022-04-28T12:42:56.991Z CNTRLR   [Node 182] added to healing queue for listening nodes
2022-04-28T12:42:56.991Z CNTRLR   [Node 202] added to healing queue for sleeping nodes
```
[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8582791/zwavejs_current.log)

jmgiaever, 2022-04-28:
Yes, they do all exists.

[image]

jmgiaever, 2022-04-29:
[image]

[image]

[image]
@AlCalzone

AlCalzone, 2022-05-05:
As discussed in Slack, I don't think this is actually a bug. Battery-powered nodes can be neighbors of mains-powered nodes. Unless the controller is reporting false info, but even then we can't do much about it.

closed 2022-05-05 by AlCalzone