zwave-js/zwave-js #4537: Binary Switch (PS9EP) Current value always returns false

shuimulinfeng, 2022-05-01:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

PS9EP (Sigma Designs (Former Zensys))

What do you observe?

current value always returns false no matter the light turns on or off.

What did you expect to happen?

"Current value" same with "Target value"

Steps to reproduce the behavior:
click the 'ON' button of ‘Target value’, the light turns on, 'Current value' changes to 'true' , but the current value changes to 'false' immediately after that.

### Device information

Manufacturer: Sigma Designs (Former Zensys)
Model name: PS9EP 
Node ID in your network: 63

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 9.0.2
`zwavejs2mqtt` version: 6.7.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

```
2022-05-01T00:40:14.549Z SERIAL » 0x010a00133f032501ff253c18                                          (12 bytes)
2022-05-01T00:40:14.550Z DRIVER » [Node 063] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      60
                                  └─[BinarySwitchCCSet]
                                      target value: true
2022-05-01T00:40:14.551Z SERIAL « [ACK]                                                                   (0x06)
2022-05-01T00:40:14.555Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-01T00:40:14.556Z SERIAL » [ACK]                                                                   (0x06)
2022-05-01T00:40:14.557Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-01T00:40:14.711Z SERIAL « 0x010700133c000010c7                                                 (9 bytes)
2022-05-01T00:40:14.711Z SERIAL » [ACK]                                                                   (0x06)
2022-05-01T00:40:14.712Z DRIVER « [REQ] [SendData]
                                    callback id:     60
                                    transmit status: OK
2022-05-01T00:40:14.716Z CNTRLR   [Node 063] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2022-05-01T00:40:15.724Z SERIAL » 0x010900133f022502253de7                                            (11 bytes)
2022-05-01T00:40:15.724Z DRIVER » [Node 063] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      61
                                  └─[BinarySwitchCCGet]
2022-05-01T00:40:15.725Z SERIAL « [ACK]                                                                   (0x06)
2022-05-01T00:40:15.729Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-01T00:40:15.730Z SERIAL » [ACK]                                                                   (0x06)
2022-05-01T00:40:15.730Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-01T00:40:15.878Z SERIAL « 0x010700133d00000fd9                                                 (9 bytes)
2022-05-01T00:40:15.878Z SERIAL » [ACK]                                                                   (0x06)
2022-05-01T00:40:15.879Z DRIVER « [REQ] [SendData]
                                    callback id:     61
                                    transmit status: OK
2022-05-01T00:40:16.007Z SERIAL « 0x01090004003f03250300e8                                            (11 bytes)
2022-05-01T00:40:16.008Z CNTRLR   [Node 063] [~] [Binary Switch] currentValue: true => false        [Endpoint 0]
2022-05-01T00:40:16.011Z SERIAL » [ACK]                                                                   (0x06)
2022-05-01T00:40:16.012Z DRIVER « [Node 063] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
```

Log file:
[zwavejs_2022-05-01.log](https://github.com/zwave-js/node-zwave-js/files/8598154/zwavejs_2022-05-01.log)
[node63_debug_info.txt](https://github.com/zwave-js/node-zwave-js/files/8598156/node63_debug_info.txt)

AlCalzone, 2022-05-02:
The device reports that it is off, one second after turning it on:
```
2022-05-01T00:40:16.012Z DRIVER « [Node 063] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
```

If you press the refresh button on the Binary Switch panel, does it give you the correct state?

shuimulinfeng, 2022-05-02:
Hi, the state is still wrong after pressing the refresh button. btw, I'm using Aeotec Z-Stick Gen5 as a secondary controller, and if I pressed the switch button, the state wasn't updated to the zwavejs2mqtt control panel.

AlCalzone, 2022-05-02:
> the state is still wrong after pressing the refresh button

Does the device have multiple endpoints/channels?

If not, I fear there's nothing we can do then, since the device doesn't report the correct status on its own or when asked.

shuimulinfeng, 2022-05-02:
Node 63 doesn't have multiple channels, but another device has multiple channels which is identified as PS9EP too,  'Current value' on the control panel is same with 'Target value'.
```
2022-05-02T13:34:38.220Z SERIAL » 0x010e00133707600d00022501ff2598db                                  (16 bytes)
2022-05-02T13:34:38.221Z DRIVER » [Node 055] [REQ] [SendData]
                                  │  transmit options: 0x25
                                  │  callback id:      152
                                  └ ─ [MultiChannelCCCommandEncapsulation]
                                    │  source:      0
                                    │  destination: 2
                                    └ ─ [BinarySwitchCCSet]
                                        target value: true
2022-05-02T13:34:38.221Z SERIAL « [ACK]                                                                   (0x06)
2022-05-02T13:34:38.226Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-02T13:34:38.228Z SERIAL » [ACK]                                                                   (0x06)
2022-05-02T13:34:38.228Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-02T13:34:38.305Z SERIAL « 0x01070013980000087b                                                 (9 bytes)
2022-05-02T13:34:38.305Z SERIAL » [ACK]                                                                   (0x06)
2022-05-02T13:34:38.306Z DRIVER « [REQ] [SendData]
                                    callback id:     152
                                    transmit status: OK
2022-05-02T13:34:38.309Z CNTRLR   [Node 055] [~] [Binary Switch] currentValue: false => true        [Endpoint 2]
2022-05-02T13:34:39.315Z SERIAL » 0x010d00133706600d00022502259924                                    (15 bytes)
2022-05-02T13:34:39.316Z DRIVER » [Node 055] [REQ] [SendData]
                                  │  transmit options: 0x25
                                  │  callback id:      153
                                  └ ─ [MultiChannelCCCommandEncapsulation]
                                    │  source:      0
                                    │  destination: 2
                                    └ ─ [BinarySwitchCCGet]
2022-05-02T13:34:39.317Z SERIAL « [ACK]                                                                   (0x06)
2022-05-02T13:34:39.322Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-02T13:34:39.323Z SERIAL » [ACK]                                                                   (0x06)
2022-05-02T13:34:39.324Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-02T13:34:39.429Z SERIAL « 0x010700139900000b79                                                 (9 bytes)
2022-05-02T13:34:39.430Z SERIAL » [ACK]                                                                   (0x06)
2022-05-02T13:34:39.430Z DRIVER « [REQ] [SendData]
                                    callback id:     153
                                    transmit status: OK
2022-05-02T13:34:39.455Z SERIAL « 0x010d0004003707600d02002503018e                                    (15 bytes)
2022-05-02T13:34:39.456Z SERIAL » [ACK]                                                                   (0x06)
2022-05-02T13:34:39.456Z DRIVER « [Node 055] [REQ] [ApplicationCommand]
                                  └ ─ [MultiChannelCCCommandEncapsulation]
                                    │  source:      2
                                    │  destination: 0
                                    └ ─ [BinarySwitchCCReport]
                                        current value: undefined
```
Node 55:
[image]

Node 63:
[image]

AlCalzone, 2022-05-02:
This is really weird. Both have the same ID and firmware, so they shouldn't have different capabilities.

Can you re-interview both (one after the other) and post a log of that? (Attach as a file please, or this will be too long).

shuimulinfeng, 2022-05-03:
Hi, please view the logs:
[node55_interview.txt](https://github.com/zwave-js/node-zwave-js/files/8607779/node55_interview.txt)
[node63_interview.txt](https://github.com/zwave-js/node-zwave-js/files/8607780/node63_interview.txt)

AlCalzone, 2022-05-03:
So, the interviews looked fine, but you really seem to have two different devices, despite them identifying exactly the same. Node 63 has a single endpoint and allows 5 nodes in the association groups, Node 55 has two endpoints and allows 3 nodes 🤷‍♂️.

This causes the lifeline association to be set up differently. Node 63 (not reporting?) has a node association (target node 64, target endpoint None), while Node 55 has a multi channel association (target node 64, target endpoint 0).
Try changing the association on Node 63 to be like the one on Node 55. If it works then, we can change the config file to treat both variants the same.

shuimulinfeng, 2022-05-03:
Hi, I'm not familiar with changing the association, then I tried tp create new association on control panel:
[image]

but the target endpoint is still to be 'none':
[image]

btw the Node 64 is "Z‐Stick Gen5 USB Controller", and the below is the 'Groups' of Node 53:
[image]

AlCalzone, 2022-05-03:
Ok that's weird. To be sure I don't misunderstand: Node 53 works correctly?

shuimulinfeng, 2022-05-03:
Not sure which node do you mean, both node 55 and node 63 can control the lights, but node 63 returns a bad state of 'Current Value' if 'Target Value' changes to 'true'.

AlCalzone, 2022-05-04:
Right, I meant 55.

AlCalzone, 2022-05-04:
I have a potential fix I'd like you to test before I release it.

Here's a description: https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files
And here's the file to test: https://raw.githubusercontent.com/zwave-js/node-zwave-js/4cbcf165be9eacafd3a7a2edd5656f2521830266/packages/config/config/devices/0x0000/ps9ep.json

Don't forget to re-interview both nodes so we can check if this works as intended.

shuimulinfeng, 2022-05-04:
Sadly, it didn't work...

```
2022-05-04T07:41:10.139Z SERIAL » 0x010a00133f032501ff25caee                                          (12 bytes)
2022-05-04T07:41:10.140Z DRIVER » [Node 063] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      202
                                  └─[BinarySwitchCCSet]
                                      target value: true
2022-05-04T07:41:10.141Z SERIAL « [ACK]                                                                   (0x06)
2022-05-04T07:41:10.145Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-04T07:41:10.146Z SERIAL » [ACK]                                                                   (0x06)
2022-05-04T07:41:10.147Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-04T07:41:10.301Z SERIAL « 0x01070013ca00001031                                                 (9 bytes)
2022-05-04T07:41:10.301Z SERIAL » [ACK]                                                                   (0x06)
2022-05-04T07:41:10.302Z DRIVER « [REQ] [SendData]
                                    callback id:     202
                                    transmit status: OK
2022-05-04T07:41:10.306Z CNTRLR   [Node 063] [~] [Binary Switch] currentValue: false => true        [Endpoint 0]
2022-05-04T07:41:11.315Z SERIAL » 0x010900133f02250225cb11                                            (11 bytes)
2022-05-04T07:41:11.316Z DRIVER » [Node 063] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      203
                                  └─[BinarySwitchCCGet]
2022-05-04T07:41:11.317Z SERIAL « [ACK]                                                                   (0x06)
2022-05-04T07:41:11.321Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-05-04T07:41:11.322Z SERIAL » [ACK]                                                                   (0x06)
2022-05-04T07:41:11.322Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-05-04T07:41:11.470Z SERIAL « 0x01070013cb00000f2f                                                 (9 bytes)
2022-05-04T07:41:11.470Z SERIAL » [ACK]                                                                   (0x06)
2022-05-04T07:41:11.471Z DRIVER « [REQ] [SendData]
                                    callback id:     203
                                    transmit status: OK
2022-05-04T07:41:11.595Z SERIAL « 0x01090004003f03250300e8                                            (11 bytes)
2022-05-04T07:41:11.596Z CNTRLR   [Node 063] [~] [Binary Switch] currentValue: true => false        [Endpoint 0]
2022-05-04T07:41:11.597Z SERIAL » [ACK]                                                                   (0x06)
2022-05-04T07:41:11.598Z DRIVER « [Node 063] [REQ] [ApplicationCommand]
                                  └─[BinarySwitchCCReport]
                                      current value: false
```

and re-interview logs: [node63_interview2.txt](https://github.com/zwave-js/node-zwave-js/files/8618613/node63_interview2.txt)
 
groups info:
[image]

AlCalzone, 2022-05-04:
Ok, next try with this variant of the file please: https://raw.githubusercontent.com/zwave-js/node-zwave-js/fbca7d10431288d81051e4c90f3d93e64ca872e3/packages/config/config/devices/0x0000/ps9ep.json

Although I'm not sure if this is going to work - the device didn't accept the multi channel association (target endpoint 0) which node 55 took without problems.

shuimulinfeng, 2022-05-04:
Amazing! it works, thanks a lot for your time.

Below is the newest result:
[image]

AlCalzone, 2022-05-04:
perfect, I'd almost lost hope.

AlCalzone, 2022-05-04:
Next update will include the updated config file - then you can delete your copy.

closed 2022-05-04 by AlCalzone

shuimulinfeng, 2022-05-04:
Got it, thanks again.