zwave-js/zwave-js #4548: Missing device configuration: CLEVERIO Smart Smoke Alarm

Aquilo77, 2022-05-01:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

CLEVERIO Smart Smoke Alarm

### Manufacturer ID

0x041C

### Product Type

0x8002

### Product ID

0x1000

### Firmware Version

3.01

### Is the device listed on the Z-Wave Alliance website?

https://products.z-wavealliance.org/products/3365

### Do you have a manual?

[51114 user manual.pdf](https://github.com/zwave-js/node-zwave-js/files/8599668/51114.user.manual.pdf)

AlCalzone, 2022-05-02:
@zwave-js-bot import config 3365 from zwa

AlCalzone, 2022-05-02:
We do have a config file for this:
https://devices.zwave-js.io/?jumpTo=0x041c:0x8002:0x1000:0.0

closed 2022-05-02 by AlCalzone