zwave-js/zwave-js #4566: [fw 7.17.2] Cannot heal any node

darkbasic, 2022-05-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

Fw 7.17.1/7.17.2 seemed to fix my network issues for a while, but after replacing some dead nodes (and all the battery powered ones) I've done a full heal and the same old pre 7.17.1 issues are back once again.
One of the nodes has been marked as dead right after being successfully healed and now I cannot heal any node period.

This is what I got in the zwavejs2mqtt logs:
```
2022-05-04 11:49:49.630 INFO ZWAVE: Calling api healNode with args: [ 84, [length]: 1 ]
2022-05-04 11:49:49.639 INFO ZWAVE: Success zwave api call healNode false
```

but **nothing** is shown in the zwavejs ones, despite having debug logs enabled.

I'll attach zwavejs2mqtt and zwavejs debug logs of yesterday (when I started removing/adding some nodes) and today (when the "cannot heal" issue appeared).

Linked issue: https://github.com/zwave-js/node-zwave-js/issues/3906

### Device information

Manufacturer: Aeotec
Model name: Z-Pi 7

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.7.1.2a08abd
zwave-js: 9.0.2
home id: 3578308563
home hex: 0xd548a7d3

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

fw 7.17.1/7.17.2 used to fix it for a while, but today I've replaced some dead nodes, removed the battery powered ones and did a network heal. Same old pre-7.17.1 issues happen again now.

### Attach Driver Logfile

[zwavejs2mqtt_2022-05-04.log](https://github.com/zwave-js/node-zwave-js/files/8619402/zwavejs2mqtt_2022-05-04.log)
[zwavejs2mqtt_2022-05-03.log.gz](https://github.com/zwave-js/node-zwave-js/files/8619404/zwavejs2mqtt_2022-05-03.log.gz)
[zwavejs_2022-05-04.log](https://github.com/zwave-js/node-zwave-js/files/8619405/zwavejs_2022-05-04.log)
[zwavejs_2022-05-03.log.gz](https://github.com/zwave-js/node-zwave-js/files/8619406/zwavejs_2022-05-03.log.gz)

blhoward2, 2022-05-04:
Your title says can't heal any nodes, but as you describe it appears they all healed except the last one, correct? Also, please confirm that your stick is plugged into a usb extension cable.

darkbasic, 2022-05-04:
> Your title says can't heal any nodes, but as you describe it appears they all healed except the last one, correct?

Nope, they **all** healed including the last one, but after doing so one node has been marked as dead and I has not been able to heal any node.

I've debugged this with Al Calzone on Slack and it appears like this might be due to the controller being stuck in a never ending heal, possibly because I've tried to heal a node marked as dead without pinging it first. I'm testing a PR with additional debugging logic right now.

darkbasic, 2022-05-05:
I still didn't manage to reproduce the original issue w/ https://github.com/zwave-js/node-zwave-js/pull/4567
I'll keep an eye on this and let you know.

AlCalzone, 2024-10-08:
There have been quite a few changes, both in terms of Z-Wave SDK/Firmware, and handling of route rebuilding (formerly called healing).
If this is still an issue, please open a new one with updated information.

closed 2024-10-08 by AlCalzone