zwave-js/zwave-js #4575: [Bug]: Device disappears, auto-removed after 1 missed ping

karalabe, 2022-05-05:


AlCalzone, 2022-05-05:
FWIW, the removal happens **10 seconds** after the node is dead.

Z-Wave JS definitely doesn't remove devices on its own, except when the device itself sent a `Device Reset Locally Notification`, which means the device has been factory reset by physical interaction and is de facto no longer a part of the network.

Unfortunately, this is impossible to diagnose without driver logs, since the application logs don't include any Z-Wave traffic.

karalabe, 2022-05-05:
Well, this was at 5AM in the morning, so pretty much nothing happened then, nobody touched it, nor did any autmations run that could have triggered it. Is there a way to enable/collect driver logs so maybe we can have it available next time?

AlCalzone, 2022-05-05:
zwavejs2mqtt lets you write [driver logfiles](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs) which are rotated every 7 days by default.

AlCalzone, 2022-05-05:
Alright, I'm going to close here then. When you have something new, feel free to reopen.

closed 2022-05-05 by AlCalzone

karalabe, 2022-05-05:
I still think there's a fault somewhere which caused an error to be escalated into exclusion, since everything was idle. But yeah, not much more information to say.