zwave-js/zwave-js #4582: Devices with "unknown" information after migrating from 500 series to 700 series controller

rcdailey, 2022-05-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I'm using zwavejs2mqtt 0.39.0 in Home Assistant. I migrated from my Aeotec Gen5+ USB controller to my SiLabs 700 series usb controller. I exported the NVM from my 500 series and imported it to my 700 series stick. In addition to this, I had to also set the Region to US via Simplicity Studio because for whatever that was blank, resulting in a bunch of dead nodes.

After this change, nearly all of my zwave devices are registered properly in my zwave dashboard. See the attachment below:
[image]

However, there are two nodes (24 and 25) that show up as "Unknown" but not really dead. My attempt at doing a reinterview seems to succeed but the data does not get populated. Logs are attached as well. In the logs, I did a reinterview for node 24.

What is the issue with these two nodes and how may I fix them?

### Device information

Manufacturer: SiLabs
Model name: SLUSB001A
Node ID in your network: 1

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.39
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-05-05.log](https://github.com/zwave-js/node-zwave-js/files/8635990/zwavejs_2022-05-05.log)

AlCalzone, 2022-05-06:
There are some weird things going on here. Can you send me your cache files `<homeid>.json` and/or `<homeid>.jsonl` as well as NVM backups from the 500 and 700 series via e-mail? I need to take a closer look at them.

rcdailey, 2022-05-06:
Happy to but I don't know your email. Where are the cache files?

rcdailey, 2022-05-06:
Sorry, I see your email in your profile. I will send you the files. Thank you!

AlCalzone, 2022-05-12:
I'm not really sure what the issue is, but the driver attempts communication with a security class that is lower than the highest granted one, and it was certain that the higher ones were definitely not granted. This doesn't seem to be caused by the migration.

Try re-interviewing nodes 24 and 25 again, this time setting the checkbox for "reset security classes".