zwave-js/zwave-js #4584: Report MultiLevelSwitchCC event type as a string instead of as a number

j9brown, 2022-05-06:
**Is your feature request related to a problem? Please describe.**

In the course of identifying various issues with the new MultiLevelSwitchCC event feature, I've found it a bit hard to interpret the `event_type` field associated with these messages.  I had to dig through the zwave-js source code to [find the correct values](https://github.com/zwave-js/node-zwave-js/blob/master/packages/zwave-js/src/lib/commandclass/_Types.ts#L1029) (though I could have looked at the Zwave spec).

**Describe the solution you'd like**

Instead of reporting `event_type` as a number, consider using a string: such as `start` (event type 4) and `stop` (event type 5). 

**Describe alternatives you've considered**

Instead of changing the field type, you could add a new field alongside it to avoid breaking changes.  That said, since this is a new feature that's still being shaken out, there may not be many clients affected by such a change.

AlCalzone, 2022-05-09:
Keeping the enum values lets us stay in sync with the specification and opens up an easier possibility for future extension. IMO this is mainly a documentation issue - the relevant (auto-generated) documentation should mention all this.

AlCalzone, 2023-02-06:
I believe this is solved with https://github.com/zwave-js/node-zwave-js/pull/4652/files

closed 2023-02-06 by AlCalzone