zwave-js/zwave-js #4585: ZooZ devices (ZEN17, ZEN32) fail to interview (ZST10-700 controller, FW: v7.17.2 SDK: v7.17.2)

jfroy, 2022-05-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Re-interview my ZEN17 or ZEN32 device.

What do you observe?
Interview never finishes. Log says the interview fails. zwave-js tries 5 times then gives up.

What did you expect to happen?
Interview succeeds.

Steps to reproduce the behavior:
  1. Toggle open the device.
  2. Click on Advanced.
  3. Click on INTERVIEW under "Re-interview Node"

### Device information

Manufacturer: Zooz
Model name: ZEN17
Node ID in your network: 8

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

zwavejs2mqtt: 6.8.1
zwavejs: 9.0.7
zwavejsmqtt HASS add-on:  0.39.0
HASS core: 2022.5.1
HASS supervisor: 2022.05.0
HASS OS: 7.6

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_2022-05-06.log](https://github.com/zwave-js/node-zwave-js/files/8644069/zwavejs_2022-05-06.log)

AlCalzone, 2022-05-09:
So, the issue here is that the Zooz switches send config parameter information like this
> Restore state after power failure: always off, always or follow the status before power failure.

in unnecessarily short chunks (4 characters):
```
2022-05-06T23:36:35.377Z DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]                          [partial]
                                  │ RSSI: -71 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 56
                                    └─[ConfigurationCCInfoReport]
                                        parameter #:       1
                                        info:              Rest
                                        reports to follow: 24
2022-05-06T23:36:35.422Z SERIAL « 0x011e00a8000108159f0339003656e7065e9d37124aacc177dc811c3c0300b900  (32 bytes)
2022-05-06T23:36:35.424Z SERIAL » [ACK]                                                                   (0x06)
2022-05-06T23:36:35.427Z DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]                          [partial]
                                  │ RSSI: -71 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 57
                                    └─[ConfigurationCCInfoReport]
                                        parameter #:       1
                                        info:              ore
                                        reports to follow: 23
2022-05-06T23:36:35.473Z SERIAL « 0x011e00a8000108159f033a0094df7635f2f03e3b7e2fa4cda3bd7f413800b965  (32 bytes)
2022-05-06T23:36:35.476Z SERIAL » [ACK]                                                                   (0x06)
2022-05-06T23:36:35.479Z DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]                          [partial]
                                  │ RSSI: -71 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 58
                                    └─[ConfigurationCCInfoReport]
                                        parameter #:       1
                                        info:              stat
                                        reports to follow: 22
2022-05-06T23:36:35.525Z SERIAL « 0x011e00a8000108159f033b00be70f981d9120987ca35b128e3b3e31bf800b98e  (32 bytes)
2022-05-06T23:36:35.528Z SERIAL » [ACK]                                                                   (0x06)
2022-05-06T23:36:35.530Z DRIVER « [Node 008] [REQ] [BridgeApplicationCommand]                          [partial]
                                  │ RSSI: -71 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 59
                                    └─[ConfigurationCCInfoReport]
                                        parameter #:       1
                                        info:              e af
                                        reports to follow: 21
```

This takes so long that eventually Z-Wave JS will give up waiting. We can work around this using compat flags, but it would be good if you complain to their support aswell. 

We reported this a while back, but they didn't seem to care.

closed 2022-05-09 by AlCalzone

laheym, 2022-06-21:
> but it would be good if you complain to their support aswell.

@AlCalzone,  I have sent zooz support a request to address this issue, or at least get you a list of every device that needs the workaround applied.