zwave-js/zwave-js #4589: [Bug]: all devices goes unavailable when high load

prilly-dev, 2022-05-09:


AlCalzone, 2022-05-09:
I can't do anything with the first log since it is not on loglevel "debug", but the 2nd one indicates that Node 11 has connectivity issues, causing it to flood the network with duplicated reports.

If you haven't already, please follow the steps in https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting to try and troubleshoot that.

prilly-dev, 2022-05-09:
Yes did notice node 11, but that is a respons to a congested network with node 25 and that nonce crap.There are something wrong with zwavejs and how it handles locks with s0 
-------- Opprinnelig melding --------Fra: AlCalzone ***@***.***> Dato: 09.05.2022  14:43  (GMT+01:00) Til: zwave-js/node-zwave-js ***@***.***> Ko: prilly-dev ***@***.***>, Author ***@***.***> Emne: Re: [zwave-js/node-zwave-js] [Bug]: all devices goes unavailable when high load (Issue #4589) 
I can't do anything with the first log since it is not on loglevel "debug", but the 2nd one indicates that Node 11 has connectivity issues, causing it to flood the network with duplicated reports.
If you haven't already, please follow the steps in https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting to try and troubleshoot that.

AlCalzone, 2022-05-09:
> There are something wrong with zwavejs and how it handles locks with s0 

Unless you can prove this with a driver log on "debug", I disagree.

closed 2022-05-20 by zwave-js-assistant[bot]