zwave-js/zwave-js #4596: [question] Unknown manufacturer 0xXXXX listed for device in database

mew1033, 2022-05-09:
I have a device (Schlage Connect BE469) that is connected to my Zwave network. It works fine as far as I can tell and has for a few years. However, for some reason zwavejs2mqtt can't seem to get the manufacturer or product. This is strange because in the debug info, the manufacturer and model are clearly shown in the `hassDevices` section. Is there a way to tell zwavejs what device this is? Or some debug logs I can generate to see why it's not getting detected?

Thanks!

AlCalzone, 2022-05-09:
@nathan-curtis 
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.
While the log is running, re-interview the lock.

WhiteNight121, 2022-05-10:
thanks for re-opening this, would it be possible to add the BE469NX model if not i can look at doing it myself but getting the PIN stuff to work might be a bit above me im just trying to avoid buying another 200$ lock. thanks all.

AlCalzone, 2022-05-11:
@WhiteNight121 in general the Z-Wave functionality should be mostly working even if the device isn't identified due to a missing configuration file. For many devices, these are only cosmetical and/or convenience (labels and pre-defined configuration parameters).

If not, see my comment directly above yours.

WhiteNight121, 2022-05-11:
Hello @AlCalzone thanks so much for the help, i have attached the logs, this was working before i updated to zwavejs so as far as i know the lock should be fine, i have had some issues getting to add but it says that its connected in zwavejs it just will not let me control it.

and im on version 6.9.0 i updated after i took the logs but its still acting the same and HA can see the lock (only as the node ID) and just says everything is Unavailable that has to do with that node.

thanks again for the help, if more info is needed please let me know.

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8673666/zwavejs2mqtt-store.zip)
.

AlCalzone, 2022-05-12:
@WhiteNight121 At the beginning of the interview, your stick is unable to transmit anything for several seconds. Right when the secure communication is figured out:
```
2022-05-11T21:44:09.568Z DRIVER » [Node 010] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      147
                                  └─[SecurityCCNonceGet]
2022-05-11T21:44:09.570Z SERIAL « [ACK]                                                                   (0x06)
2022-05-11T21:44:09.571Z SERIAL « 0x0104011300e9                                                       (6 bytes)
2022-05-11T21:44:09.572Z SERIAL » [ACK]                                                                   (0x06)
2022-05-11T21:44:09.572Z DRIVER « [RES] [SendData]
                                    was sent: false
2022-05-11T21:44:09.573Z CNTRLR   The controller response indicated failure after 1/3 attempts. Scheduling next 
                                  try in 100 ms.
```
(this goes on for a while).

As a result, Z-Wave JS is unable to communicate with the lock which is why it isn't working properly.

Can you do the same again to see if it is reproducible? Please remove logs first though so you don't have to transmit 30 MB and it is easier to compare.

WhiteNight121, 2022-05-12:
@AlCalzone see attached and yes sorry for the large file.

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8683007/zwavejs2mqtt-store.zip)

AlCalzone, 2022-05-13:
@WhiteNight121 different error this time, again right at the beginning when it counts:
> transmit status: NoAck

This means you do have a physical connectivity issue - either the node doesn't get some messages or the acknowledgement from the node doesn't reach the controller.

Your first steps should be:
1. Follow https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting to exclude the most common source of these issues
2. Tone down the power reporting of your nodes (report on **significant** changes, not every 0.01 kWh or 0.1 W, timed reports only as backup, every hour or so).
During each transmission attempt, there are on average 5 to 10 incoming MeterCCReports from several nodes, sometimes in bursts of multiple messages per second. In the short log of 2.5 minutes, this amounts to 142 such reports alone. This can easily cause other commands to physically get lost.

WhiteNight121, 2022-05-14:
Thanks so much I'll give that a try I was starting to think the lock could not see my hub... Just strange as that it was working

WhiteNight121, 2022-05-21:
Hello @AlCalzone I have made some progress, i put zwaveJS on a Pi and moved it back to the location it was working before i swapped over to zwaveJS and it now shows the lock name ect, but im still unable to control it i looked at the logs and it looks like it finishes the add if im reading it right.. thanks again for all your help.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8748230/zwavejs2mqtt-store.zip)

[image]

WhiteNight121, 2022-05-29:
Hello so i have moved the Zwave receiver 2in from the lock and removed it and then re-added it and as far as i can tell its acting the same way.

AlCalzone, 2022-06-09:
@WhiteNight121 sorry this slipped through. One thing I can imagine is that the lock was not paired with the same security key you have configured in Z-Wave JS, which makes it impossible to communicate with it properly.

You can try excluding the lock, then factory resetting it to keep it from re-using an old/outdated S0 key and then include it again. Make a driver log of the inclusion process and if things don't work, send this to me via e-mail (see my profile).

WhiteNight121, 2022-06-09:
np im sure you are busy, i did factory reset it as i thought that might be the issue as well and then re-add but as far as i can tell it acted the same way it did before.

i will grab a log this weekend and send over to you thanks so much

AlCalzone, 2022-06-10:
Oh you can actually post the log here - we've stopped logging network keys in 9.x

WhiteNight121, 2022-06-12:
@AlCalzone thanks again for your time, see attached.
[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8886874/zwavejs2mqtt-store.zip)

AlCalzone, 2022-06-13:
So, zwave-js assigns the node a route back to the controller, which succeeds:
```
2022-06-12T21:46:36.118Z CNTRLR » [Node 010] Assigning SUC return route...
2022-06-12T21:46:36.129Z SERIAL » 0x010500510a10b1                                                     (7 bytes)
2022-06-12T21:46:36.130Z DRIVER » [Node 010] [REQ] [AssignSUCReturnRoute]
                                    payload: 0x0a10
2022-06-12T21:46:36.132Z SERIAL « [ACK]                                                                   (0x06)
2022-06-12T21:46:36.139Z SERIAL « 0x0104015101aa                                                       (6 bytes)
2022-06-12T21:46:36.140Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T21:46:36.141Z DRIVER « [RES] [AssignSUCReturnRoute]
                                    was executed: true
...
2022-06-12T21:46:39.956Z DRIVER « [REQ] [AssignSUCReturnRoute]
                                    callback id:     16
                                    transmit status: OK
```
But the very next command which is used to start the key exchange fails already (it gets acknowledged by the node, but yields no response):
```
2022-06-12T21:46:40.035Z DRIVER » [Node 010] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      17
                                  └─[SecurityCCSchemeGet]
2022-06-12T21:46:40.037Z SERIAL « [ACK]                                                                   (0x06)
2022-06-12T21:46:40.040Z SERIAL « 0x0104011301e8                                                       (6 bytes)
2022-06-12T21:46:40.041Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T21:46:40.042Z DRIVER « [RES] [SendData]
                                    was sent: true
2022-06-12T21:46:40.086Z SERIAL « 0x010500131100f8                                                     (7 bytes)
2022-06-12T21:46:40.088Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T21:46:40.089Z DRIVER « [REQ] [SendData]
                                    callback id:     17
                                    transmit status: OK
2022-06-12T21:46:41.172Z CNTRLR   [Node 010] Timed out while waiting for a response from the node (ZW0201)
```

The node responds to nonce requests fairly quickly, so this doesn't seem like a reception issue:
```
2022-06-12T21:46:41.199Z DRIVER » [Node 010] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      18
                                  └─[SecurityCCNonceGet]
[...]
2022-06-12T21:46:41.254Z DRIVER « [Node 010] [REQ] [ApplicationCommand]
                                  └─[SecurityCCNonceReport]
                                      nonce: 0xe07fcda741b7dc6d
```

---

Looking at the specifications for the command that yields no response:
> A node is considered newly included if it has been included for less than 10 seconds.
A newly included node MUST return the Security Scheme Report Command in response to this
command.
Whether a node has been included securely or non-securely, the node MUST NOT respond to the
Security Scheme Get command if it is not newly included.

Either this means the node's timer of 10s starts before the inclusion was fully completed, maybe at the "node found":
> 2022-06-12T21:46:30.606Z CNTRLR   handling add node request (status = NodeFound)
(but this would still be slightly less than 10s before the request).

Or it was not reset completely and therefore does not consider itself newly-included.

---

I'd say this is either an issue with the lock, or user error. In any case this doesn't look like something I can help you with and you should try contacting the manufacturer support about it.

closed 2022-06-30 by zwave-js-assistant[bot]