zwave-js/zwave-js #4606: Heltun HE-HLS01 missin Thermostat mode parameters

jarmokentta, 2022-05-10:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

The device can be installed in two modes, binary switch or thermostat.
Both modes have slightly different parameter lists.
Regardless of the state the device is set to, the same binary switch parameters are displayed.
Can you also add support for thermostat mode parameters.

The manual where the parameters are displayed can be downloaded from the Heltun website
Direct link to the document:
https://drive.google.com/file/d/1Cq096wYg-5X-7n8yloEaAXbWGpXze-Mw/view?usp=sharing

### Device information

Manufacturer:  Heltun
Model name:  High Load Switch HE-HLS01
Node ID in your network:  50

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 9.0.7
serverversion: 1.16.1

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Z-Wave JS -lokiviestit tilattu…
2022-05-10T11:34:43.350Z SERIAL » 0x0118001332119f038700949bd8532c6848295d83aa9978255f82              (26 bytes)
2022-05-10T11:34:43.352Z DRIVER » [Node 050] [REQ] [SendData]
                                  │ transmit options: 0x25
                                  │ callback id:      95
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 135
                                    └─[ConfigurationCCSet]
                                        parameter #:      7
                                        reset to default: false
                                        value size:       1
                                        value format:     SignedInteger
[... 215 lines cut ...]
2022-05-10T11:34:51.491Z DRIVER « [Node 043] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      3
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       V
                                        rate type:   Consumed
                                        value:       239.039
                                        time delta:  60 seconds
                                        prev. value: 238.807
2022-05-10T11:34:51.697Z SERIAL « 0x01180004002b12600d01003202a16c00000000003c0000000077              (26 bytes)
2022-05-10T11:34:51.699Z CNTRLR   [Node 043] [~] [Meter] value[66817]: 0 => 0                       [Endpoint 1]
2022-05-10T11:34:51.702Z SERIAL » [ACK]                                                                   (0x06)
2022-05-10T11:34:51.705Z DRIVER « [Node 043] [REQ] [ApplicationCommand]
                                  └─[MultiChannelCCCommandEncapsulation]
                                    │ source:      1
                                    │ destination: 0
                                    └─[MeterCCReport]
                                        type:        Electric
                                        scale:       A
                                        rate type:   Consumed
                                        value:       0
                                        time delta:  60 seconds
                                        prev. value: 0

AlCalzone, 2022-05-10:
@jarmokentta please **attach** the log as a .txt file by dragging it into the comment box. Copy paste is pretty unreadable, even with code formatting.

AlCalzone, 2022-05-10:
We could add the missing ones and mention which ones are only available in which mode.

But it looks like some of the parameters are actually different in size and ranges etc. depending on the mode. We definitely don't support that.

To set those parameters correctly, you will need to use zwavejs2mqtt and enter these values according to the manual:
[image]

AlCalzone, 2022-05-10:
Accepting PRs to add the missing parameters. Please do mention in the description which mode they belong to.

For those schizophrenic parameters please do the same and refer to setting the parameter manually for the other mode.