zwave-js/zwave-js #4611: Controller becomes unresponsive when trying to remove dead node

kendrome, 2022-05-11:


AlCalzone, 2022-05-11:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

AlCalzone, 2022-05-12:
Okay, that's one I haven't seen before. The entire stick becomes unresponsive when trying to ping the node during the startup procedure. When you attempt to remove it, it is already too late. Unfortunately, there's no way to disable the pinging, since it determines the node status. Manually editing the controller memory is out of the question since your stick firmware is too old.

If you have access to a Windows PC, you can try to remove the node using Silabs' PC controller (part of Simplicity Studio).

closed 2022-05-15 by zwave-js-assistant[bot]