zwave-js/zwave-js #4617: Aeon labs / aeotec water sensor 6 zw122 tamper / vibration notify not clearing

apaperclip, 2022-05-13:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

(Using the zwavejs2mqtt ui for visibility)

When the device is moved it will beep and report it was tampered with (value 3)  This tampering event never clears. 
 

What do you observe?
After a fresh inclusion or re-interview the tamper value is `0`.  When the device is moved it will beep and report it was tampered with (value 3)  This tampering never clears.  Even though the case is closed, the` [6-113-0-Home Security-Cover status] Cover status` is updated to `[3] Tampering, product cover removed` when the device is moved.  If I continue to move the device it continually reports value `3 => 3`

What did you expect to happen?
Should case open, vibration, and tilt all use the same notification? It looks like all three are under the `home security` notification class.

Regardless, I would expect the device to report a clear like it does for water (tested) and heat (untested). The manual indicates it supports this in section 5.6

I let the logs go for a few more hours to catch the scheduled wakeups from the device.

Steps to reproduce the behavior:
  1. Discover or refresh zw122 water sensor 6 
  2. See tamper value is off
  3. Shake device, tamper turns on
  4. Tamper stays on, seemingly never clears even after several hours.

### Device information

Manufacturer:   Aeon labs / aeotec
Model name:  Water Sensor 6 / zw122
Node ID in your network:  6

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:   9.2.2 (in zwjs2mqtt 6.9.1)
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2022-05-12.log](https://github.com/zwave-js/node-zwave-js/files/8683412/zwavejs_2022-05-12.log)

AlCalzone, 2022-05-13:
> Should case open, vibration, and tilt all use the same notification? It looks like all three are under the `home security` notification class.

There exist several independent status variables under the Home Security notification, but this is really a question for Aeotec. This device only supports a single one:

> notification type: Home Security
> supported events:
> · Tampering, product cover removed

As for the original problem: The device implements Notification CC v7 and as such isn't required to idle the notification variables itself. We can force auto-idling notifications after a certain time via a compat flag, but this doesn't make sense here. If you just remove the cover and leave it off, the variable really shouldn't be reset.

All I see in your log are three seperate instances of "Tampering, product cover removed". Never idle.

IMO you need to take this to Aeotec, this isn't something that can be solved on the driver side.

apaperclip, 2022-05-13:
Thank you for the review and reply.

apaperclip, 2022-05-16:
Thanks for the comment. Thankfully the base function seems to be reliable, water yes/no is what is most important.

I will close this issue.

closed 2022-05-16 by apaperclip