zwave-js/zwave-js #4623: `options is not a MeterCCResetOptions` when calling zwave_js.meter_reset

gregsheremeta, 2022-05-15:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Calling the following from home assistant:
```
service: zwave_js.reset_meter
target:
  entity_id: sensor.home_energy_meter_leg_1_electric_consumption_kvarh
```

```
service: zwave_js.reset_meter
target:
  device_id: 3d78bea7b7a667a9868cb469e8d3ec15
```

What do you observe?
home-assistant.log contains something like the following:
```
WSMessage(type=<WSMsgType.TEXT: 1>, data='{"type":"result","success":false,"messageId":"dd70b577137a4d40ae03d31b5e41158f","errorCode":"zwave_error","zwaveErrorCode":322,"zwaveErrorMessage":"options is not a MeterCCResetOptions (ZW0322)"}', extra='')
```
Oddly, I see absolutely nothing in the zwavejsmqtt app log or the driver log, despite having them both on debug. I might have something configured wrong. Or maybe because this device is S0?

What did you expect to happen?
Successful meter reset

Other information

meter_reset on the device has been working fine nightly for the last several months. Last week I updated both my homeassistant and zwavejsmqtt containers to latest, and that's when this started happening.

The meter is an Aeotec ZW095. I reinterviewed it, but that didn't help.

### Device information

Manufacturer: Aeotec
Model name: ZW095
Node ID in your network: 5

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:

zwavejs/zwavejs2mqtt:master
homeassistant/home-assistant:latest

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

container images from ~March 2022.

### Attach Driver Logfile

I see nothing happen in the logs

gregsheremeta, 2022-05-15:
I forgot to mention -- clicking the "Reset accumulated values" button in zwavejsmqtt control panel works -- it does reset the meter if I do that.

raman325, 2022-05-15:
Ok so then it's a HA bug not a driver issue. That was my mistake. I think we can close this one, can you add the comment about z2m to the ha/core issue?

raman325, 2022-05-15:
Actually leave this open until I get a chance to investigate

AlCalzone, 2022-05-15:
Like I wrote on discord, the API method argument should be made optional.

closed 2022-05-16 by AlCalzone