zwave-js/zwave-js #4630: Inovelli Red on/off switch  - lzw30-sn  security column has gray question mark circle in zwjs2mqtt

apaperclip, 2022-05-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Added an lzw30-sn with no encryption

What do you observe?
Security is unknown (no encryption was selected during inclusion) Other no security devices have red hyphen circle (they are not inovelli)

Node debug info json has  isSecure as unknown as well

What did you expect to happen?
Security status should be red hyphen circle (no security mouse over)

Steps to reproduce the behavior:
  1. Include device with no security
  2. Put device in incusion mode (single up-click)
  3. Observe control panel device overview table.
  4. Note gray question mark for Security

### Device information

Manufacturer: Inovelli
Model name: lzw30-sn
Node ID in your network: 10

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 9.2.2 (in zwjs2mqtt 6.9.1)
node-zwave-js branch:
zwavejs2mqtt branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Logs include a re-interview with Reset Security Classes checked
[zwavejs_2022-05-18.log](https://github.com/zwave-js/node-zwave-js/files/8717387/zwavejs_2022-05-18.log)
.

AlCalzone, 2022-05-18:
This is why:
> [Node 010] supports Security S2, but no S2 network keys were configured.

The driver cannot determine if the node has any S2 keys granted.

apaperclip, 2022-05-18:
I can add them, but since I don't need anything with S2 I was not going to define them.

my thought process is:

Is the security value supposed to indicate how the node is associated or what is possible?  
I would expect if the device is using not encryption, then defining s2 keys would not matter. The driver knows its not using encryption so the security value should be none. Why would the driver care if s2 keys are not provided if they're not needed for a node?

AlCalzone, 2022-05-18:
> The driver knows its not using encryption

It doesn't. The current logic is, as long as at least one security class is not a definite yes or no, the security status is unknown.

AlCalzone, 2022-05-19:
Ok so this issue is twofold:
1. S0 and non-secure inclusion didn't correctly remember which keys were **not** granted, resulting in this (?) status. This will be fixed in https://github.com/zwave-js/node-zwave-js/pull/4637
2. When S2 keys are not configured, after re-interview with "reset security classes" it is impossible to programmatically determine the security status of an **insecure** device for sure. It could be that the device was previously included with S2, yet without keys this cannot be tested.

closed 2022-05-19 by AlCalzone

apaperclip, 2022-05-19:
Thank you for looking at this again!

PR #4637 (2022-05-19): fix: during bootstrapping, remember which security classes were not granted
fixes: #4630