zwave-js/zwave-js #4633: no events in developer tools

bp47025, 2022-05-18:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

First posted here: https://github.com/home-assistant/core/issues/72110

Was advised to post here. 

I am new to Z-wave JS (migrating from vera to ha) and I am trying to get some automation configured.

When I go directly to the Z-wave add-on log I can see all activity on the z-wave network. As an example, I grabbed the output on a button press from a minimote below:

```
2022-05-18T19:49:56.500Z DRIVER « [Node 015] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -93 dBm
                                  └─[SceneActivationCCSet]
                                      scene id:         3
                                      dimming duration: 0s
```
When I try to listen to any event either zwave_js_notification or zwave_js_value_notification there are 0 events. I have tried numerous paddle switches, outlets, and door sensors. None receive any events but do show in the Z-wave log.

I have provided diagnostic data below.
[config_entry-zwave_js-09333b058d96facf9f4ea14037a1caf6.json.txt](https://github.com/zwave-js/node-zwave-js/files/8720766/config_entry-zwave_js-09333b058d96facf9f4ea14037a1caf6.json.txt)

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

When I go to the Z-Wave JS Logs, i see no logs. I have changed log level and restarted both the core and host numerous times. There are no logs.

raman325, 2022-05-18:
@bp47025 it looks like your issue is not a bug, it's that your node isn't ready: https://github.com/home-assistant/core/issues/72110#issuecomment-1130519304

bp47025, 2022-05-19:
I just posted an update in the other thread, I referenced a different device that was ready. Still not seeing events on a device that was ready.

AlCalzone, 2023-02-06:
Is this still an issue?

closed 2023-02-17 by zwave-js-assistant[bot]