zwave-js/zwave-js #4641: Yale locks missing alarmLevel for Manual Lock Operation notifications

elmigbot, 2022-05-19:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

For the Yale lock YDR256, the `alarmLevel `is not being consumed for the `alarmType` of 21. I see this at:
`/packages/config/config/devices/0x0129/templates/yale_template.json`

What do you observe?

This causes the system to be unaware of the type of Manual Lock performed. It should report "by key cylinder or inside thumb-turn" for `alarmLevel=1`, and "by touch function (lock and leave)" for `alarmLevel=2`. (There is a third level, "by internal button" that must be present on other models).

I'm getting this documentation from: [here](https://products.z-wavealliance.org/ProductManual/File?folder=&filename=Manuals/3267/YRL216_226_236_256_YaleRealLivingAssureLever_Z-WavePlus_SystemIntegrator.._.pdf)

What did you expect to happen?

Knowing which type of lock operation occurred lets you infer if the cat sitter has left the house. :)

Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: 
Model name: 
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

latest addon version  0.1.58

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Not applicable

AlCalzone, 2022-05-19:
The "lock reasons" you mention are not represented in the standardized notifications, so they cannot be mapped. You'll have to manually react to changes in the `alarmType` and `alarmLevel` values if you need that information.

elmigbot, 2022-05-19:
Hi,

How can I see `alarmType `and `alarmLevel ` in home assistant? I thought the event was all I had access to, which only includes:

```
"{
            "command_class": 113,
            "command_class_name": "Notification",
            "label": "Access Control",
            "type": 6,
            "event": 1,
            "event_label": "Manual lock operation",
            "parameters": {}

}
```

AlCalzone, 2022-05-19:
That's a question you need to ask in the Home Assistant support channels to get a definite answer. I'm guessing they may just be hidden.

elmigbot, 2022-05-19:
I was under the assumption that this raw data never leaves node-zwave-js, hence the mapping in yale_template.json.

elmigbot, 2022-05-19:
I've confirmed the only data that leaves the zwave-js server is (schema v16):

```
{
    "source": "node",
    "event": "notification",
    "nodeId": 12,
    "ccId": 113,
    "args": {
        "type": 6,
        "event": 4,
        "label": "Access Control",
        "eventLabel": "RF unlock operation"
    }
}
```

If you are not keen on adding a new notification type, could we add the `alarmLevel `to the `eventParameters `similar to how keypad unlock uses it:

```
"alarm_map_keypad_unlock": {
    "from": {
        "alarmType": 19
    },
    "to": {
        "notificationType": 6, // Access Control
        "notificationEvent": 6, // Keypad unlock operation
        "eventParameters": {
            **"userId": "alarmLevel"**
        }
    }
}
```
Thanks

AlCalzone, 2022-05-20:
Alarm level and type are not part of the Notification events, which are reserved for standardized notifications.
They are normal values (or sensors?), you're probably looking in the wrong place.

kpine, 2022-05-20:
The alarm level and type should be available as numeric sensors in HA. @elmigbot Can you attach the Device Diagnostic file (download from Device page) for this node so we can verify if they exist? Also, since you have the server debug messages, can you post a log of the entire sequence of that? Probably wouldn't hurt to include driver debug logs as well.

elmigbot, 2022-05-20:
Hi AlCalzone, I'm not sure I understand your point. Its clear `alarmLevel `and `alarmType `are mapped to `notificationtype `and `notificationevent`. This is evident in the attached device file. I'm ok having these mapped values instead.

But, the mapping is simply incomplete for the manual lock notification (`alarmType=21`), because its missing inputs `alarmLevel` 1,2, and ostensibly 3. Including these to the mapping by either creating 2 additional `notificationevents`, or simply including them as `eventParameters `would solve the issue. 

Kpine, I just enabled the sensors, but they are not reporting any values. (Makes sense... those values are not part of zwave server message.) 

I've attached the files. The logs cover a manual unlock by cylinder and manual lock by touch button. The relevant node is 12.

Thank you both.
[zwave_js-a47b01a52874147d13d600cebed90c71-Assure Touchscreen Deadbolt-4637ffcdea57624b8cc0c3577a4217c9.json.txt](https://github.com/zwave-js/node-zwave-js/files/8741503/zwave_js-a47b01a52874147d13d600cebed90c71-Assure.Touchscreen.Deadbolt-4637ffcdea57624b8cc0c3577a4217c9.json.txt)
[zwaveResponseV16.json.txt](https://github.com/zwave-js/node-zwave-js/files/8741802/zwaveResponseV16.json.txt)
[driver.txt](https://github.com/zwave-js/node-zwave-js/files/8741804/driver.txt)

firstof9, 2022-05-20:
The manual lock notification is complete.

When you physically turn the lock, there's no user and the lock will always report `alarmLevel` 0 or 1, nothing else, as it has no way to know which user's fingers moved the lock latch.

elmigbot, 2022-05-20:
There are 3 types of manual lock notifications according to the manufacturer. (I've included the supplier documentation above.) And indeed, I do have a use case for deciphering between them.

Here they are for convenience:
[image]

[image]

`AlarmLevel `does not always represent `userID`. In this case, it indicates the method of locking. Your example above did not include pushing the lock button from outside. 

It's precisely this case I'd like to trigger automations, because its the only way to determine when an otherwise untracked person has left the house, instead of just locking the door behind themselves as they enter.

The 3rd method is not available on my unit, I can only lock from inside by rotating the cylinder. However, probably best to support it.

firstof9, 2022-05-20:
To be fair my lock only reports `alarmLevel` 1 even when locking from the outside button.

kpine, 2022-05-20:
FYI, your lock is not using alarm mappings (otherwise it would say so in the logs), so either the lock doesn't actually report those alarm levels, or Z-Wave JS is ignoring them because it uses Notification CC version 4. I'll have to defer to Al to determine that.

elmigbot, 2022-05-20:
> To be fair my lock only reports `alarmLevel` 1 even when locking from the outside button.

Yikes, I hope you don't have the same model as me, Yale YDR256. It would be a bummer if the lock did not actually do what it says it does. Looking at the zwave log I could not determine what is actually being received from the device...

elmigbot, 2022-05-20:
> FYI, your lock is not using alarm mappings (otherwise it would say so in the logs), so either the lock doesn't actually report those alarm levels, or Z-Wave JS is ignoring them because it uses Notification CC version 4. I'll have to defer to Al to determine that.

Hmmm, if it doesn't use alarm mappings, then how would it report the userID when doing a keypad unlock? I can confirm it does do that properly. 

Fingers crossed.

Thanks

firstof9, 2022-05-20:
> if it doesn't use alarm mappings, then how would it report the userID when doing a keypad unlock?

Via the notification command class.

elmigbot, 2022-05-20:
> > if it doesn't use alarm mappings, then how would it report the userID when doing a keypad unlock?
> 
> Via the notification command class.

LOL, I must be missing something. The alarm mappings are part of the notification command class, no?

kpine, 2022-05-20:
> Hmmm, if it doesn't use alarm mappings, then how would it report the userID when doing a keypad unlock? I can confirm it does do that properly.

It's part of the Z-Wave standard, for Notification CC V4. They are defined in the document "SDS13713 Notification Command Class". User code is part of the event data for an unlock notification.

The alarm mappings are to make older locks support the same set of "modern" notifications. V1 alarm values are manufacturer specific.

elmigbot, 2022-05-20:
ah ok, understood. I'm possibly looking at unused code, if it is indeed using Notification CC V4.

AlCalzone, 2022-05-23:
@elmigbot can you please send me your S0 network key via email? You'll find it in my github profile. I need it to see the raw data for the `NotificationCCReport`.
At a high level, the log looks like the report isn't actually mapped but interpreted as a V2+ Notification, for which the specs require the alarm level and type to be zero. This is probably the reason you're not seeing these values, since they are being ignored.

elmigbot, 2022-05-24:
> @elmigbot can you please send me your S0 network key via email? You'll find it in my github profile. I need it to see the raw data for the `NotificationCCReport`. At a high level, the log looks like [...]

Key has been emailed, thanks.

elmigbot, 2022-06-01:
No stale!

AlCalzone, 2022-06-10:
Hey @elmigbot I can't access the key. Can you send it again?

elmigbot, 2022-06-15:
Sorry about that, sent again.

elmigbot, 2022-06-27:
Sent again... thanks.

AlCalzone, 2022-06-28:
Thanks, this time it worked. This should be solved in the next release.
This was no mapping FYI, but zwave-js ignoring the alarm values for V2+ notifications. I thought that made sense to cut down on "useless" values, but in your case they actually add information.

PR #4756 (2022-06-30): feat: preserve V1 alarm values in V2+ notifications
fixes: #4641
by preserving the alarm values for V2+ notifications unconditionally. The example in the fixed issue is one where alarm values provide a bit more specific information than the notification, so both are good.
closed 2022-06-30 by AlCalzone