zwave-js/zwave-js #4643: CT101 Thermostat Device Driver Not reporting humidity

ur7x, 2022-05-20:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?  The ZWave Device Driver for the CT101 thermostat does not report humidity

What do you observe?  Added the Thermostat to ZWave JS and ZWave2MQTT and it reports temperature fine, but there is no defined property/event for Humidity.  There is a setting for humidity sensitivity.  Which should trigger a report, it never reports.

What did you expect to happen?  The thermostat will report temperature and humidity.. 

Steps to reproduce the behavior:
1 Add Thermostat as a new Zwave  node
2 Check Multi Level Sensor V2 in ZWave2MQTT
3) Only Air Temp... No Humidity report. 

This was reported and a fix was attempted:
https://github.com/zwave-js/node-zwave-js/pull/3079

But on my device this does not work.  

### Device information

Manufacturer: Radio Thermostat
Model name: CT110
Node ID in your network: 2

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:  9.2.2
`zwavejs2mqtt` branch:  0.40.0

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

Device Diagnostics 

[zwave_js-3333c9afebd50ae21deb4c660e5cafa6-Family Room CT101-ed7c91b32e7cd1c5d8740a79fde9e007.json(2).txt](https://github.com/zwave-js/node-zwave-js/files/8734983/zwave_js-3333c9afebd50ae21deb4c660e5cafa6-Family.Room.CT101-ed7c91b32e7cd1c5d8740a79fde9e007.json.2.txt)

Zwave Log

[zwavejs2mqtt_2022-05-19(2).log](https://github.com/zwave-js/node-zwave-js/files/8735180/zwavejs2mqtt_2022-05-19.2.log)

ur7x, 2022-05-20:
I left both a Log file from the device diagnostics and from Zwave2MQTT 
If a different log file is needed please let me know which one is needed and how to get it. 
Since I'm using the a Zwave2MQTT addon  the ZWave JS integration is not creating a log file (maybe that is the issue?)

ur7x, 2022-05-20:
[zwavejs_2022-05-20.log](https://github.com/zwave-js/node-zwave-js/files/8742341/zwavejs_2022-05-20.log)

Hi there, I think this is what you need.

kpine, 2022-05-20:
You should post the driver log that includes a Re-interview of the device.

ur7x, 2022-05-20:
Got it...  Sorry this Zwave to JS  and Zwave2MQTT is a little new to me...  is this the complete and correct log?

[zwavejs_2022-05-20(3).log](https://github.com/zwave-js/node-zwave-js/files/8742501/zwavejs_2022-05-20.3.log)

This is from the ZWave JS start right through a re-interview of the thermostat.

elbarsal, 2022-05-20:
This is a known quirk in the implementation for that version of device. I had previously sorted a solution on OpenZwave. ZWaveJS does handle the issue appropriately, but it seems you need to get it to listen to all endpoints to read the humidity (as duplicate endpoints get ignored).

My solution was to copy the default CT101 config file into config\0x0098\ and modify the file, adding a line to the "compat" section to preserve endpoints, like this:

`
"compat": {
                // The device has two endpoints. The temperature gets reported via the root, humidity is on endpoint 2.
                "preserveEndpoints": "*",
                "mapRootReportsToEndpoint": 1
        }`

I may have added too much (preserving * is non-selective) but it works. I am currently polling the value, but I think it may even notify on changes as well (I haven't stopped polling it).

ur7x, 2022-05-21:
> 

> My solution was to copy the default CT101 config file into config\0x0098\ and modify the file, adding a line to the "compat" section to preserve endpoints, like this

Can I get a little more info on your solution.  I found the current ct101.json file at https://github.com/zwave-js/node-zwave-js/blob/cd1995e49724edd1c284ea901b9c15865f22e8cb/packages/config/config/devices/0x0098/ct101.json

the "config\0x0098' directory is that within the storage section of Zwave JS? or is that within the config folder that Home Assistant uses?

Once that is done do you re-interview the device? stop and restart the Zwave JS service? Anything else?

Thanks in advance.

AlCalzone, 2022-05-21:
> My solution was to copy the default CT101 config file into config\0x0098\ and modify the file, adding a line to the "compat" section to preserve endpoint

Hm, the device classes for the affected variant are probably different from the one I originally added the fix for.
We now have the ability to selectively apply compat flags to certain variants included in a config file:
https://github.com/zwave-js/node-zwave-js/pull/4501

Would you be willing to open a PR that adds the `preserveEndpoints` flag for productID 0x000d?
Otherwise, I can do this on monday.

ur7x, 2022-05-21:
I have tried various "Compat" variables including the recommendation above and none will get the Thermostat to report humidity.
Do we need to manually define the two End Points?

ebarsaloucc, 2022-05-21:
I didn't need to do so - but I did need to re-interview the device - have you done so?

ur7x, 2022-05-22:
Re-interview WORKED... 

Funny with each JSON edit I would only see it applied if I restarted Zwave JS...  Never thought to re-interview. 

Re interview with your suggested change and Humidity is now reporting!  Thanks.

AlCalzone, 2022-05-22:
> Never thought to re-interview.

It's mentioned in the documentation for the custom config files though 😉

https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files?id=using-custom-device-configuration-files

▶️ Second big blue box

PR #4647 (2022-05-23): fix(config): preserve endpoints for RTC CT101
fixes: #4643
closed 2022-05-23 by AlCalzone