zwave-js/zwave-js #4644: Missing device configuration: Fakro ARF Solar

szeryf85, 2022-05-20:
### Checklist

- [X] It is not in the [configuration DB](https://devices.zwave-js.io/?search=3)

- [X] It was not [merged recently](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Apr+is%3Aclosed) or has a [pending PR](https://github.com/zwave-js/node-zwave-js/pulls?q=is%3Aopen+is%3Apr)

### Which device is missing?

Fakro ARF Solar

### Manufacturer ID

0x0085

### Product Type

0x0004

### Product ID

0x0012

### Firmware Version

1.1

### Is the device listed on the Z-Wave Alliance website?

No

### Do you have a manual?

[ARF_FAKRO_instrukcja_uzytkowania.pdf](https://github.com/zwave-js/node-zwave-js/files/8740378/ARF_FAKRO_instrukcja_uzytkowania.pdf)

AlCalzone, 2022-05-20:
Is there an English manual? 😅

szeryf85, 2022-05-20:
> Is there an English manual? 😅

I can't find in English :(

OXINARF, 2022-05-20:
https://www.fakro.com/att/COMMON/offer/servis/user_manuals/EN/ARF-ARF-D-ARP-SOLAR_EN.pdf

I don't know Polish and didn't compare the two completely but this seems to be it.

szeryf85, 2022-05-21:
Yes, this manual is correct.

futurexdesign, 2022-06-22:
Looking at the parameters in the manual, the device takes the same configuration values, and association groups as the ARZ ZWave Solar:  https://devices.zwave-js.io/?jumpTo=0x0085:0x0003:0x0112:0.0

@AlCalzone Would the appropriate fix be to add this product type & product ID to that existing config file, or to create a new one with the same parameter details?

AlCalzone, 2022-06-22:
@futurexdesign To avoid confusion between ARF/ARZ/... variants, I'd make a copy of the file and change label and product type/id in the copy.

PR #4740 (2022-06-27): feat(config): add Fakro ARF Solar, template Fakro configs
Add configuration for Fakro ARF Solar roller blind.   Based on configuration for ARZ Z-Wave Solar, as they share the same configuration parameters when looking at the respective manuals. 

Fixes #4644
closed 2022-06-27 by AlCalzone