zwave-js/zwave-js #4653: Node inclusion fails

klontje85, 2022-05-26:


AlCalzone, 2022-05-26:
Please make a [driver log](https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs), loglevel `debug` or `silly` and attach it here as a file.

klontje85, 2022-05-26:
Please find both logs attached. I disabled logging, deleted old logs, re-enabled logging, and started the inclusion.

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8778512/zwavejs2mqtt-store.zip)

klontje85, 2022-05-26:
I was able to include a Fibaro relay module, but unable to include any of the three remaining Qubino modules (of which I already have 8 of them that did successfully include). Node ID numbering did go up with every (failed) attempt as the Fibaro module got assigned ID 17 and the last successful module before that was 11.

klontje85, 2022-05-28:
@AlCalzone I was able to include another Qubino module, the only modules that are still failing are all of the same type: https://qubino.com/mini-dimmer-tech-2/

klontje85, 2022-05-30:
@AlCalzone I did, my controller is on a 3m passive USB extension cord away from interference. I have just ordered a 20m active USB extension cable and will try to bring the controller next to the module that fails to include and re-try. It does seem weird there is no reason provided, and would expect inclusion to not even be attempted if it were out of range.

AlCalzone, 2022-05-30:
I don't think it is out of range. Maybe the end device is affected by interference and not the stick? Unfortunately without a Zniffer, you won't be able to see what is being exchanged and what isn't. 

> will try to bring the controller next to the module that fails to include and re-try

Although this might be an option, do know that this should not be done regularly: https://github.com/zwave-js/node-zwave-js/issues/1995

closed 2022-06-17 by zwave-js-assistant[bot]