zwave-js/zwave-js #4656: Poorly performing nodes that won't heal

bepsoccer, 2022-05-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

When I try to heal a poorly performing node I know should be atleast a hop away and isn't I get , "DRIVER no handlers registered!" a few times and then it finishes the heal as successful and assigns no hops(direct to Node 1).  This should not be the case as it's a fair distance, is unreliable, and fails/constantly times out when trying to run a network health check on the node.  And the other weird things is it is assigned downstream neighbors...

### Device information

Manufacturer:  Zooz
Model name:  Various Zooz devices; ZEN71, ZEN76 are two know models with problematic nodes
Node ID in your network: 186

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
zwavejs2mqtt: 6.10.0
zwave-js: 9.3.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

```
2022-05-26 14:58:17.916 INFO ZWAVE: Calling api healNode with args: [ 186, [length]: 1 ]
2022-05-26T19:58:17.916Z CNTRLR [Node 186] Healing node...
2022-05-26T19:58:17.916Z CNTRLR » [Node 186] refreshing neighbor list (attempt 1)...
2022-05-26T19:58:17.920Z SERIAL » 0x01050048ba6961 (7 bytes)
2022-05-26T19:58:17.920Z DRIVER » [Node 186] [REQ] [RequestNodeNeighborUpdate]
callback id: 105
2022-05-26T19:58:17.924Z SERIAL « [ACK] (0x06)
2022-05-26T19:58:25.730Z SERIAL « 0x010500486921fa (7 bytes)
2022-05-26T19:58:25.731Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:25.732Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
callback id: 105
update status: UpdateStarted
2022-05-26T19:58:34.321Z SERIAL « 0x010500486922f9 (7 bytes)
2022-05-26T19:58:34.323Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:34.324Z DRIVER « [REQ] [RequestNodeNeighborUpdate]
callback id: 105
update status: UpdateDone
2022-05-26T19:58:34.326Z CNTRLR « [Node 186] neighbor list refreshed...
2022-05-26T19:58:34.326Z CNTRLR » [Node 186] Deleting SUC return route...
2022-05-26T19:58:34.332Z SERIAL » 0x01050055ba6a7f (7 bytes)
2022-05-26T19:58:34.332Z DRIVER » [Node 186] [REQ] [DeleteSUCReturnRoute]
payload: 0xba6a
2022-05-26T19:58:34.337Z SERIAL « [ACK] (0x06)
2022-05-26T19:58:34.338Z SERIAL « 0x0104015501ae (6 bytes)
2022-05-26T19:58:34.338Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:34.339Z DRIVER « [RES] [DeleteSUCReturnRoute]
was executed: true
2022-05-26T19:58:34.613Z SERIAL « 0x012100556a0000000000000000000000000000000000000000000000000000000 (35 bytes)
000e1
2022-05-26T19:58:34.614Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:34.615Z DRIVER « [REQ] [DeleteSUCReturnRoute]
callback id: 106
transmit status: OK
2022-05-26T19:58:34.618Z CNTRLR » [Node 186] Assigning SUC return route...
2022-05-26T19:58:34.619Z SERIAL » 0x01050051ba6b7a (7 bytes)
2022-05-26T19:58:34.620Z DRIVER » [Node 186] [REQ] [AssignSUCReturnRoute]
payload: 0xba6b
2022-05-26T19:58:34.625Z SERIAL « [ACK] (0x06)
2022-05-26T19:58:34.626Z SERIAL « 0x0104015101aa (6 bytes)
2022-05-26T19:58:34.626Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:34.628Z DRIVER « [RES] [AssignSUCReturnRoute]
was executed: true
2022-05-26T19:58:35.125Z SERIAL « 0x012100516b0000000000000000000000000000000000000000000000000000000 (35 bytes)
000e4
2022-05-26T19:58:35.126Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:35.127Z DRIVER « [REQ] [AssignSUCReturnRoute]
callback id: 107
transmit status: OK
2022-05-26T19:58:35.130Z CNTRLR » [Node 186] deleting return routes (attempt 1)...
2022-05-26T19:58:35.133Z SERIAL » 0x01050047ba6c6b (7 bytes)
2022-05-26T19:58:35.133Z DRIVER » [Node 186] [REQ] [DeleteReturnRoute]
payload: 0xba6c
2022-05-26T19:58:35.138Z SERIAL « [ACK] (0x06)
2022-05-26T19:58:35.139Z SERIAL « 0x0104014701bc (6 bytes)
2022-05-26T19:58:35.140Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:35.140Z DRIVER « [RES] [DeleteReturnRoute]
has started: true
2022-05-26T19:58:35.424Z SERIAL « 0x012100476c0000000000000000000000000000000000000000000000000000000 (35 bytes)
000f5
2022-05-26T19:58:35.425Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:35.426Z DRIVER « [REQ] [DeleteReturnRoute]
callback id: 108
transmit status: OK
2022-05-26T19:58:35.429Z CNTRLR » [Node 186] assigning return routes to the following nodes:
1
2022-05-26T19:58:35.429Z CNTRLR » [Node 186] assigning return route to node 1 (attempt 1)...
2022-05-26T19:58:35.442Z SERIAL » 0x01060046ba016d69 (8 bytes)
2022-05-26T19:58:35.442Z DRIVER » [Node 186] [REQ] [AssignReturnRoute]
payload: 0xba016d
2022-05-26T19:58:35.448Z SERIAL « [ACK] (0x06)
2022-05-26T19:58:35.448Z SERIAL « 0x0104014601bd (6 bytes)
2022-05-26T19:58:35.448Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:35.449Z DRIVER « [RES] [AssignReturnRoute]
has started: true
2022-05-26T19:58:35.935Z SERIAL « 0x012100466d0000000000000000000000000000000000000000000000000000000 (35 bytes)
000f5
2022-05-26T19:58:35.936Z SERIAL » [ACK] (0x06)
2022-05-26T19:58:35.937Z DRIVER « [REQ] [AssignReturnRoute]
callback id: 109
transmit status: OK
2022-05-26T19:58:35.941Z CNTRLR [Node 186] healed successfully
2022-05-26 14:58:35.941 INFO ZWAVE: Success zwave api call healNode true
```

AlCalzone, 2022-05-30:
Yeah I can't do anything with this short snippet. Thats not even half of what happens when a single node is healed.

bepsoccer, 2022-05-30:
> Yeah I can't do anything with this short snippet. Thats not even half of what happens when a single node is healed.

That's not a snippet, that's what gets outputted in its entirety with logging turned to silly.  Is there somewhere else or another method to obtain something that may be missing?

bepsoccer, 2022-05-30:
> Uhhh this is embarrassing... Somehow I missed the scrollbar next to the log...

No worries, happens to the best of us.

AlCalzone, 2022-05-31:
Ok, so the log above is one where the heal goes through. Do you have a log where it doesn't?

> it finishes the heal as successful and assigns no hops(direct to Node 1)

Where are you taking this information from? The heal log doesn't include this.

bepsoccer, 2022-06-01:
> Ok, so the log above is one where the heal goes through. Do you have a log where it doesn't?
> 
> > ... assigns no hops(direct to Node 1)
> 
> Where are you taking this information from? The heal log doesn't include this.

This comes from the UI debug console.  I realize it says it healed successfully, but it doesn't make sense because it's on the far edge of the network and always fails the network health check with 0/10 scores.  It should be finding a neighbor to use.  I have some devices much closer to the controller that have a hop.  This is a ZEN71.  I mention it because I was going to open another issue for my large fleet of ZEN76 and ZEN71s failing to interview fully, but I see the issues/PRs already exist for that problem.

AlCalzone, 2022-06-08:
> always fails the network health check with 0/10 scores

Do you have a driver log of that? When the stick tells us it successfully assigned the node a new route, we'll have to trust that information.

closed 2022-06-19 by zwave-js-assistant[bot]