zwave-js/zwave-js #4657: I'm getting compilation errors in the zwave-js library..

iklein99, 2022-05-27:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Getting compilation errors after trying to upgrade to the latest version of the library.
What do you observe?
Here are the errors:
node_modules/zwave-js/build/lib/controller/MockControllerBehaviors.d.ts:1:45 - error TS2307: Cannot find module '@zwave-js/testing' or its corresponding type declarations.

1 import type { MockControllerBehavior } from "@zwave-js/testing";
                                              ~~~~~~~~~~~~~~~~~~~

node_modules/zwave-js/build/lib/driver/Driver.d.ts:76:22 - error TS2420: Class 'Driver' incorrectly implements interface 'ZWaveHost'.
  Type 'Driver' is missing the following properties from type 'ZWaveHost': controllerLog, securityManager, securityManager2, ownNodeId, and 3 more.

76 export declare class Driver extends TypedEventEmitter<DriverEventCallbacks> implements ZWaveHost {
                        ~~~~~~

Found 2 errors in 2 files.

### Device information

NA

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [X] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js 9.3.0

### Did you change anything?

no

### Did this work before?

Yes (please describe)

### If yes, where did it work?

Was working well until I tried to update the library.  

### Attach Driver Logfile

N/A

AlCalzone, 2022-05-30:
You'll have to enable `skipLibCheck` for now in `tsconfig.json`. I'll look into fixing the errors when I'm done with the current refactor.

iklein99, 2022-05-31:
Thanks, Al!

AlCalzone, 2025-03-12:
I don't think there's an easy way to avoid `skipLibCheck: true`. Given that even the `@tsconfig/nodeXX` packages set this by default, I think it's fair to keep this as a requirement.

closed 2025-03-12 by AlCalzone