zwave-js/zwave-js #4659: [Bug]: Zooz Zen72  with v10.0 fails inclusion

laheym, 2022-05-30:


AlCalzone, 2022-05-30:
I'm not seeing any reference to that node in the log you posted. Please make another driver log, re-interview the node and then post the log. I don't need the zwavejs2mqtt logs.

laheym, 2022-05-31:
I've subsequently tried to include many additional nodes.  It's seems that the issue may be with the 10.0 firmware from Zooz.  v1.3 on zen77 and zen76s work the first try, but after upgrading to v10, I get the same failed interviews.

with the zen72, etc on v10 firmware, I am eventually able a successful interview the.  I am able to do this by completing the following in various order, sometimes multiple times each: restart of the container, soft reset the controller and reinterview the node. I have not yet isolated a sequence that will reliably interview the noded

laheym, 2022-05-31:
> I'm not seeing any reference to that node in the log you posted. Please make another driver log, re-interview the node and then post the log. I don't need the zwavejs2mqtt logs.

it's Node 28 in the original logs.  I will repeat on another zen72 and give you that log,

laheym, 2022-05-31:
I am including two nodes that fail interviews.  The first (Node37) is a Zen 77 that I successfully included with v1.3 firmware on the first attempt, but when I upgraded the firmware to 10.10 it attempted to reinterview and failed.  I have repeatedly tried to reinterview, reset the container, reset the controller, but it is stubbornly stuck at nodeinfo.   the second (node38) is a zen 72 that came from the factory with firmware 10.0.  Like other v10 zen72s that I have tried to include it fails the interview.

[image]
[zwavejs2mqtt-store (4).zip](https://github.com/zwave-js/node-zwave-js/files/8802438/zwavejs2mqtt-store.4.zip)

AlCalzone, 2022-05-31:
Ok it looks like this is caused by the same issue that https://github.com/zwave-js/node-zwave-js/pull/4595 works around. https://github.com/zwave-js/node-zwave-js/issues/4585#issuecomment-1121140769 has an explanation.

So, this workaround needs to be applied for Zen72, 76 and 77 too, did I forget anything?

laheym, 2022-05-31:
those are the three devices that I have been with this issue, thanks!

kaechele, 2022-05-31:
There is a newer version of the firmware available. Maybe this issue is fixed there.
Oddly, the version number is 2.0 (released in May 2022, whereas 10.0 was released in June 2021).
See: https://www.support.getzooz.com/kb/article/849-zen72-700-series-dimmer-change-log/

PR #4666 (2022-06-07): fix(config): work around Configuration Info reporting bug in ZEN71/72/76/77
fixes: #4659
closed 2022-06-07 by AlCalzone