zwave-js/zwave-js #4660: [Bug]: zwavejs2mqtt does not reconnect if the TCP (ser2net) serial port was drop

dick902, 2022-05-30:
### Checklist

- [X]  I am not using Home Assistant. Or: a developer has told me to come here.
- [X] I have checked the troubleshooting section and my problem is not described there.
- [X] I have read the changelog and my problem is not mentioned there.

### Deploy method

Docker

### Zwavejs2Mqtt version

6.10.0

### ZwaveJS version

9.3.0

### Describe the bug

This is a similar issues as zwave-js/zwavejs2mqtt#1246.
Once the TCP serial connection drop, it will nerve reconnect until manual restart zwavejs2mqtt or press the SAVE button in the setting menu (another method to restart) also make the TCP serial port reconnect.

I had tested with  v4.3.1 and found the issues was resolved before.
But it is happening again on v5.x.x and v6.x.x of zwavejs2mqtt again. 

### To Reproduce

1. Connect the Z-Wave dongle via ser2net
2. Stop ser2net
3. Wait for the lost connection in zwavejs2mqtt
4. Restart ser2net

### Expected behavior

After 4., the TCP serial port connection should be auto-restored.

kvs, 2022-10-09:
I'm seeing the same issue - everything works fine, ser2net machine restarts properly but zwavejs never notices it was gone so it never receives anything from the zwave-stick again. Maybe a periodic status-poll from the stick would be enough to trigger a tcp-closed and reconnect (ie. a keepalive function)?

robertsLando, 2022-10-10:
We need to wait for @AlCalzone to come back from his parental leave in order to fix this

AlCalzone, 2022-12-09:
👉🏻 https://github.com/zwave-js/node-zwave-js/issues/5129

chrish987, 2023-02-19:
is there any ETA on this one being looked at?

PR #5715 (2023-04-27): fix: detect TCP disconnects
fixes: #4660
closed 2023-04-27 by zwave-js-bot