zwave-js/zwave-js #4664: Kwikset Lock Manufacturer and Model show as <unknown>

rpruden, 2022-05-30:
I hope this is the correct place. I have a Kwikset lock (I believe it’s a 912). It pairs successfully to Home Assistant via Zwave JS but, the device manufacturer and model shows up as <unknown> and doesn’t appear to receive any logging back from the lock. Due to this I can not get notifications when the lock is operated manually. I CAN perform basic lock and unlock commands. It seems to me that it’s just not finding an exact match in the database so it doesn’t know how to handle the device. Here is a copy of the Zwave logs when re-interviewing the device. Can you tell why it’s not able to properly determine the model? Hopefully this is something that can be fixed.

`Kwikset Z-Wave Interview

2022-05-30T20:20:36.001Z CNTRLR   [Node 015] Beginning interview - last completed stage: None
2022-05-30T20:20:36.003Z CNTRLR   [Node 015] new node, doing a full interview...
2022-05-30T20:20:36.004Z CNTRLR » [Node 015] querying protocol info...
2022-05-30T20:20:36.044Z CNTRLR « [Node 015] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Secure Keypad Door Lock
                                  node type:             End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2022-05-30T20:20:36.051Z CNTRLR   [Node 015] Interview stage completed: ProtocolInfo
2022-05-30T20:20:36.053Z CNTRLR » [Node 015] querying node info...
2022-05-30T20:20:37.296Z CNTRLR « [Node 015] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Manufacturer Specific
                                  · Device Reset Locally
                                  · Security
                                  · Powerlevel
                                  · Firmware Update Meta Data
                                  controlled CCs:
2022-05-30T20:20:37.300Z CNTRLR   [Node 015] Interview stage completed: NodeInfo
2022-05-30T20:20:37.304Z CNTRLR » [Node 015] Querying securely supported commands (S0)...
2022-05-30T20:20:37.490Z CNTRLR « [Node 015] received secure commands (S0)
                                  supported CCs:
                                  · Version
                                  · Battery
                                  · Door Lock
                                  · User Code
                                  · Association
                                  · Association Group Information
                                  · Notification
                                  · Configuration
                                  · Schedule Entry Lock
                                  · Time Parameters
                                  · Door Lock Logging
                                  · Anti-Theft
                                  controlled CCs:
2022-05-30T20:20:37.495Z CNTRLR   [Node 015] Interviewing Manufacturer Specific...
2022-05-30T20:20:37.496Z CNTRLR » [Node 015] querying manufacturer information...
2022-05-30T20:20:37.553Z CNTRLR « [Node 015] received response for manufacturer information:
                                    manufacturer: Spectrum Brands (0x90)
                                    product type: 0x03
                                    product id:   0x0436
2022-05-30T20:20:37.558Z CNTRLR   [Node 015] Interviewing Version...
2022-05-30T20:20:37.559Z CNTRLR » [Node 015]   querying the CC version for Version...
2022-05-30T20:20:37.724Z CNTRLR   [Node 015]   supports CC Version (0x86) in version 2
2022-05-30T20:20:37.725Z CNTRLR » [Node 015] querying node versions...
2022-05-30T20:20:37.884Z CNTRLR « [Node 015] received response for node versions:
                                    library type:      Enhanced Slave (0x03)
                                    protocol version:  4.34
                                    firmware versions: 4.10, 54.13
                                    hardware version:  4
2022-05-30T20:20:37.887Z CNTRLR » [Node 015] querying CC versions...
2022-05-30T20:20:37.889Z CNTRLR » [Node 015]   querying the CC version for Basic...
2022-05-30T20:20:38.056Z CNTRLR   [Node 015]   does NOT support CC Basic (0x20)
2022-05-30T20:20:38.057Z CNTRLR » [Node 015]   querying the CC version for Door Lock...
2022-05-30T20:20:38.219Z CNTRLR   [Node 015]   supports CC Door Lock (0x62) in version 2
2022-05-30T20:20:38.220Z CNTRLR » [Node 015]   querying the CC version for User Code...
2022-05-30T20:20:38.358Z CNTRLR   [Node 015]   supports CC User Code (0x63) in version 1
2022-05-30T20:20:38.359Z CNTRLR » [Node 015]   querying the CC version for Manufacturer Specific...
[... 94 lines cut ...]
2022-05-30T20:20:42.069Z CNTRLR » [Node 015] querying supported notification types...
2022-05-30T20:20:42.223Z CNTRLR « [Node 015] received supported notification types:
                                  · Access Control
                                  · Home Security
                                  · Power Management
2022-05-30T20:20:42.225Z CNTRLR » [Node 015] querying supported notification events for Access Control...
2022-05-30T20:20:42.386Z CNTRLR « [Node 015] received supported notification events for Access Control: 1, 2, 3,
                                   4, 5, 6, 8, 9, 10, 11, 12, 13, 14, 15, 19, 20, 21
2022-05-30T20:20:42.387Z CNTRLR » [Node 015] querying supported notification events for Home Security...
2022-05-30T20:20:42.541Z CNTRLR « [Node 015] received supported notification events for Home Security: 3
2022-05-30T20:20:42.543Z CNTRLR » [Node 015] querying supported notification events for Power Management...
2022-05-30T20:20:42.702Z CNTRLR « [Node 015] received supported notification events for Power Management: 10, 11
2022-05-30T20:20:42.707Z CNTRLR » [Node 015] enabling notifications for Access Control...
2022-05-30T20:20:42.804Z CNTRLR » [Node 015] enabling notifications for Home Security...
2022-05-30T20:20:42.915Z CNTRLR » [Node 015] enabling notifications for Power Management...
2022-05-30T20:20:43.005Z CNTRLR   [Node 015] Interviewing Configuration...
2022-05-30T20:20:43.007Z CNTRLR   [Node 015] ConfigurationCC: skipping interview because CC version is < 3 and t
                                  here is no config file
2022-05-30T20:20:43.011Z CNTRLR   [Node 015] Interviewing Door Lock Logging...
2022-05-30T20:20:43.013Z CNTRLR » [Node 015] querying supported number of records...
2022-05-30T20:20:43.189Z CNTRLR « [Node 015] supports 10 records
2022-05-30T20:20:43.192Z CNTRLR   [Node 015] Interview stage completed: CommandClasses
2022-05-30T20:20:43.195Z CNTRLR   [Node 015] Interview stage completed: OverwriteConfig
2022-05-30T20:20:43.196Z CNTRLR   [Node 015] Interview completed
2022-05-30T20:20:43.198Z CNTRLR   [Node 015] The node is ready to be used `

AlCalzone, 2022-05-31:
We simply don't have the identifier for this one in the DB. Easy fix.
@zwave-js-bot add fingerprint `0x0003:0x0436` to `0x0090:0x0003:0x0336`

rpruden, 2022-05-31:
> We simply don't have the identifier for this one in the DB. Easy fix.

so does this mean it will end up getting pushed in a future update?

AlCalzone, 2022-05-31:
yep, but you'll likely have to re-interview the device for it to get picked up.

rpruden, 2022-06-18:
I have another one of these missing from the database. It’s a model Smartcode 913 though (should be essentially the same as a 912 or 914 I think. Here is the interview log. Also, the first one that you added never ended up showing up (it also might be a 913 instead of a 912). How long will it take for those previous commits to go live in the database?

Subscribed to Z-Wave JS Log Messages…
2022-06-18T01:59:55.584Z CNTRLR   [Node 019] Beginning interview - last completed stage: None
2022-06-18T01:59:55.588Z CNTRLR   [Node 019] new node, doing a full interview...
2022-06-18T01:59:55.590Z CNTRLR » [Node 019] querying protocol info...
2022-06-18T01:59:55.666Z CNTRLR « [Node 019] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Secure Keypad Door Lock
                                  node type:             End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2022-06-18T01:59:55.671Z CNTRLR   [Node 019] Interview stage completed: ProtocolInfo
2022-06-18T01:59:55.674Z CNTRLR » [Node 019] querying node info...
2022-06-18T01:59:56.932Z CNTRLR « [Node 019] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Manufacturer Specific
                                  · Security
                                  controlled CCs:
2022-06-18T01:59:56.937Z CNTRLR   [Node 019] Interview stage completed: NodeInfo
2022-06-18T01:59:56.941Z CNTRLR » [Node 019] Querying securely supported commands (S0)...
2022-06-18T01:59:57.140Z CNTRLR « [Node 019] received secure commands (S0)
                                  supported CCs:
                                  · Version
                                  · Anti-Theft
                                  controlled CCs:
2022-06-18T01:59:57.146Z CNTRLR   [Node 019] Interviewing Manufacturer Specific...
2022-06-18T01:59:57.147Z CNTRLR » [Node 019] querying manufacturer information...
2022-06-18T01:59:57.208Z CNTRLR « [Node 019] received response for manufacturer information:
                                    manufacturer: Spectrum Brands (0x90)
                                    product type: 0x03
                                    product id:   0x0845
2022-06-18T01:59:57.210Z CNTRLR   [Node 019] Interviewing Version...
2022-06-18T01:59:57.211Z CNTRLR » [Node 019]   querying the CC version for Version...
2022-06-18T01:59:57.414Z CNTRLR   [Node 019]   supports CC Version (0x86) in version 2
2022-06-18T01:59:57.416Z CNTRLR » [Node 019] querying node versions...
2022-06-18T01:59:57.574Z CNTRLR « [Node 019] received response for node versions:
                                    library type:      Enhanced Slave (0x03)
                                    protocol version:  4.34
                                    firmware versions: 4.10, 69.3
                                    hardware version:  8
2022-06-18T01:59:57.576Z CNTRLR » [Node 019] querying CC versions...
2022-06-18T01:59:57.578Z CNTRLR » [Node 019]   querying the CC version for Basic...
2022-06-18T01:59:57.753Z CNTRLR   [Node 019]   does NOT support CC Basic (0x20)
2022-06-18T01:59:57.755Z CNTRLR » [Node 019]   querying the CC version for Door Lock...
2022-06-18T01:59:57.895Z CNTRLR   [Node 019]   does NOT support CC Door Lock (0x62)
2022-06-18T01:59:57.896Z CNTRLR » [Node 019]   querying the CC version for User Code...
2022-06-18T01:59:58.050Z CNTRLR   [Node 019]   does NOT support CC User Code (0x63)
2022-06-18T01:59:58.052Z CNTRLR » [Node 019]   querying the CC version for Manufacturer Specific...
2022-06-18T01:59:58.221Z CNTRLR   [Node 019]   supports CC Manufacturer Specific (0x72) in version 2
2022-06-18T01:59:58.223Z CNTRLR » [Node 019]   querying the CC version for Z-Wave Plus Info...
2022-06-18T01:59:58.391Z CNTRLR   [Node 019]   supports CC Z-Wave Plus Info (0x5e) in version 2
2022-06-18T01:59:58.393Z CNTRLR   [Node 019]   skipping query for Security (0x98) because max implemented versio
                                  n is 1
2022-06-18T01:59:58.395Z CNTRLR   [Node 019]   skipping query for Anti-Theft (0x5d) because max implemented vers
                                  ion is 0
2022-06-18T01:59:58.398Z CNTRLR   [Node 019] No device config found
2022-06-18T01:59:58.400Z CNTRLR   [Node 019] Interviewing Z-Wave Plus Info...
2022-06-18T01:59:58.400Z CNTRLR » [Node 019] querying Z-Wave+ information...
2022-06-18T01:59:58.456Z CNTRLR « [Node 019] received response for Z-Wave+ information:
                                  Z-Wave+ version: 1
                                  role type:       SleepingListeningSlave
                                  node type:       Node
                                  installer icon:  0x0300
                                  user icon:       0x0300
2022-06-18T01:59:58.458Z CNTRLR   [Node 019] Interview stage completed: CommandClasses
2022-06-18T01:59:58.459Z CNTRLR   [Node 019] Interview stage completed: OverwriteConfig
2022-06-18T01:59:58.460Z CNTRLR   [Node 019] Interview completed
2022-06-18T01:59:58.461Z CNTRLR   [Node 019] The node is ready to be used

rpruden, 2022-06-18:
Can confirm. These are both Smartcode 913 locks. The only difference is that the 913 doesn’t natively support Z-Wave. These have been upgraded with a Z-Wave module and only have 16 User Codes instead of 30. Other than that, they should be the same as the 914. Would it be possible to get both of these models added as a 913 model? You added the first as a 912 which actually has less buttons than this one.

AlCalzone, 2022-06-18:
```
2022-06-18T01:59:57.140Z CNTRLR « [Node 019] received secure commands (S0)
supported CCs:
· Version
· Anti-Theft
```

Looks like your lock has the Anti-Theft feature enabled. Unfortunately, you can only disable it on the same software that enabled it. Which isn't Z-Wave JS.

rpruden, 2022-06-18:
> Looks like your lock has the Anti-Theft feature enabled, which means there is no meaningful functionality exposed while it is enabled. Unfortunately, you can only disable it on the same software that enabled it. And that isn't Z-Wave JS.

I’ve heard of this before. The lock is brand new, but the Z-Wave Module came from E-bay. I will contact the seller to see if he will replace it.

As for the categorization of the two locks, would it be possible to create a 913 category, or would it make sense to just lump them in with the 914? The only difference really between the two is the amount of code slots that they each have. And that the 913 doesn’t come with Z-wave natively. It needs to be upgraded with a Z-Wave module.

AlCalzone, 2022-06-18:
Depends how it is identified. Unfortunately that functionality  is also locked behind the anti theft feature.

rpruden, 2022-06-18:
> Depends how it is identified. Unfortunately that functionality is also locked behind the anti theft feature.

Interesting. I appreciate your help and insight on this. I just removed both locks from my config swapped their Z-Wave modules and then re-added them and I am still getting the “Anti-Theft” feature on the new 913 lock. Just for clarification this lock was brand new in the box retail Kwikset 913. The Z-Wave Modules were purchased on e-bay. The original first lock came with the house and I don’t know the exact model but it looks the same as my 913. Is there any other information that I could gather that might help?

Here is the latest Interview;

```
2022-06-18T19:31:17.221Z CNTRLR   [Node 023] Beginning interview - last completed stage: None
2022-06-18T19:31:17.226Z CNTRLR   [Node 023] new node, doing a full interview...
2022-06-18T19:31:17.232Z CNTRLR » [Node 023] querying protocol info...
2022-06-18T19:31:17.273Z CNTRLR « [Node 023] received response for protocol info:
                                  basic device class:    Routing Slave
                                  generic device class:  Entry Control
                                  specific device class: Secure Keypad Door Lock
                                  node type:             End Node
                                  is always listening:   false
                                  is frequent listening: 1000ms
                                  can route messages:    true
                                  supports security:     false
                                  supports beaming:      true
                                  maximum data rate:     100000 kbps
                                  protocol version:      3
2022-06-18T19:31:17.278Z CNTRLR   [Node 023] Interview stage completed: ProtocolInfo
2022-06-18T19:31:17.279Z CNTRLR » [Node 023] querying node info...
2022-06-18T19:31:18.561Z CNTRLR « [Node 023] node info received
                                  supported CCs:
                                  · Z-Wave Plus Info
                                  · Manufacturer Specific
                                  · Security
                                  controlled CCs:
2022-06-18T19:31:18.567Z CNTRLR   [Node 023] Interview stage completed: NodeInfo
2022-06-18T19:31:18.570Z CNTRLR » [Node 023] Querying securely supported commands (S0)...
2022-06-18T19:31:18.758Z CNTRLR « [Node 023] received secure commands (S0)
                                  supported CCs:
                                  · Version
                                  · Anti-Theft
                                  controlled CCs:
2022-06-18T19:31:18.761Z CNTRLR   [Node 023] Interviewing Manufacturer Specific...
2022-06-18T19:31:18.762Z CNTRLR » [Node 023] querying manufacturer information...
2022-06-18T19:31:18.815Z CNTRLR « [Node 023] received response for manufacturer information:
                                    manufacturer: Spectrum Brands (0x90)
                                    product type: 0x03
                                    product id:   0x0845
2022-06-18T19:31:18.817Z CNTRLR   [Node 023] Interviewing Version...
2022-06-18T19:31:18.818Z CNTRLR » [Node 023]   querying the CC version for Version...
2022-06-18T19:31:18.967Z CNTRLR   [Node 023]   supports CC Version (0x86) in version 2
2022-06-18T19:31:18.970Z CNTRLR » [Node 023] querying node versions...
2022-06-18T19:31:19.135Z CNTRLR « [Node 023] received response for node versions:
                                    library type:      Enhanced Slave (0x03)
                                    protocol version:  4.34
                                    firmware versions: 4.10, 69.3
                                    hardware version:  8
2022-06-18T19:31:19.136Z CNTRLR » [Node 023] querying CC versions...
2022-06-18T19:31:19.137Z CNTRLR » [Node 023]   querying the CC version for Basic...
2022-06-18T19:31:19.298Z CNTRLR   [Node 023]   does NOT support CC Basic (0x20)
2022-06-18T19:31:19.299Z CNTRLR » [Node 023]   querying the CC version for Door Lock...
2022-06-18T19:31:19.512Z CNTRLR   [Node 023]   does NOT support CC Door Lock (0x62)
2022-06-18T19:31:19.516Z CNTRLR » [Node 023]   querying the CC version for User Code...
2022-06-18T19:31:19.664Z CNTRLR   [Node 023]   does NOT support CC User Code (0x63)
2022-06-18T19:31:19.666Z CNTRLR » [Node 023]   querying the CC version for Manufacturer Specific...
2022-06-18T19:31:19.823Z CNTRLR   [Node 023]   supports CC Manufacturer Specific (0x72) in version 2
2022-06-18T19:31:19.825Z CNTRLR » [Node 023]   querying the CC version for Z-Wave Plus Info...
2022-06-18T19:31:19.975Z CNTRLR   [Node 023]   supports CC Z-Wave Plus Info (0x5e) in version 2
2022-06-18T19:31:19.977Z CNTRLR   [Node 023]   skipping query for Security (0x98) because max implemented versio
                                  n is 1
2022-06-18T19:31:19.978Z CNTRLR   [Node 023]   skipping query for Anti-Theft (0x5d) because max implemented vers
                                  ion is 0
2022-06-18T19:31:19.989Z CNTRLR   [Node 023] No device config found
2022-06-18T19:31:19.992Z CNTRLR   [Node 023] Interviewing Z-Wave Plus Info...
2022-06-18T19:31:19.994Z CNTRLR » [Node 023] querying Z-Wave+ information...
[... 177 lines cut ...]
2022-06-18T19:42:56.254Z CNTRLR « [Node 024] received supported notification types:
                                  · Access Control
                                  · Home Security
                                  · Power Management
[... cut ...]
2022-06-18T19:42:57.027Z CNTRLR   [Node 024] Interviewing Door Lock Logging...
2022-06-18T19:42:57.029Z CNTRLR » [Node 024] querying supported number of records...
2022-06-18T19:42:57.208Z CNTRLR « [Node 024] supports 10 records
2022-06-18T19:42:57.216Z CNTRLR   [Node 024] Interview stage completed: CommandClasses
2022-06-18T19:42:57.220Z CNTRLR   [Node 024] Interview stage completed: OverwriteConfig
2022-06-18T19:42:57.222Z CNTRLR   [Node 024] Interview completed
2022-06-18T19:42:57.223Z CNTRLR   [Node 024] The node is ready to be used
```

AlCalzone, 2022-06-19:
Ok I was mistaken. The first one does identify with the following info:
```
manufacturer: Spectrum Brands (0x90)
product type: 0x03
product id:   0x0845
```
which we don't have in our database yet. We can create an entry (this is the 913?), but we'll need to know some more details about configuration parameters if possible. Do you have a manual for it?
Then open a new issue with the missing configuration template and provide the information you have:
https://github.com/zwave-js/node-zwave-js/issues/new?assignees=&labels=config+%E2%9A%99&template=device_config.yml&title=Missing+device+configuration%3A+%3Center+manufacturer+and+device+name%3E

Note that this won't help you operate it though, since it is still in Anti-Theft mode.

PR #4667 (2022-05-31): fix(config): add fingerprint `0x0003:0x0436` to "Kwikset 912"
fixes: #4664