zwave-js/zwave-js #4669: ZEN71 repeated interview failures

RyanWor, 2022-05-31:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Attempting to interview a Zooz ZEN71 switch securely included in an existing network

What do you observe?
Interview is never able to complete successfully

What did you expect to happen?
Interview process completes and device is ready for use

Steps to reproduce the behavior:
  1. Factory reset ZEN71 switch
  2. Exclude ZEN71 switch
  3. Include ZEN71 switch securely (S2_Authenticated,S2_Unauthenticated classes) via manual inclusion and entering DSK PIN.
  4. Wait until failed interview completes
  5. Move controller as close to switch as possible for reinterview attempt (this has significantly helped many other nodes that failed to interview in my network)
  6. Re-interview node (interview again fails, SPAN errors in logs)

### Device information

Manufacturer: Zooz
Model name: ZEN71
Node ID in your network: 44

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 6.11.0
`zwavejs2mqtt` branch: latest

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

Excerpts from logs of two different attempts to interview this device
[failed_interview_zwavejs_log2.txt](https://github.com/zwave-js/node-zwave-js/files/8808388/failed_interview_zwavejs_log2.txt)
[failed_interview_zwavejs_log.txt](https://github.com/zwave-js/node-zwave-js/files/8808389/failed_interview_zwavejs_log.txt)

Additionally have excerpt from zniffer logs as well from one attempt if that would be helpful please let me know and if those are okay to upload publicly or should be sent privately.

RyanWor, 2022-05-31:
Update: I tested the fix that is in the works for ZEN72/ZEN76/ZEN77 adding the same compat section in #4666 for ZEN71.json and after restarting my device interviewed almost instantly.

AlCalzone, 2022-05-31:
Cool, I'll amend the PR tomorrow

AlCalzone, 2022-06-07:
fixed by https://github.com/zwave-js/node-zwave-js/pull/4666

closed 2022-06-07 by AlCalzone