zwave-js/zwave-js #4688: Device (Heatit thermostat) doesn't report binary switch as it should with updated firmware

Fr00t, 2022-06-04:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

YES, BUT a ZWaveJS2MQTT developer has told me to come here

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Unclear. Maybe the device was set up using documentation of a previous firmware version of the device that didn't support the binary switch?

What do you observe?
The thermostat reports setpoints as expected, but the new firmware is also capable of reporting wether or not the relay is open which is the feature I want. 

What did you expect to happen?
That binary switch for heating status is correctly reported

Steps to reproduce the behavior:
I've tried re-interviewing the nodes multiple time. I've got two identical thermostats. They reported correctly using Homeseers implementation so I know it's actually possible. 

I turn up the setpoint above current temperature, see the red LED-indicator light up, and look for changes in the events tab, but all I get is the new setpoint. 

I have attached the documentation of this firmware version if that could be of any use. 

[HeatIt_1v9.pdf](https://github.com/zwave-js/node-zwave-js/files/8837771/HeatIt_1v9.pdf)

### Device information

Manufacturer: HeatIt
Model name: Z-trm2fx
Firmware: 1.92
Node ID in your network: 96

It's added as "Thermofloor Thermostat TF016 / TF021" in Zwavejs2mqtt

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### If yes, where did it work?

It worked in Homeseer before I migrated to zwavejs2mqtt and Homeassistant

### Attach Driver Logfile

[zwavejs_2022-06-04.log](https://github.com/zwave-js/node-zwave-js/files/8837779/zwavejs_2022-06-04.log)

Fr00t, 2022-06-04:
It seems to actually report correctly when using the "Refresh" button in zwavejs2mqtt. I'm puzzled by why it wouldn't update automatically?

Are my groups/associations off?

AlCalzone, 2022-06-04:
If the new firmware changed capabilities, you might need to exclude and re-include the device.

I won't be able to look at this before Monday, so please make sure to capture the inclusion and/or an interview in the driver log. I'm on mobile so I Can't verify that the posted log is sufficient right now.

Fr00t, 2022-06-04:
> If the new firmware changed capabilities, you might need to exclude and re-include the device.
>
> I won't be able to look at this before Monday, so please make sure to capture the inclusion and/or an [...]

Thanks, but it was always at 1.92 when I added it to zwavejs

Fr00t, 2022-06-04:
It seems like configuration (parameters) introduced in this firmware isn't supported either. Parameter 14-18 in the documentation linked in OP

Fr00t, 2022-06-04:
> It seems like configuration (parameters) introduced in this firmware isn't supported either. Parameter 14-18 in the documentation linked in OP

As for this issue I think the solution should be fairly simple. I've forked and created a [commit here](https://github.com/Fr00t/node-zwave-js/commit/c0d9a93cf6d363ab0dba6bd543fec2b11299df40). I do need some guidance on how I can test it though? How can I move that device-json into my zwavejs2mqtt container to give it a go?

AlCalzone, 2022-06-04:
There is documentation on testing device files in the zwavejs2mqtt repo.

Fr00t, 2022-06-04:
> There is documentation on testing device files in the zwavejs2mqtt repo.

Yea I just found it and that makes testing significantly easier :) I imagine with FW >= 1.9 it's possible to drop `disableBasicMapping`.  

I'm still unable to get it to update the binary switch current value by itself, I have to manually trigger the refresh and then I get the correct information immediately. Any ideas about this?

Attached is a log file in which I at `2022-06-04T18:56:33` set a new lower setpoint which turns the relay off and then at `2022-06-04T18:56:51` I guery the basic switch and see that it's turned off. 
[zwavejs_2022-06-04 (1).log](https://github.com/zwave-js/node-zwave-js/files/8838489/zwavejs_2022-06-04.1.log)

Is there an assiciation group or endpoint I'm missing? If so, where would I start to look for it?

[image]

AlCalzone, 2022-06-07:
```
2022-06-04T18:56:38.043Z DRIVER   Dropping message because it could not be deserialized: BinarySwitchCCSet: dese
                                  rialization not implemented (ZW0320)
```
The thermostat is using a `BinarySwitchCCSet` which is meant to control a binary switch and not to report status. See manual:
[image]

According to the manual, there is another association group on Endpoint 1, which uses Basic Report commands to do the same. This sounds like a better fit:
[image]

I'm not sure how this one is labeled though, since we don't have any associations defined for the endpoints in the config file.

You do seem to have a few duplicated associations though. The controller doesn't have any endpoints, so specifying a different target endpoint than "None" (or 0 for some lifelines) just results in duplicate reports.

AlCalzone, 2022-06-08:
I'm still not seeing any unsolicited Basic reports from the thermostat, only the Binary Switch Set commands.

You'll have to ask the Heat It support how the thermostat needs to be configured so it sends the Basic Reports mentioned in the device manual. They should be able to tell you.

closed 2023-02-17 by zwave-js-assistant[bot]

nesheimt, 2024-10-24:
Official information about the Heatit TRM2 TF021 v1.92 can be found here:
https://github.com/Heatit-Controls/no.thermofloor/tree/main/drivers/TF_Thermostat_FW1.92

I do believe that Heatit TRM2 TF021 with firmware versions less that v1.92 are behaving different and having different configurations i the drivers. Most like I think that the v1.92 should have it's own device.

AlCalzone, 2024-10-25:
Do you have an actual manual for that firmware?

nesheimt, 2024-10-25:
Yes, I got this from the manufacturer this morning when asking for v1.92 documentation(even though it says v1.9 on the front page)

[Appendix_Firmware_update_1_9_Heatit_Z-Wave.pdf](https://github.com/user-attachments/files/17517269/Appendix_Firmware_update_1_9_Heatit_Z-Wave.pdf)

nesheimt, 2024-10-28:
The manufacturer implemented the relay-status going from the thermostat to the controller using association Groups 2. According to the datasheet they used "Binary Switch Set" to do this, and this seems to be so according to your observations. However, this relay-status message should have been implemented using the "Basic Reports" (according to the z-wave standard?).

I have asked the manufacturer to confirm what they have done...but I believe that they probably have (wrongly) used the "Binary Switch Set" command for this, also, as you observe only unsolicited "Binary Switch Set" commming from the thermostat.

Would the fix here be to make a workaround? I will ask the manufacturer to fix this i a new FW version but I do not have too high hopes for that. Another problem with that is that this thermostat is not OTA-upgradable, and this USB-dongle is required.

nesheimt, 2024-10-28:
For what it's worth, here is a Zniffer-file the TRM2v1.92 starting at setpoint 20, going to 25(turning on the relay) and back to 20(turning off the relay). Association group settings are:
[image]

[Heatit-TRM2v192.zip](https://github.com/user-attachments/files/17544683/Heatit-TRM2v192.zip)

nesheimt, 2024-11-01:
Here is the an allegendly working config file from Open-Zwave: https://github.com/OpenZWave/open-zwave/blob/master/config/thermofloor/heatit021-v1.92.xml