zwave-js/zwave-js #4690: General Level of Inconsistency in Z-Wave Command Functionality.

RJ-Make, 2022-06-06:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

I'm noticing a general inconsistency with commands succeeding.  It's not any one device, or area of the house and it's quite random. I have an 84 Node Z-Wave Network, comprising of many powered devices in a 2 level 1200 sq ft house. So considering the size of my house and the number of powered z-wave devices (both 500 and 700 series) I should have a very strong mesh. 

1) Latest FW for Controller.
2) Controller on a 3' Ext. Cable
3) Connected to a USB2 Powered Hub
4) Controller is almost centered in the middle of the house

I'm coming from HomeSeer (Z-NET) (~7 Years) so that is my comparative. While that setup's inclusion and exclusion system was extremely temperamental, the system seemed more stable. I would get the occasional failed command, but it was a once or twice a months thing, not an every other day thing. 

Not sure if this is a normal failure rate?
[image]

### Device information

The controller is a Zooz ZST 10 700

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
Server Version: 1.17.0 
zwavejs2mqtt: 6.11.0
zwave-js: 9.3.0

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs.zip](https://github.com/zwave-js/node-zwave-js/files/8847795/zwavejs.zip)

AlCalzone, 2022-06-07:
While I can't say exactly what is going on, I have a pretty good idea:
1. You have a few nodes with bad connectivity, which take extraordinarily long to reach. There are several instances in the log where the controller gives up after 10-20 seconds.
In some instances the controller tries really hard to reach a device that it was just sending to sleep. This is dumb and I've opened https://github.com/zwave-js/node-zwave-js/issues/4693 to improve on this.

2. You have lots of nodes that report a lot of power meter data. There are a few stretches where the controller is getting reports roughly every 20-50ms. With this amount of traffic on the network, sending commands will absolutely take longer and be less reliable.
While individual nodes seem to have a reporting frequency of several minutes, they send data in bursts and the mass of devices adds up. There are several factors that make this worse:
a) Some devices repeat themselves, e.g.:
```
2022-06-06T15:15:42.959Z SERIAL « 0x011700a80001620e320221640000000900780000000900c4e4                (25 bytes)
2022-06-06T15:15:42.960Z CNTRLR   [Node 098] [~] [Meter] value[65537]: 0.009 => 0.009               [Endpoint 0]
2022-06-06T15:15:42.961Z SERIAL » [ACK]                                                                   (0x06)
2022-06-06T15:15:42.962Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -60 dBm
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       0.009
                                      time delta:  120 seconds
                                      prev. value: 0.009
2022-06-06T15:15:43.125Z SERIAL « 0x011700a80001620e320221640000000900780000000900c3e3                (25 bytes)
2022-06-06T15:15:43.126Z CNTRLR   [Node 098] [~] [Meter] value[65537]: 0.009 => 0.009               [Endpoint 0]
2022-06-06T15:15:43.126Z SERIAL » [ACK]                                                                   (0x06)
2022-06-06T15:15:43.127Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -61 dBm
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       0.009
                                      time delta:  120 seconds
                                      prev. value: 0.009
2022-06-06T15:15:43.746Z SERIAL « 0x011700a80001620e320221640000000900780000000900c4e4                (25 bytes)
2022-06-06T15:15:43.747Z CNTRLR   [Node 098] [~] [Meter] value[65537]: 0.009 => 0.009               [Endpoint 0]
2022-06-06T15:15:43.748Z SERIAL » [ACK]                                                                   (0x06)
2022-06-06T15:15:43.748Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -60 dBm
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       0.009
                                      time delta:  120 seconds
                                      prev. value: 0.009
2022-06-06T15:15:43.766Z SERIAL « 0x011700a80001620e320221640000000900780000000900c3e3                (25 bytes)
2022-06-06T15:15:43.766Z CNTRLR   [Node 098] [~] [Meter] value[65537]: 0.009 => 0.009               [Endpoint 0]
2022-06-06T15:15:43.767Z SERIAL » [ACK]                                                                   (0x06)
2022-06-06T15:15:43.767Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -61 dBm
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       0.009
                                      time delta:  120 seconds
                                      prev. value: 0.009
```
This might be an indication of a lost acknowledgement, so the devices re-transmit to make sure the reports reach their target. This might also be a side effect of the high traffic, unfortunately making it worse.

b) Some devices report on a fixed schedule, even if the changes are insignificant (0.003 kWh):
```
2022-06-06T15:15:47.177Z DRIVER « [Node 101] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -91 dBm
                                  └─[MeterCCReport]
                                      type:        Electric
                                      scale:       kWh
                                      rate type:   Consumed
                                      value:       335.327
                                      time delta:  60 seconds
                                      prev. value: 335.324
[... 2 lines cut ...]

c) Some devices send the reports with Supervision encapsulation, requiring an answer to the report, even if the reports are useless because nothing changed (0W -> 0W):
```
[... cut ...]
The failures are mainly dropped (identically) repeated commands from nodes. What you should worry about are the almost 20k incoming reports.
Your first course of action should be configuring all devices to report way less. Where possible, prefer reporting on significant (!) changes over timed reports. If you need timed reports e.g. as a backup, reduce the frequency significantly (to sth. like once per hour).
Check which reports you really need. E.g. do you need to know the Voltage or is W/kWh enough? For multisensors, check if you need UV levels or if temperature/humidity is enough, etc.

RJ-Make, 2022-06-07:
Thank You for taking a look at this, I will go through, see if I can make the suggested changes and report back.

RJ-Make, 2022-06-12:
After turning off ALL automatic reporting and raising the threshold to a point of almost uselessness for all energy reporting devices the problem persists. 

[zwavejs_After_Adjustments.zip](https://github.com/zwave-js/node-zwave-js/files/8885883/zwavejs_After_Adjustments.zip)

AlCalzone, 2022-06-13:
Ok, it does look a bit better though. At least I am getting a clearer picture of whats going on.

The long delays when sending nodes to sleep are still a problem https://github.com/zwave-js/node-zwave-js/issues/4693.

Node 006 is spamming a lot, I count 38 reports during the time it took to send an S2 nonce to node 128, about a third are the same illuminance report
```
2022-06-12T10:24:25.960Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:25.961Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 0 => 1            [Endpoint 0]
2022-06-12T10:24:25.962Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:25.962Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -59 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:25.986Z SERIAL « 0x010f00a8000106063105030a000100c4a1                                (17 bytes)
2022-06-12T10:24:25.987Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:25.988Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:25.989Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:25.990Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -60 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:26.161Z SERIAL « 0x010f00a8000106063105030a000100c3a6                                (17 bytes)
2022-06-12T10:24:26.162Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:26.162Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:26.162Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:26.163Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -61 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:26.339Z SERIAL « 0x010f00a8000106063105030a000100d0b5                                (17 bytes)
2022-06-12T10:24:26.339Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:26.340Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:26.341Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:26.341Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -48 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:26.401Z SERIAL « 0x010f00a8000106063105030a000100c5a0                                (17 bytes)
2022-06-12T10:24:26.401Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:26.402Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:26.403Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:26.403Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -59 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:26.429Z SERIAL « 0x010f00a8000106063105030a000100cfaa                                (17 bytes)
2022-06-12T10:24:26.429Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:26.430Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:26.430Z SERIAL » [ACK]                                                                   (0x06)
2022-06-12T10:24:26.431Z DRIVER « [Node 006] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -49 dBm
                                  └─[MultilevelSensorCCReport]
                                      type:  Illuminance
                                      scale: Lux
                                      value: 1
2022-06-12T10:24:26.436Z SERIAL « 0x010f00a8000106063105030a000100c5a0                                (17 bytes)
2022-06-12T10:24:26.436Z CNTRLR   [Node 006] [Multilevel Sensor] Illuminance: metadata updated      [Endpoint 0]
2022-06-12T10:24:26.437Z CNTRLR   [Node 006] [~] [Multilevel Sensor] Illuminance: 1 => 1            [Endpoint 0]
2022-06-12T10:24:26.437Z SERIAL » [ACK]                                                                   (0x06)
[... cut ...]
transmit status:        NoAck, took 18560 ms
routing attempts:       21
protocol & route speed: Z-Wave, 40 kbit/s
TX channel no.:         1
route failed here:      4 -> 128
TX power:               0 dBm
measured noise floor:   0 dBm
```

---

TL;DR: Your next steps should be using zwavejs2mqtt to do health checks on every node, working outwards from the controller. If there are delays or routing issues or low signal strength, check if you need to move the node or add repeaters in between to improve the link strength.

RJ-Make, 2022-06-18:
Sorry, some how I missed your reply. :-(

I just can't image that I need more repeaters for a 2 level 1200 sq ft house.  My house measures exactly 20' x 30' (Townhouse) and as you can see has a ton of powered nodes.  I truly would have no place to add them.  

What is further puzzling is I didn't have these problems with HomeSeer and a Z-NET (Albeit on a 500 controller).  

Something must be causing these signal issues, and perhaps the 700 controller has something to do with this. 

I will work through the suggestions and report back.  

Thank you again for taking a look at this, it really is MUCH appreciated.

RJ-Make, 2022-06-18:
Interesting, Node 006 is in a closed door closet (storage underneath step) which rarely gets opened. I shut off the automatic reporting on that node to see if it helps. 

Another note: Node 128 makes perfect sense as it's inside a refrigerator, but can one low signal battery node bring down a network?

I can't seem to do any more with nodes 118, 117 to try and turn off report with Supervision encapsulation. 

Nodes 006, 118 and 117 are 10' from the controller. (117 and 118 in the same room) and 006 in the closet next to controller (2 walls).

AlCalzone, 2022-06-20:
> Node 128 makes perfect sense as it's inside a refrigerator, but can one low signal battery node bring down a network?

The stick can only send messages to one node at a time and there is a cycle of steps involved that needs to complete before the next message can be sent. If sending commands to one node can take 10-20 seconds and frequently fail altogether, this definitely has an influence on the entire network. Especially in situations like yours where:
1. the problematic node is included with S2 and the frequent errors cause the node and controller to go out of sync, requiring further communication to restore the shared state
2. the problematic node uses Supervision, requiring a response from the controller quickly, essentially making 1. much worse
3. other nodes use Supervision aswell and get impatient when they don't get the confirmation ASAP

So yes, one battery powered node with bad reception can bring down a network when it forces the controller to try and send to it frequently.

---

As for the other problematic nodes, definitely use the health check tool to try and figure out which nodes have a bad connection to the controller. The built-in algorithm for selecting routes doesn't seem to be particularly sophisticated, so a bad repeater connectivity may increase your problems aswell.

closed 2022-07-08 by zwave-js-assistant[bot]