zwave-js/zwave-js #4702: Non-responsiveness of battery powered devices, zwave communication issues, possible s2 collisions

RyanWor, 2022-06-09:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

In this instance this is an attempt to heal the network but the errors observed happen throughout the day during normal operation as well. 

What do you observe?
Inability to heal many battery powered nodes even after 24+ hours. Only 6 of 37 battery powered nodes successfully healed after 24+ hours. Nodes are attempting to communicate and waking up but possibly ZWave traffic and s2 collisions are preventing proper communication
The logs are full of tons of the following errors:
 error: Security2CC_CannotDecode 
 error: Security2CC_NoSPAN 
 error: Duplicate command 

What did you expect to happen?
Nodes to slowly heal over a long period of time. 

Steps to reproduce the behavior:
No steps needed to reproduce the errors, any usage of the network and zwave traffic seems to cause the errors. In this instance however, the seem to impact nodes ability to heal after performing a network heal (as well as healing individual nodes)

### Device information

Controller Manufacturer: Zooz
Controller Model: ZST10-700
Node ID in your network: many
35 powered nodes across cover the entire home and alone are very responsive and achieve great mesh 
37 battery powered nodes are highly unresponsive, in instances where they do work they are very delayed  (Zooz ZSE40 motion/4in1 sensors and Aeotec ZW187 contact sensors)

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 6.11.0 / 9.3.0
`zwavejs2mqtt` branch: latest

### Did you change anything?

yes (please describe)

### If yes, what did you change?

Continued slowly adding battery powered devices to the network once all powered devices were working properly and fully healed (after implementing compat fix for zen77/zen71 causing issues interviewing previously)

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwavejs_networkheal.log](https://github.com/zwave-js/node-zwave-js/files/8865774/zwavejs_networkheal.log)
Truncated log attached, logs from both today and yesterday are too large to attach. Clipped log from start of network heal at 2022-06-07T23:28:37.049Z  to approx. 21 hours later. Heal is still running 25+ hours later and failed or pending on the vast majority of battery powered devices.

RyanWor, 2022-06-09:
Thanks for taking a look @AlCalzone I appreciate your time. I am working with Zooz and Aeotec to see if supervision reports can be disabled. For timed reports I know the ZSE40s default to 86400s (24h) however I have several set to 3600s just in an attempt to get them to wake up and heal faster. All the ZW187s wake up every 3600s by default. All the nodes numbers you mentioned are ZSE40 specifically (big surprise, more issues with Zooz devices lol). I've included my configuration values for one of the ZSE40s and unless I am reading it incorrect it should only be reporting for every 1F change (10x 0.1F), 10% brightness changes, or 10% humidity changes, and of course motion. My stick is already updated to 7.17.2 and I've got it on extension and I've ran through the other general troubleshooting steps a number of times over the last few months (including testing multiple sticks, different hosts, different extensions, etc.). Honestly wondering if it would just be better if I abandoned s2/secure pairing of all my sensor devices. I would assume doing so would remove the need for them to get and wait for nonces. Personally, I am less concerned about encrypting sensor traffic as opposed to things that can actually be turned on/off.

RyanWor, 2022-06-09:
Well having a fun new issue in my attempt to start excluding and reincluding with no security. A lot of my nodes are including back at lower numbers and not working. For instance my highest node number is 98 so presumably the next node will be 99. But when I exclude a ZW187 and then re-include it it's picking up 57 as a node ID and going dead immediately before even getting through protocolinfo stage of interview. Very strange.

AlCalzone, 2022-06-10:
> For timed reports I know the ZSE40s default to 86400s (24h) however I have several set to 3600s just in an attempt to get them to wake up and heal faster.

That does not make sense. Reporting can happen while the devices are asleep. If you want to wake them up more frequently, you need to set the wakeup interval of the `Wake Up CC`, which is **not** a configuration parameter.

> I've included my configuration values for one of the ZSE40s and unless I am reading it incorrect it should only be reporting for every 1F change (10x 0.1F), 10% brightness changes, or 10% humidity changes, and of course motion.

This is possible - I haven't looked at every single report here and part of what I've seen may very well be timed reports. You could check yourself by searching for the corresponding node ID and see if the reports are spaced as you'd expect.

> Honestly wondering if it would just be better if I abandoned s2/secure pairing of all my sensor devices. I would assume doing so would remove the need for them to get and wait for nonces. 

Retrieving the S2 nonces is only necessary after a collision (both parties trying to send at the same time with the same shared state) and is normally not an issue. S0 is much worse in that regard. I really think the 10+ seconds of delay with some nodes make all the difference here.
If you cannot turn off the Supervision for sensor reports, re-including them insecurely may be an option though. Many devices start using supervision (necessary or not) when S2 is involved.

> For instance my highest node number is 98 so presumably the next node will be 99. But when I exclude a ZW187 and then re-include it it's picking up 57 as a node ID

Is it possible that you've looped through the 232 node IDs back to the beginning? Or was the node assigned 57 before exclusion? Then you might need to factory reset it after exclusion to delete the old node ID. Some devices require that.

closed 2022-06-27 by zwave-js-assistant[bot]