zwave-js/zwave-js #4703: [Bug]: Zooz ZSE44 sensor is inconsistently identified by zwavejs

laheym, 2022-06-10:


AlCalzone, 2022-06-10:
Honestly, I have no clue what the hell is going on here. The device seems to randomly not understand S2-encrypted commands.

The communication is a bit like this:
```
zwave-js -> node

-> encrypted command
<- response to encrypted command

-> next encrypted command
<- could not decode

[... establish shared secret again ...]

-> encrypted command
<- response to encrypted command

-> next encrypted command
<- could not decode 🤷🏻
```

Meanwhile some commands from the node are re-transmitted, which might indicate connectivity issues. I have my test stick plugged directly into USB3 ports on my laptop (worst case!), a test sensor is plugged into a different USB port on a different PC and yet I'm getting ACK RSSI in the order of -75 dBm. Yours are at -90 for direct range and -107 when routing through node 29 - no clue why it chooses that route. That might be a tad close to the background noise. Can you perform a health check for the node (network map, click node, click on health check button in the dialog) and post the results?

Also, it seems you're running into https://github.com/zwave-js/node-zwave-js/issues/3964 aswell. Which exact stick do you have? Wonder if that is actually the Zooz 700 series stick? AFAIK zst10-700 gets shown for all 700 series sticks.

laheym, 2022-06-11:
The stick is labed as a "Zooz s2 stick 700 zst10 700"

It's pretty weird.  Earlier today , I couldn't get a health check to even finish on Node 48 OR 49.  And when I tried other nodes (between neighbors and with the zooz stick) they would finish with 0/10 heath. I did a reset of the hypervisor (rpi) and power cycled the stick (by temporarily disconnecting it).  now heath statuses are coming in at 10/10 for both of the ZSE44 nodes.  

I'm attaching a picture of my testing setup for node 48. (normally it's mounted on the wall in a central location). node 49 is mounted in the bathroom (with door open) about 10 feet away.  AFAIK there are no other 900 mhz radios in my house.

Do you think I should try buying a new zwave stick? Do you have any other recommendations?

[image]

[image]

[image]

Botched1, 2022-06-15:
I have eight zse44, but I haven't seen this issue - likely because all of mine are paired with no security.

AlCalzone, 2022-06-15:
> they would finish with 0/10 heath

It would be interesting to know what caused this result. Latency, failures, SNR, commands not being sent at all?

> now both ZSE44 nodes heath statuses are completing with reasonable results.
> node 49 is mounted in the bathroom (with door open) about 10 feet away. AFAIK except for the zwave network there are no other 900 mhz radios in my house.

You mention 48 and 49, but posted a screenshot of 50, I'm assuming thats the one in the bathroom? The SNR margins are pretty low (should be above 17 dBm), despite the other measurements being good. That means there is either a lot of noise or the node has a borderline connectivity and the slightest additional noise will lead to dropped packets.

There could be other sources of interference, e.g. water heater or other electrical devices near the nodes. Maybe the device is located somewhere where there is a lot of metal between it and the stick or near to it, causing reflections?

AlCalzone, 2022-06-21:
If there are radio issues on the device, individual bits might get flipped in a way that the packet still has a correct checksum, and gets identified as an S2 encapsulation command, but the command fails to decrypt.

Can't verify this for certain though since we don't have access to the raw data the device receives.

closed 2022-07-09 by zwave-js-assistant[bot]