zwave-js/zwave-js #4707: Zooz ZEN32 Missing Parameters 22 and 23.

efaden, 2022-06-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
Zooz ZEN32 is missing parameters 22 and 23.
What do you observe?

What did you expect to happen?
They should exist per https://www.support.getzooz.com/kb/article/608-zen32-scene-controller-advanced-settings/
Steps to reproduce the behavior:
  1. Go to '...'
  2. Click on '...'
  3. Scroll down to '...'
  4. See error

### Device information

Manufacturer: Zooz
Model name: Zen32
Node ID in your network: 

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

See https://www.support.getzooz.com/kb/article/608-zen32-scene-controller-advanced-settings/ 

Parameters 22 and 23 should exist, but don't in the template.

efaden, 2022-06-11:
Exist in FW 10.20 per https://www.support.getzooz.com/kb/article/706-zen32-scene-controller-change-log/

futurexdesign, 2022-06-13:
Fix for this was merged in #4619

AlCalzone, 2022-06-13:
And released in v9.4.0

closed 2022-06-13 by AlCalzone