zwave-js/zwave-js #4709: Danalock v3 BTZE not added securely, all features missing

kvisle, 2022-06-11:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

I don't know

What do you observe?

The lock does not pair securely.  No features available.

What did you expect to happen?

Lock paired securely.  Features available.  🤷‍♂️

Steps to reproduce the behavior:

1. Have Danalock V3-BTZE
2. Pair it
3. Start z-wave-js-server
4. Observe secure pairing failing in logs

### Device information

Manufacturer: Poly-control
Model name: Danalock V3-BTZE
Node ID in your network: 19

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [X] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 1.17.0
`node-zwave-js` branch: commit 6fa92e7120b3728e7bee1fb198a1d61c9c387229 (HEAD -> master, tag: 1.17.0, origin/master, origin/HEAD)

`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

Don't know, this is a new device

### Attach Driver Logfile

[zwave_2022-06-11.log](https://github.com/zwave-js/node-zwave-js/files/8884747/zwave_2022-06-11.log)

kvisle, 2022-06-11:
Uploaded a new log-file with silly loglevel.  Note, in this logfile, the node ID is 20, not 19 as in the previous file.
[zwave_2022-06-11-silly.log](https://github.com/zwave-js/node-zwave-js/files/8884780/zwave_2022-06-11-silly.log)

jmgiaever, 2022-06-12:
I have 20+ Dana Locks. All added secure S2. Is your lock very far away when you try to include it?

kvisle, 2022-06-12:
I've figured out what my problem was, and I'm writing it here in this issue in case it will be useful for others running into the same problem.

1. Initially (prior to this issue), my standalone z-wave-js-server installation didn't have network keys defined.  Copying example_config.js, and ensuring these entries exist:
```javascript
  securityKeys: {
    S0_Legacy: Buffer.from("32_CHARACTER_LONG_RANDOM_HEX_STRING", "hex"),
    S2_AccessControl: Buffer.from("32_CHARACTER_LONG_RANDOM_HEX_STRING", "hex"),
    S2_Authenticated: Buffer.from("32_CHARACTER_LONG_RANDOM_HEX_STRING", "hex"),
    S2_Unauthenticated: Buffer.from("32_CHARACTER_LONG_RANDOM_HEX_STRING", "hex")
  },
```
... this is not done for you by home assistant, and I couldn't find the keys documented anywhere.
The configuration file is effective if you run the process with the `--config configfile.js` parameter.

2. I paired the device incorrect.  I got a USB stick, and the way I've always done this is by moving the stick next to the device to be paired - set it to inclusion mode, and do it that way.  When starting z-wavejs-server again, things just seem to work.  Well - not this time.  Going through "Add device" in home assistant presented a popup that asked me to provide the user code.
3. I used the incorrect user code.  Earlier on, I scanned the QR code hidden in the battery lid of the lock.  It could might as well have been the correct code for all I knew -- well, it wasn't.  The correct code is the first section of the series of numbers.

Adding the device with network keys, using home assistant, and typing in the correct user code -- and voila.

During troubleshooting I did attempt to move the coordinator closer, but when doing the actual successful inclusion, I did no such thing.  The coordinator is 3-4 meters away from the lock.

AlCalzone, 2022-06-13:
> 2\. the way I've always done this is by moving the stick next to the device to be paired

I'm just gonna leave this here:
https://github.com/zwave-js/node-zwave-js/issues/1995

closed 2022-06-13 by AlCalzone