zwave-js/zwave-js #4715: [Bug]: Second Endpoint of Zooz ZEN30 double switch not listed in Target Endpoint dropdown of New Association dialog

laheym, 2022-06-15:


AlCalzone, 2022-06-15:
> Zooz ZEN30 double switch has two target endpoints

It only reports a single endpoint in the interview though:
```
2022-06-14T20:50:30.628Z DRIVER « [Node 062] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -63 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 97
                                    └─[MultiChannelCCEndPointReport]
                                        endpoint count (individual): 1
                                        count is dynamic:            false
                                        identical capabilities:      false
                                        endpoint count (aggregated): 0
```

Endpoint 0 is the root device (the node itself), which per the specs provides access to a subset of the functionality of the real endpoints, but is also often used to control all of them at once (e.g. for multi-channel relays).

I would assume that endpoint **1** is the relay and (the here non-existent) endpoint **2** the dimmer (or vice versa)?

In the past, devices often used endpoints unnecessarily, e.g. 1 = temperature sensor, 2 = humidity sensor, etc. although all of that information would also be provided on the root device. So in order to hide these unnecessary values that tend to confuse users and raise questions, Z-Wave JS has a heuristic when to ignore endpoints altogether. This happens when all of the endpoints have different device classes, i.e. provide entirely different functionality.

We can prevent this through a compat flag, but I need to be sure we're actually talking about endpoints 0+1 (which would be a weird choice IMO) instead of 1+2.

laheym, 2022-06-15:
> > Zooz ZEN30 double switch has two target endpoints
> 
> It only reports a single endpoint in the interview though:

I've forwarded this discrepancy to the Zooz developers. Hopefully we will get more information soon.

AlCalzone, 2022-06-22:
This should be fixed by #4736

To test, follow the instructions at https://zwave-js.github.io/zwavejs2mqtt/#/device-support/custom-device-files

you'll need these files in the custom config folder:
- [zen30.json](https://raw.githubusercontent.com/zwave-js/node-zwave-js/b27d403f815b66acdee91181458015bb5a90d0dc/packages/config/config/devices/0x027a/zen30.json)
- [templates/master_template.json](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/templates/master_template.json)
- [templates/zooz_template.json](https://raw.githubusercontent.com/zwave-js/node-zwave-js/master/packages/config/config/devices/0x027a/templates/zooz_template.json)

The 2nd and 3rd go into a new subdirectory `templates` if that was not obvious.

Don't forget to re-interview the device.

laheym, 2022-07-03:
hey AlCalzone, sorry for the delay testing the fix.  unfortunately  #4736  did not work as expected.  I installed the custom config as directed, and reset the container.   

I then tried to reinterview node 72, but it got stuck on protocolinfo. So... I tried to exlcude the node, but it stubbornly refused to be exclduded. Eventually, I disconnected power to the device and used remove failed node.  I then factory reset the device and tried to do a manual inclusion (as Node 73).   Now it is stuck at NodeInfo.

logs attached

[zwavejs2mqtt-store (12).zip](https://github.com/zwave-js/node-zwave-js/files/9035600/zwavejs2mqtt-store.12.zip)

laheym, 2022-07-03:
it seems that after the update, zen30 is now suffering from the same issue as #4585.  I applied #4595 and now the interview works.

the additional endpoints are showing now.... but it's still not quite right.  I can make associations to Endpoint 1 (which I presume to be the relay), but when I activate the switch/sensor with associations, the relay doesn't respond.    Associations to Endpoint 0 (the dimmer) works as expected. I am not sure if this is a zwavejs problem or a zooz problem though? 

updated config file (with #4595) attached

[zwavejs2mqtt-store (13).zip](https://github.com/zwave-js/node-zwave-js/files/9035634/zwavejs2mqtt-store.13.zip)

AlCalzone, 2022-07-04:
I'm in contact with them anyways, I'll ask about the association issue. Can you show how you defined the association?

PR #4736 (2022-07-19): fix(config): preserve endpoint 1 for Zooz ZEN30
fixes: #4715 
fixes: #4724
closed 2022-07-19 by AlCalzone