zwave-js/zwave-js #4717: HomeSeer HS-WS200+ Missing Parameter 6

mdw73, 2022-06-16:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug? HomeSeer HS-WS200+ switches do not allow you to change parameter 6 when you go into Z-Wave Device Configuration for the node.

What do you observe? Parameter 6 is not available.

What did you expect to happen? I expected parameter 6 to be available.

### Device information

Manufacturer: HomeSeer
Model name: HS-WS200+
Firmware: 5.13

Note: Parameter 6 (Enable/Disable Central Scene) was added in firmware 5.12:
https://homeseer.com/wp-content/uploads/2020/09/HS-WS200-Manual-v9.pdf

It's not available in Z-Wave JS:
https://devices.zwave-js.io/?jumpTo=0x000c:0x4447:0x3035:5.13

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version: 0.1.60
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

N/A

futurexdesign, 2022-06-18:
@mdw73 If you could try the patch I suggested in the PR, that would be a help to know if it fixes the issue.

mdw73, 2022-06-18:
@futurexdesign I'd be more than happy to--are there instructions somewhere on how to do that?

futurexdesign, 2022-06-18:
If you are using the Z-Wave JS to MQTT addon, you can just upload the [file](https://raw.githubusercontent.com/zwave-js/node-zwave-js/206ca5967afedab8fbe96bc6f135f62a93a67437/packages/config/config/devices/0x000c/hs-ws200.json) into the config priority directory using the built-in file manager.  You might also need to upload the templates, but I can't recall exactly at the moment.    The new config file should get loaded when the add-on restarts. 

If you are using the home-assistant provided add-on, I'm not sure how to inject the configuration into that, as I don't use it.

AlCalzone, 2022-06-20:
Or you can wait for the next release :)

closed 2022-06-20 by AlCalzone

mdw73, 2022-07-26:
I'm now running Home Assistant 2022.7.6 and Z-Wave JS 0.1.64 and the option still isn't displaying. I also don't see Parameter 6 listed on this page, either:

https://devices.zwave-js.io/?jumpTo=0x000c:0x4447:0x3035:5.13

AlCalzone, 2022-07-26:
Did you re-interview your device? The config DB website isn't up to date.

kpine, 2022-07-26:
Z-Wave JS 0.1.64 add-on is still using driver v9.4.0.

PR #4720 (2022-06-20): fix(config): add param 6 to HomeSeer HS-WS200+, FW 5.12+
Add missing parameter 6 for the HomeSeer HS-WS200+  Fixes #4717