zwave-js/zwave-js #4719: [Bug]: Zooz Zen34 fails inclusion

laheym, 2022-06-17:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

## What causes the bug?

## What do you observe?
Zoon Zen34 inclusion fails - it is stuck at NodeInfo.  Behavior is identical to #4659.  Possibly this device behavies like this https://github.com/zwave-js/node-zwave-js/issues/4585#issuecomment-1121140769 ?

## What did you expect to happen?
device includes correctly
## Steps to reproduce the behavior:
put zwavejs into inclusion mode (via gui)
include node

### Device information

Manufacturer: Zooz
Model name: Zen34
Node ID in your network: 
Inclusion failed when I first attempted to include it as Node66.   Then I added a custom config entry with the #4666 compat flag workaround and reset the container.  I was excluded and then reincludes (successfully) as Node 67

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:
`zwavejs2mqtt` branch:

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### Attach Driver Logfile

[zwavejs_2022-06-17.log.gz](https://github.com/zwave-js/node-zwave-js/files/8931324/zwavejs_2022-06-17.log.gz)

AlCalzone, 2022-06-20:
@laheym Unfortunately your logfile didn't upload, can you try again? I want to verify that this is indeed the same issue.

RJ-Make, 2022-06-20:
Sorry to butt in.. I had the same exact problem, (I should still have the logs)

After something like 4 attempts, over several hours. I removed the node, performed a factory reset, pulled the batteries, added the device using Smart Scan this time, reinstalled batteries and shortly after kept waking the device. (7 up taps)

On the 2nd or 3rd wake it completed the interview.

RJ-Make, 2022-06-21:
I've added my logs. Failed node interviews were 136 and 137. Succeed interview on node 138 via Smart Scan

[zwavejs2mqtt-store.zip](https://github.com/zwave-js/node-zwave-js/files/8949430/zwavejs2mqtt-store.zip)
n

AlCalzone, 2022-06-21:
Ok this definitely is https://github.com/zwave-js/node-zwave-js/issues/4585 and should be fixed by doing the same as in https://github.com/zwave-js/node-zwave-js/pull/4666

If anyone would like to help, please go ahead.

closed 2022-06-21 by AlCalzone

AlCalzone, 2022-06-21:
@laheym I'm happy to take a look, but that's not the correct file.

PR #4731 (2022-06-21): fix(config): work around Configuration Info reporting bug in ZEN34
Add config workaround for Zooz ZEN34.  Fixes #4719