zwave-js/zwave-js #4729: [Bug]:

anonymous66va, 2022-06-21:


AlCalzone, 2022-06-21:
What @robertsLando said.

And the information from previous interview shouldn't just get forgotten, which it did here. Looks like somehow your upgrade caused the cache files to get lost. I'd look into that if I was you.

anonymous66va, 2022-06-25:
[zwavejs_2022-06-24.zip](https://github.com/zwave-js/node-zwave-js/files/8984405/zwavejs_2022-06-24.zip)
Here the log file  for nodes 40 , 41 ,48

anonymous66va, 2022-06-27:
> I don't know what you did to your log, but this only includes the raw serial data, no human-readable interpretation. If you have any filters on, disable them please and try again. Please delete the existing logfile before starting and make sure to re-interview at least one of the problematic nodes while the log is being made.

Ciao , what am I doing wrong ?
Thanks a lot
[image]

AlCalzone, 2022-06-27:
Remove these:
[image]

anonymous66va, 2022-06-27:
Ciao , as per your request I removed nodes 3 & 40 & tried to re-interviewed node 64.
I attach the log result.
I don't know where to check if the filer log is still working.
BR
antonio
[zwavejs_2022-06-27(1).log](https://github.com/zwave-js/node-zwave-js/files/8991887/zwavejs_2022-06-27.1.log)

AlCalzone, 2022-06-30:
Ok, so the issue is that your devices report bogus alarm sensor types:
```
2022-06-27T11:05:45.096Z CNTRLR   [Node 064] [~] [Alarm Sensor] supportedSensorTypes: 8, [Endpoint 0] [internal]
                                  9,17,24,32,34,42 => 8,9,17,24,32,34,42
2022-06-27T11:05:45.097Z SERIAL » [ACK]                                                                   (0x06)
2022-06-27T11:05:45.098Z DRIVER « [Node 064] [REQ] [ApplicationCommand]
                                  └─[AlarmSensorCCSupportedReport]
                                      supported sensor types: unknown (0x08), unknown (0x09), unknown (0x11), un
                                  known (0x18), unknown (0x20), unknown (0x22), unknown (0x2a)
2022-06-27T11:05:45.101Z CNTRLR « [Node 064] received supported sensor types: 
                                  · unknown (0x08)
                                  · unknown (0x09)
                                  · unknown (0x11)
                                  · unknown (0x18)
                                  · unknown (0x20)
                                  · unknown (0x22)
                                  · unknown (0x2a)
```
and these break a stricter validation of API method arguments I put in place a while ago. I'll just tone down the strictness of that check in the next release, don't think it makes much sense to work around this.

closed 2022-06-30 by AlCalzone