zwave-js/zwave-js #4746: Timeouts during interview with Allegion/Schalge BE469ZP

mattstocum, 2022-06-26:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

NO, my problem is NOT within Home Assistant or the ZWave JS integration

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?

Interview a Schalage BE469ZP

What do you observe?

Repeated timeouts during interview

```
2022-06-26T18:36:30.791Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-06-26T18:36:30.791Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:30.791Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-06-26T18:36:30.811Z SERIAL « 0x011d00a97600000100ac7f7f7f7f01010300000000020100007f7f7f7f7fef    (31 bytes)
2022-06-26T18:36:30.812Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:30.812Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            118
                                    transmit status:        OK, took 10 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -84 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2022-06-26T18:36:31.844Z CNTRLR   [Node 090] Timed out while waiting for a response from the node (ZW0201)

```

What did you expect to happen?

Interview to complete without errors.

Steps to reproduce the behavior:
1. Include or trigger a re-interview of a BE469ZP.

### Device information

Manufacturer: Allegion
Model name: BE469ZP
Node ID in your network: 90 and 91 (I have two different devices showing the same results)

### How are you using `node-zwave-js`?

- [X] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [ ] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch: 9.4.0
`zwavejs2mqtt` branch: 6.12.1

### Did you change anything?

yes (please describe)

### If yes, what did you change?

I finally upgraded the firmware on my Zooz 700 series controller and moved my network back to it. I was getting issues with repeated notifications from the locks when the door was locked or unlocked so I decided to try excluding and re-including the device. This did solve the multiple notification issue, but now I'm having trouble interviewing the device.

### Did this work before?

Yes (please describe)

### If yes, where did it work?

At some point in the past the device had interviewed properly.

### Attach Driver Logfile

[zwavejs_current.log](https://github.com/zwave-js/node-zwave-js/files/8987297/zwavejs_current.log)

mattstocum, 2022-06-26:
I suspect the issue is related to the Beaming frequency for the locks. It wakes up once every second, and it looks like the timeout is occurring at 1 second. I've found if I refresh all values on the device, and trigger a re-interview while the refresh is happening it completes more of the interview. Just to see if I could increase the timeout value, I added the following to /usr/src/app/lib/ZwaveClient.ts in the zwavejs2mqtt container before restarting the container:

```
                        // extend options with hidden `options`
                        const zwaveOptions: utils.DeepPartial<ZWaveOptions> = {
                                timeouts: {
                                        report: 2000,
                                },
                                storage: {
```

It didn't make any difference however.

AlCalzone, 2022-06-27:
Honestly, I believe this is just the lock acting up. I can't remember how many issues I've seen related to this exact model and its always the lock doing something strange.

The timeout is computed automatically from (round-trip-time + report timeout). The round-trip-time (RTT) is the time delta between sending the request and receiving the node's acknowledgement - meaning the node did already answer, so it cannot be asleep.

During the stretch of commands you quoted, it consistently had a RTT of 30-40 ms and replies in ~300-400 ms:
```
2022-06-26T18:36:29.575Z DRIVER » [Node 090] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      115
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 22
                                    └─[NotificationCCEventSupportedGet]
                                        notification type: Access Control
2022-06-26T18:36:29.581Z SERIAL « [ACK]                                                                   (0x06)
2022-06-26T18:36:29.582Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-06-26T18:36:29.582Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:29.582Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-06-26T18:36:29.603Z SERIAL « 0x011d00a97300000200ad7f7f7f7f01010300000000020100007f7f7f7f7fe8    (31 bytes)
2022-06-26T18:36:29.604Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:29.604Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            115
                                    transmit status:        OK, took 20 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -83 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2022-06-26T18:36:29.907Z DRIVER « [Node 090] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -83 dBm
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 237
                                    └─[NotificationCCEventSupportedReport]
                                        notification type: Access Control
                                        supported events:
                                        · Manual lock operation
                                        · Manual unlock operation
                                        · Keypad lock operation
                                        · Keypad unlock operation
                                        · Auto lock locked operation
                                        · Lock jammed
                                        · Keypad temporary disabled
                                        · New program code entered: unique code for lock configuration

2022-06-26T18:36:29.909Z DRIVER » [Node 090] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      116
                                  └─[Security2CCMessageEncapsulation]
                                    │ sequence number: 23
                                    └─[NotificationCCEventSupportedGet]
                                        notification type: Home Security
2022-06-26T18:36:29.916Z SERIAL « [ACK]                                                                   (0x06)
2022-06-26T18:36:29.916Z SERIAL « 0x010401a90152                                                       (6 bytes)
2022-06-26T18:36:29.916Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:29.917Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2022-06-26T18:36:29.937Z SERIAL « 0x011d00a97400000100ad7f7f7f7f01010300000000020100007f7f7f7f7fec    (31 bytes)
2022-06-26T18:36:29.937Z SERIAL » [ACK]                                                                   (0x06)
2022-06-26T18:36:29.938Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            116
                                    transmit status:        OK, took 10 ms
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -83 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1

2022-06-26T18:36:30.340Z DRIVER « [Node 090] [REQ] [BridgeApplicationCommand]
[... 30 lines cut ...]
                                    ACK RSSI:               -84 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1

... more communication, each time with almost immediate ACKs

2022-06-26T18:36:35.534Z DRIVER « [Node 090] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -84 dBm
[... cut ...]
                                        · System hardware failure
```

You should be able to get around most of this by upping the report timeout to 10000. 2000 is not enough for these hiccups.
You can also do that in the settings.json, so you don't have to edit the code:
https://zwave-js.github.io/zwavejs2mqtt/#/usage/setup?id=z-wave -> Hidden settings. Basically you need to add
```json
"timeouts": {
  "report": 10000
}
```
inside the "zwave" key in the settings.json (you can reformat it to be readable).

mattstocum, 2022-06-27:
And for anyone else who might find this issue in the future while trying to get these stupid locks working, here's the "zwave" portion of settings.json with the increased timeout. I have removed the "securityKeys" section, you should obviously leave yours intact. The part I added is the entire "options" section. After the interview is complete, I think the section should be removed, to not cause unnecessary delays.

```
    "zwave": {
        "port": "/dev/zwave",
        "commandsTimeout": 60,
        "logLevel": "silly",
        "logEnabled": true,
        "deviceConfigPriorityDir": "/usr/src/app/store/config",
        "logToFile": true,
        "serverEnabled": true,
        "enableSoftReset": true,
        "enableStatistics": true,
        "serverPort": 3000,
        "nodeFilter": [],
        "disclaimerVersion": 1,
        "options": {
            "timeouts": {
                "report": 10000
            }
        }
    }
```

closed 2022-07-01 by zwave-js-assistant[bot]