zwave-js/zwave-js #4757: Firmware updates arent working [question]

no1warr1or, 2022-06-29:


AlCalzone, 2022-06-29:
Without seeing logs, probably a flaky connection. The update process is driven by the end device and if it stops requesting the update fragments for a while, the driver will abort the update.

no1warr1or, 2022-07-06:
Apologies for the delay. What I'm noticing is the devices are communicating fine then the response time will shoot through the roof 1000+ ms. Some of the nodes go dead then come back alive. And I have a ton of dropped commands. But when the log shows the update is "done" its not experiencing these symptom at that time. I will grab the logs on the next update (I'm waiting for a response from zooz on one) the others I just kept trying until they actually completed.

I've gone through some of the zwave troubleshooting. It's on a USB extension and it's as far away from my rack as I can get it. I'm wondering if the usb cable is picking up interference? 

I'll provide more information once I get the update file and get the time to diagnose the zwave network alittle more

closed 2022-07-17 by zwave-js-assistant[bot]