zwave-js/zwave-js #4773: Response suddenly very slow - is quick when pinging first

dawiinci, 2022-07-04:
Since a few days 6.13.0? I have a strange behavior of devices. 

If I click on the switch in the frontend of home assistant I get an immediate response. If I use the wall switch I get a delay of about two seconds. If I ping the device right before I switch it on, the response is immediate.

rtt is 68 ms.

Is there something wrong in my settings or is this a bug?

Log is attached:
[zwavejs_2022-07-03.log](https://github.com/zwave-js/zwavejs2mqtt/files/9034241/zwavejs_2022-07-03.log)
 
I first switched on and off using the wall switch. It was delayed. The last two events were using the ping before pressing the switch - no delay.

dawiinci, 2022-07-04:
I healed the network multiple times, I reconfigured all devices multiple times.

Health check revealed some devices are 0/10. But I am not sure how to handle that. So assume there is a faulty device that leads to this if routed over this device, how can I find it?

AlCalzone, 2022-07-05:
You can enable driver logs https://zwave-js.github.io/zwavejs2mqtt/#/troubleshooting/generating-logs?id=driver-logs, open them in a new window and ping the device.
You should see something like this in the log:
```
2022-07-03T07:13:29.881Z DRIVER « [REQ] [SendData]
                                    callback id:            7
                                    transmit status:        OK, took 90 ms
                                    repeater node IDs:      66, 126, 26
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    ACK RSSI:               -74 dBm
                                    ACK RSSI on repeaters:  N/A, N/A, N/A
                                    ACK channel no.:        1
                                    TX channel no.:         1
```

The line with `repeater node IDs` tells you which nodes the packet gets routed through. If there are routing issues, there will also be a line "route failed here" which tells you between which nodes the problem was.
You should also be able to see this information on the network map if you click the node. This shows a panel with the route info.

> So assume there is a faulty device that leads to this if routed over this device, how can I find it?

To really be sure, test the connection of all nodes using the health check tool you'll find on the network map (after clicking on a node) - this will take a while. Start with the ones close to the controller and work your way outward. Select the controller as the target node, 5 rounds should be good.

dawiinci, 2022-07-06:
Most devices are 0/10 even though they are close. They do work though, but sometimes have a delay or are „dead“.  

Could this be a faulty controller?

What USB controller is recommended to replace the UZB1? Aeotec Gen7?

AlCalzone, 2022-07-06:
> Most devices are 0/10 even though they are close

What causes the 0 score? Is there any more information in the report?
A driver log of one such node test would also be good to see.

dawiinci, 2022-07-06:
I have more information:
[image]
[zwavejs_2022-07-06.log](https://github.com/zwave-js/node-zwave-js/files/9052444/zwavejs_2022-07-06.log)

The device actually works.

AlCalzone, 2022-07-06:
> SNR margin of -194

I think that's actually an error in the calculation. This happens when the background noise is below the detectable threshold.

> But this is in line with what I experience. If I switch the device from the controller it is immediate, but the response is slow.

```
2022-07-06T07:06:28.575Z DRIVER « [REQ] [SendData]
                                    callback id:            42
                                    transmit status:        OK, took 150 ms
                                    repeater node IDs:      126
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 9.6 kbit/s
                                    ACK RSSI:               -66 dBm
                                    ACK RSSI on repeaters:  N/A
                                    ACK channel no.:        1
                                    TX channel no.:         1
```

All commands to node 122 go through 126 and one of them has a really slow connection. `9.6 kbit/s` is the lowest that Z-Wave will go and this is usually just a fallback if nothing else works.

I'm relatively sure this is happening because of the extremely high background noise:
```
2022-07-06T07:06:40.875Z DRIVER « [RES] [GetBackgroundRSSI]
                                    channel 0: -89 dBm
                                    channel 1: -63 dBm
```
The commands are using channel 1 and have RSSI values that are at or sometimes even below (!) the background noise.

I'm usually seeing background RSSI values of around -90..-100 dBm, so there is something blasting heavily on the Z-Wave frequency. I'm afraid you'll have to figure out what that is and eliminate it.

dawiinci, 2022-07-06:
Make sense. How can I find the source?

Is it another zwave device or can it be anything? Can I live check monitor rssi?

AlCalzone, 2022-07-06:
Can be anything really. You can build your own live monitor in z2m by opening the log in a separate window and executing the following driver function:

```ts
const measurements = 5; // how many measurements
const frequency = 1000; // how many ms between measurements

let i = 0;
let interval = setInterval(async ()=>{
  await driver.controller.getBackgroundRSSI();
  if (++i > measurements) clearInterval(interval);
}, frequency);
```
You can tune it by changing the frequency and number of measurements. This example does 5 measurements in 1s intervals.

[image]

And here's the corresponding log:
[image]

dawiinci, 2022-07-09:
Background noise is reported much lower when I connect it to a Windows Laptop using Z-Way instead of the M1 Mac mini using Z2M (-101 vs. -88). It occasionally drops to ~90.

Although I am not sure if Z2M just reports the max value in that case, that is in line with Z-Way, since it reports between -101 and ~-90. It seems to be better when using USB-A instead of the USB-C port with a hub on the Mac (-77 vs -88).

But what I don't understand is why a device fails with 0/10 because the node didn't send anything. That makes no sense since it actually reports and works. Could this be an incompatibility with the health check? Or is it because it isn't routing at all when choosing "Controller" (node 1)?

dawiinci, 2022-07-17:
To add to this weirdness:
[image]

Why is the first test 10/10 when the back route wasn't tested?

Is there a bug that makes the answers fail or is it the device that is not fully supporting this type of check?

AlCalzone, 2022-07-18:
Sorry for the late reply, I was on vacation.

The reported SNR values are the worst case, since this is what you'll be working with normally.
The one line where there were no errors on the return direction (hence none were shown) is also the only case where a power level reduction of -6dBm still resulted in failure-free communication. "Normal power" means there can be no reduction in power without communication failures, which is bad.

Do you have the stick plugged in directly to the USB ports or are you using an extension cord? If plugged in directly, you should definitely use an extension cord. 30cm/1' should be enough. USB ports generate a ton of noise, especially USB3.
Not sure why I didn't ask this first thing...
See also https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting

closed 2022-08-07 by zwave-js-assistant[bot]