zwave-js/zwave-js #4780: zWave 700 series drop nodes on latest firmware

kdober, 2022-07-05:
### Is your problem within Home Assistant (Core or Z-Wave JS Integration)?

YES, BUT a Home Assistant developer has told me to come here

### Is your problem within ZWaveJS2MQTT?

NO, my problem is NOT within ZWaveJS2MQTT

### Checklist

- [X] I have checked the [troubleshooting section](https://zwave-js.github.io/node-zwave-js/#/troubleshooting/index) and my problem is **not** described there.

- [X] I have read the [changelog](https://github.com/zwave-js/node-zwave-js/blob/master/CHANGELOG.md) and my problem was **not** mentioned there.

### Describe the bug

What causes the bug?
I'm not sure.

What do you observe?
Some Zwave Nodes go to dead status, and they usually don't come back by them self. When this happens, the device remains dead and the control of them are no longer possible from dashboards (gray out). However, just doing "ping/press button" within HA device brings them back immediately. 

What did you expect to happen?
Not sure, if something really happens with the node, and it became unavailable, it should come back 'online' without manual intervention?

Steps to reproduce the behaviour:
Just sit and wait for a node to became death. It doesn't get back, except I do a ping to the device.

Current system (all at the latest version):
Zwave: Aeotec 700
ZST10-700
by Silicon Labs
Firmware: 7.17.2 

zWaveJs: 
Current version: 0.1.64
Driver Version: 9.4.0
Server Version: 1.20.0
Server URL: ws://core-zwave-js:3000 

HAOS
Version | core-2022.6.7
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
User | root
Virtual Environment | false
Python Version | 3.9.12
Operating System Family | Linux
Operating System Version | 5.15.45
CPU Architecture | x86_64

### Device information

Manufacturer: Aeotec
Model name: 700 series
Node ID in your network: Node 001 (for aeotec stick). Node 002 for failing example (if this is what you mean)

### How are you using `node-zwave-js`?

- [ ] `zwavejs2mqtt` Docker image (latest)
- [ ] `zwavejs2mqtt` Docker image (dev)
- [ ] `zwavejs2mqtt` Docker manually built _(please specify branches)_
- [ ] `ioBroker.zwave2` adapter _(please specify version)_
- [X] `HomeAssistant zwave_js` integration  _(please specify version)_
- [ ] `pkg`
- [ ] `node-red-contrib-zwave-js` _(please specify version, double click node to find out)_
- [ ] Manually built from GitHub _(please specify branch)_
- [ ] Other _(please describe)_

### Which branches or versions?

version:
`node-zwave-js` branch:  0.1.64
`zwavejs2mqtt` branch: Not installed

### Did you change anything?

no

### Did this work before?

No, it never worked anywhere

### If yes, where did it work?

All devices working, but a few are dropping from time to time and they don't go back if I don't make a 'Ping'
My zwave installation in HA is new (about 2 weeks). Before they were under Hubitat (working fine) and linked to HA with the hubitat-HA integration. However, I bought the Aeoteck Stick 7 and migrated all of them

### Attach Driver Logfile

I attach a few logs where I was able to catch the full sequence. This is a node becoming death, waiting a long time and then just doing a ping to bring it back.
Pay attention to Node 002 at 2022-07-05T18:25:36 (line 9327 of zwzje_js_20220705_3.log, last entry for "node 002")
Then, 3 hours later, the execution of the ping to that node at 2022-07-05T21:13:35  (line 21478 of zwzje_js_20220705_4.log, first entry for "node 002")

I also attach the dump (which I don't remember how I got it) and the "700 Series-based Controller" diagnosis

Please, just let me know is there is anything I can do to help out to debug this issue.
Best regards,
Cristian

[zwave_js_20220705_3.log](https://github.com/zwave-js/node-zwave-js/files/9050279/zwave_js_20220705_3.log)
[zwave_js-7f56a28154daa0e0b7c8246f24bcdfd0-700 Series-based Controller-ba20ddd4b6238eb2b218cda9a9f164ee.json.txt](https://github.com/zwave-js/node-zwave-js/files/9050280/zwave_js-7f56a28154daa0e0b7c8246f24bcdfd0-700.Series-based.Controller-ba20ddd4b6238eb2b218cda9a9f164ee.json.txt)
[Uploading zwave_js_dump.4.json.txt…]()

[zwave_js_20220705_4.log](https://github.com/zwave-js/node-zwave-js/files/9050277/zwave_js_20220705_4.log)

kdober, 2022-07-06:
Hi,

I'm not sure then how to generate the logs.
Far as I understood, they are generated by going to Settings -> Devices & Services -> Zwave (Configure) -> Logs tab, then, leave browser open and download generated logs.
The link provided on the msg seems to be for ZWavejs2Mqtt, which I don't have installed.

Am I missing something?

Cheers,
Cristian

AlCalzone, 2022-07-06:
Its fine, the heuristic to consider a log a driver log just doesn't understand how you renamed the files.

AlCalzone, 2022-07-06:
The RSSI values seem low, so this could partially be caused by bad connectivity. Please read https://zwave-js.github.io/node-zwave-js/#/troubleshooting/connectivity-issues?id=general-troubleshooting and make sure you've followed the steps outlined there.

Besides that, I think the main issue is that you have a couple of very chatty nodes (2, 6 and 8) that flood the network with lots of reports up to the point that the controller is unable to send. This might not be an issue after you've optimized the controller location and healed, but optimizing the reporting won't hurt:
Try configuring all devices to report less. I see the aforementioned nodes report over and over that they are still not using any power.
Where possible, prefer reporting on significant changes (like changes of >5W or so) over timed reports. If you need timed reports e.g. as a backup, reduce the frequency significantly (to sth. like once per hour).

kdober, 2022-07-06:
Hi AlCalzone,

Thanks for your quick reply!
Hub is located in the centre of the house and devices are not far away. I will try the usb cable cord to place further away from the NUC and other electronics in the area.

You are absolutely right that nodes 2,6 and 8 are too chatty. See image below.
[image]
[image]

I'm not sure to understand why this happens. These are all Fibaro devices (Double switch and dimmers) and I don't get why some of them and reporting that often and others are not. Could you help me out to understand this (or point me where I can find this information)?
Additionally, the reporting parameters are the defaults, 20% change in power (now raised to 40%) and energy reported every 1kw increments (for Led bulbs that is quite a lot of time).
There is a third parameter called "minimal time between power reports", however, it cannot be set, and I have no clue what value it contains (see image)
[image]
Haven't found other parameters about reporting frequency.

Anyway, I will continue testing with the suggestions and will investigate this parameter I cannot change.

At last, my doubt remains about why the controller never reach the device again once it gets the 'dead status', shouldn't it try to reach it again after some time? Is this something that can be configured?

Thanks!
Cheers,
Cristian

AlCalzone, 2022-07-18:
I think this is more of a connectivity issue. The nodes don't seem to get the acknowledgements from the controller and try to reach it over and over.

This is probably related to the stick's RF range, which seems to be lower than intended. See also this issue where this is being investigated:
https://github.com/zwave-js/node-zwave-js/issues/4352#issuecomment-1174988913

You might have some luck placing additional repeaters between the controller and the problematic nodes.

kdober, 2022-07-18:
Hi 

Thanks AlCalzone
It improved quite a lot when started using the usb cord extension. It didn't initially, but that was due to own silliness.... when I installed the Zigbee dongle last Saturday I realized I placed the mouse usb receiver on the extension cord instead of the zwave stick (doh)
Anyway, it is working much better now. Still a few short drops from time to time. Additionally, I made a node-red flow to ping a node when it is dead for 3 mins.

AlCalzone, 2022-07-18:
Good - you might have some luck using zwavejs2mqtt's health check (on the network map) to figure out which repeaters are problematic now.

closed 2022-08-05 by zwave-js-assistant[bot]