zwave-js/zwave-js #4786: Update Zooz device files

blhoward2, 2022-07-08:
To account for new devices, per Zooz:

Below are the new settings which have been added to the following dimmer versions so far:

ZEN72 10.10 (ver. 1.0 hardware) and 2.0 (ver. 2.0 hardware)
ZEN74 10.10 (ver. 1.0 hardware, no 2.0 hardware available at the moment)
ZEN77 10.20 (ver. 1.0 hardware), 2.10 (ver. 2.0 hardware), and 3.0 (ver. 3.0 hardware)

Please let me know if you have any questions here!

Remote Z-Wave Dimming Duration (Multilevel Commands For Group 3 And 4)

Parameter 30: Set the time it takes to get from 0% to 100% brightness on dimmers and smart bulbs directly associated with your dimmer in Groups 3 and 4 when pressing and holding the paddle (physical dimming) on your dimmer. The number entered as value corresponds to the number of seconds.
Values: 1 – 99 (seconds). Default: 5.
Size: 1 byte dec

Ramp Rate Control

Parameter 9: Adjust the ramp rate ON for your dimmer when the top paddle is pressed (physical ramp rate on). Values correspond to the number of seconds it take for the dimmer to reach full brightness when operated manually.
Values: 1 – 99 (seconds). 0 – instant on. Default: 0.
Size: 1 byte dec
 NOTE:  On all other previous firmware versions including 1.0 and 10.0, this setting was used for all ramp rate control (to adjust behavior for on/off physical and Z-Wave ramp rate). The default value for this parameter used to be 1.
 
Parameter 27: Adjust the ramp rate OFF for your dimmer when the bottom paddle is pressed (physical ramp rate off). Values correspond to the number of seconds it take for the dimmer to turn off when operated manually.
Values: 1 – 99 (seconds). 0 – instant off. Default: 2.
Size: 1 byte dec
 
Parameter 28: Adjust the ramp rate ON for your dimmer when a Z-Wave command is sent to turn the dimmer on (Z-Wave ramp rate on). Values correspond to the number of seconds it take for the dimmer to reach full brightness when operated via Z-Wave.
Values: 1 – 99 (seconds). 0 – instant on. 255 – match physical. Default: 255.
Size: 1 byte dec
 
Parameter 29: Adjust the ramp rate OFF for your dimmer when a Z-Wave command is sent to turn the dimmer off (Z-Wave ramp rate off). Values correspond to the number of seconds it take for the dimmer to turn off when operated via Z-Wave.
Values: 1 – 99 (seconds). 0 – instant off. 255 – match physical. Default: 255.
Size: 1 byte dec

jordahlm, 2022-09-08:
I would also like this to be updated. I just got a several of these dimmers with the 2.0 hardware and firmware and can't properly set the ramp rate parameters as the current device files only have a single parameter (9) for this.

alvinchen1, 2022-09-27:
> I would also like this to be updated. I just got a several of these dimmers with the 2.0 hardware and firmware and can't properly set the ramp rate parameters as the current device files only have a single parameter (9) for this.

@jordahlm, Hi, I am trying to write the configuration parameters for your device.  I added the ones for hardware version 1.0, but don't have a way to checking for hardware version 2.0 and 3.0 of the ZEN77.  Do you have ZEN77 ver 2.0 or 3.0 devices?  If you do, could you please go into Z-WaveJS, go to your device.  Click the **Configure** Link under **Device Info**

[image]

Then click on the **Z-Wave JS Device Database link** 

[image]

and reply with the URL it takes you to?  That might help me figure out how to write the configuration file for version 2.0 and 3.0 of the ZEN77.

The ZEN72 and ZEN74 configuration parameters should trickle down in the next release or so once [PR 5125](https://github.com/zwave-js/node-zwave-js/pull/5125) gets merged.

jordahlm, 2022-09-28:
Sorry, I don't currently have any hardware version 2.0 Zen77s, only Zen72s. I may have a 2.0 Zen77 soon and will update here if I do.

alvinchen1, 2022-09-28:
No problem.  Hopefully someone does and if they submit that information, I can look to write the config file for it.

blhoward2, 2022-09-29:
The information in my original post came from Zooz, so you can just go ahead and add them. No need to verify for us to merge it.

AnthonyMaiorani, 2022-10-01:
> I added the ones for hardware version 1.0, but don't have a way to checking for hardware version 2.0 and 3.0 of the ZEN77. Do you have ZEN77 ver 2.0 or 3.0 devices? If you do, could you please go into Z-WaveJS, go to your device.

@alvinchen1 This is the link my ver 2.0 ZEN77 brings me to: https://devices.zwave-js.io/?jumpTo=0x027a:0x7000:0xa007:0.0

alvinchen1, 2022-12-01:
@AnthonyMaiorani sorry to shake off the cobwebs on this one, but there is a suggestion on how to get the Hardware Version of the ZEN77 if you are running Z-WaveJS UI you could help us out?  See https://github.com/zwave-js/node-zwave-js/pull/5080.

PR #5317 (2023-01-26): feat(config): add parameters for new Zooz ZEN72/74/77 firmwares, fix ZEN05 param 1

Revised https://github.com/zwave-js/node-zwave-js/pull/5156.

~~Partially addresses issue https://github.com/zwave-js/node-zwave-js/issues/4786~~
Added parameter 27, 28, 29 to zen77.
Added parameter 30 to zen72, zen74 and zen77 hardware v1.
Fix zooz zen05 param 1 https://github.com/zwave-js/node-zwave-js/issues/5248

Still no resolution for hardware versions where firmware drops below version 10.x (i.e. version 2.0, 3.0) differentiation see https://github.com/zwave-js/node-zwave-js/issues/5248 and https://githu [... cut]